pikvm/pikvm #1057: Can't see preboot/bios screen anymore in my Beelink pc

plia7, 2023-07-15:
**Describe the bug**
When restarting my Beelink SER5 pc the preboot/bios screen is not showing anymore

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect your pikvm to your Beelink SER5 through hdmi and usb cables.
2. Turn on your beelink or restart it

**Expected behavior**
PC Beelink Logo is showing, possibility to enter the BIOS, etc

**Actual behavior**
As soon as I press restart in windows 11 start menu, the screen stops showing, not even showing "restarting" message,
neither it shows when the pc is starting up again, I have to refresh the pikvm web interface to see the pc back in Windows.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 11
 -PC:
Beelink Mini PC, AMD Ryzen 5 5500U(6C/12T, Up to 4.0GHz), 16GB DDR4 RAM 500GB NVMe SSD, SER5 Win 11 Mini Desktop Computer Support 4K@60Hz Output/WiFi6/BT5.2/DP1.4/HDMI2.0/USB3.2 for Gaming/Office/Home
https://www.amazon.com/gp/product/B0B2RHXLDK
*All pc drivers/windows updates are up to date.
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
 kvmd-platform-v3-hdmi-rpi4 3.234-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
When I connect the beelink pc to an external monitor through hdmi, I can see the preboot screen such as the beelink logo, etc.
For reference, this used to work in kvmd 3.169 (nothing substantial changed in the beelink pc drivers wise, only pikvm version was upgraded).

plia7, 2023-07-15:
@mdevaev This is the same pikvm from two weeks ago that I flashed a new micro sd card with the latest image
I updated, set passwords, enable VNC, install tailscale, update configs as needed.
In addition, I copied the exact same EDID file from kvmd 3.169 where this used to work with this pc.

EDID command for reference:

kvmd-edidconf --set-mfc-id DEL --set-monitor-name "Dell Monitor" --set-audio yes --set-serial 3405691582 --set-product-id 16640

plia7, 2023-07-16:
@mdevaev Some additional finding that might be useful to diagnose the issue: As you know, switching to mjpeg when it becomes black screen when you click restart, does "fix" it and it makes the screen to show "Restarting...", beelink logo showing, ability to enter bios etc, so I did exactly that, switched to mjpeg, then I entered the bios hitting F7 when the logo appeared, while in bios, I changed it back to webrtc, it still worked, I was still able to see the screen. Then I exit bios, it comes back to windows, now I do another restart, and it still works while in webrtc. 
So how do you break from this "working cycle"? You simply refresh the pikvm webpage, just to be on the safe side, I open an incognito session, and when I tried to do the restart button again, it went black screen again, so I was able to reproduce and breaking from the working cycle. Hopefully this provides some additional clues on what's going on.

plia7, 2023-07-16:
@mdevaev I performed another test per amt921 which seem to be fix it as well:
Bring it to the black screen issue, by pressing restart in windows, then X it and click show stream - that seem to fix it.

plia7, 2023-07-21:
@mdevaev It's been a week since the issue was created. Are there any updates on the fix to the issue? Were you able to reproduce it? I see it still has the "question" label, shouldn't it be a "bug" label at this point?

mdevaev, 2023-07-23:
This is a problem on the web client side. Still haven't figured out the reason.

mdevaev, 2023-07-23:
Fixed, update OS.

closed 2023-07-23 by mdevaev

plia7, 2023-07-23:
> Fixed, update OS.

I just updated OS, but the issue is not fixed.

kvmd-platform-v3-hdmi-rpi4 3.237-1