pikvm/pikvm #1058: Possible to reset the USB-HDMI dongle via cmd ?

manasmgkar, 2023-07-15:
cant ge the hmdi-csi2 bridge due to chinese shipping bans in my country, with the USB-HDMI capture card/dongle if there is a black screen issue or some other issue it is possible to reset it via the command line in pikvm ? basically, simulate a disconnect and reconnect ?

mdevaev, 2023-07-15:
USB capture is a black box that we cannot influence in any way, alas.

manasmgkar, 2023-07-17:
> USB capture is a black box that we cannot influence in any way, alas.

but cant the usb port that it is connected to be reset to simulate a dissconnect/reconnect ? i have seen online people mentioning how simply dissconnecting/reseting the usb-hdmi physically solves a lot of the problems,software bassed disconnect/reconnect of the usb port itself could sort of mimic that.

thomae, 2023-07-17:
You can try if usbreset from the usbutils package suits your need.

```
rw
pacman -Syy
pacman -S usbutils
ro
usbreset

Usage:
  usbreset PPPP:VVVV - reset by product and vendor id
  usbreset BBB/DDD   - reset by bus and device number
  usbreset "Product" - reset by product name
```

mdevaev, 2023-07-17:
If it will work, you can bind these commands to web ui: https://docs.pikvm.org/gpio/#cmd

mdevaev, 2023-07-21:
Answered, I guess.

closed 2023-07-21 by mdevaev