pikvm/pikvm #1060: Suddenly no signal

Ulyouth, 2023-07-15:
Hello everyone,

I suddenly started getting a "no signal" message on my PiKVM v3 HAT with a "Persistent device timeout (unplugged)" error being displayed on the log. I tried a few things such as updating packages and also the solution proposed at pikvm/ustreamer#110, but nothing worked so far. I was wondering if anyone could shed some light on the matter.

Technical info:

Log:
```
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.server             INFO --- Registered new client session: WsSession(id=2998041840, {'stream': True}); clients now: 1
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Starting streamer ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=24460: /usr/bin/ustreamer --device=/dev/kvmd-video --persistent --dv-timings --format=uyvy --encoder=omx --workers=3 --quality=80 --desired-fps=40 --drop-same-frames=30 --last-as-blank=0 --unix=/run/kvmd/ustreamer.sock --unix-rm --unix-mode=0660 --exit-on-parent-death --process-name-prefix=kvmd/streamer --notify-parent --no-log-colors --sink=kvmd::ustreamer::jpeg --sink-mode=0660 --h264-sink=kvmd::ustreamer::h264 --h264-sink-mode=0660 --h264-bitrate=5000 --h264-gop=30
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      main] -- Starting PiKVM uStreamer 5.41 ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      main] -- Using internal blank placeholder
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      main] -- Using JPEG-sink: kvmd::ustreamer::jpeg
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      main] -- Using H264-sink: kvmd::ustreamer::h264
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504    stream] -- Using V4L2 device: /dev/kvmd-video
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504    stream] -- Using desired FPS: 40
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504      http] -- Starting HTTP eventloop ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.504    stream] -- H264: Initializing encoder ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.513    stream] -- H264: Configuring encoder: DMA=0 ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.533    stream] -- Device fd=11 opened
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.533    stream] -- Using input channel: 0
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.567    stream] -- Using resolution: 640x480
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.567    stream] -- Using format: UYVY
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.567    stream] -- Querying HW FPS changing is not supported
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.567    stream] -- Using IO method: MMAP
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.569    stream] -- Requested 5 device buffers, got 5
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- Capturing started
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- JPEG-0: Initializing encoder ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- JPEG-1: Initializing encoder ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- JPEG-2: Initializing encoder ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- Using JPEG quality: 80%
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- Creating pool JPEG with 3 workers ...
[2023-07-15 12:40:55 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [426554.578    stream] -- Capturing ...
[2023-07-15 12:40:55 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://pikvm/share/js/kvm/stream_janus.js'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0'
[2023-07-15 12:40:55 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://pikvm/share/js/kvm/janus.js'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0'
[2023-07-15 12:40:55 kvmd.service] --- aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0'
[2023-07-15 12:40:55 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => Creating new session: 4358394122180708; 0xb0a00eb8
[2023-07-15 12:40:55 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => Creating new handle in session 4358394122180708: 2667021486317727; 0xb0a00eb8 0xb0a01308
[2023-07-15 12:40:55 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => == ustreamer/main      -- Creating session 0xae601408 ...
[2023-07-15 12:40:55 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [2667021486317727] Creating ICE agent (ICE Full mode, controlling)
[2023-07-15 12:40:55 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => == ustreamer/video     -- Memsink opened; reading frames ...
[2023-07-15 12:40:56 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [426555.579    stream] -- Persistent device timeout (unplugged)
[2023-07-15 12:40:56 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [2667021486317727] Waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
[2023-07-15 12:40:57 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [2667021486317727] Failed to add some remote candidates (added 0, expected 1)
[2023-07-15 12:40:57 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [2667021486317727] The DTLS handshake has been completed
```

 v4l2-ctl -d /dev/video0 --all:
```
Driver Info:
        Driver name      : unicam
        Card type        : unicam
        Bus info         : platform:fe801000.csi
        Driver version   : 5.15.68
        Capabilities     : 0x85a00001
                Video Capture
                Metadata Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Media Driver Info:
[... 112 lines cut ...]
        Bytes per Line    : 1280
        Size Image        : 614400
        Colorspace        : SMPTE 170M
        Transfer Function : Default (maps to Rec. 709)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Limited Range)
        Flags             :

User Controls

            audio_sampling_rate 0x00981980 (int)    : min=0 max=768000 step=1 default=0 value=0 flags=read-only
                  audio_present 0x00981981 (bool)   : default=0 value=0 flags=read-only

Digital Video Controls

                  power_present 0x00a00964 (bitmask): max=0x00000001 default=0x00000000 value=1 flags=read-only
```

uname -a:
`Linux pikvm 5.15.68-3-rpi-ARCH pikvm/ustreamer#1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux`

PCI bridge:
`PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 10)`

Thanks in advance.

mdevaev, 2023-07-15:
Hello! I have passed your case to technical support, we will diagnose and find out what the problem is ;) I'm in touch

arch1mede, 2023-07-15:
Hi, it would be best to join us on [Discord](https://discord.gg/bpmXfz5) to help diagnose. If this isnt possible, please provide the following:

Can you provide details on what happens when you try this on another Target system?
Provide us a pic on how you have this wired up to the target? 
Have you tried another hdmi cables? 
Have you tried connecting this to a real monitor and are you getting a pic from the monitor? If so what is the resolution its sending?

Please try this loop test and see what the results are:
HDMI Loop test Please use the 1ST micro hdmi port closest to the usb-c port for V3, v4+ use the output1 into the in, v4 mini in incapable of doing a loop test

[image]

Ulyouth, 2023-07-16:
> Have you tried another hdmi cables? Have you tried connecting this to a real monitor and are you getting a pic from the monitor? If so what is the resolution its sending?

I tried with another hdmi cable but still no image being displayed. When I plug it to a real monitor everything works fine. I can even use keyboard and mouse from KVM web, just no image output. The resolution was 1920x1080@60Hz, I tried other resolutions/refresh rates but it still won't work.

> Please try this loop test and see what the results are: HDMI Loop test Please use the 1ST micro hdmi port closest to the usb-c port for V3, v4+ use the output1 into the in, v4 mini in incapable of doing a loop test

Unfortunately I don't have such a cable.

> Can you provide details on what happens when you try this on another Target system? Provide us a pic on how you have this wired up to the target? 

I could but this will be a bit annoying as I have to disassemble the setup, move to another place and make up some free space on the other target system for the extra cables and peripherals. I will try doing a clean install first sometime this week and see if it works (I already wanted to do that anyways). If it still doesn't work I will try it on another system and contact you on discord.

arch1mede, 2023-07-16:
> I tried with another hdmi cable but still no image being displayed. When I plug it to a real monitor everything works fine. I can even use keyboard and mouse from KVM web, just no image output. The resolution was 1920x1080@60Hz, I tried other resolutions/refresh rates but it still won't work.
> 
for the v3 its important that you are using 1080p@50hz or 720p@60hz

> Unfortunately I don't have such a cable.

I would recommend getting one just in case, its good to have.

> I could but this will be a bit annoying as I have to disassemble the setup, move to another place and make up some free space on the other target system for the extra cables and peripherals. I will [...]

Yes I agree but it beats the alternative, being a defective hat. I don't think you need to do a clean install, just try this on anything really, reboot the target as soon as the pikvm is up.

Ulyouth, 2023-07-18:
> for the v3 its important that you are using 1080p@50hz or 720p@60hz

Tried also, to no avail.

> I would recommend getting one just in case, its good to have.

Ok, I'll get one.

> Yes I agree but it beats the alternative, being a defective hat. I don't think you need to do a clean install, just try this on anything really, reboot the target as soon as the pikvm is up.

Yeah I agree, it would be important to document the issue, so I'll reformat my KVM later. I plugged the KVM to another system today and exactly the same thing happens - keyboard/mouse fine, but no signal. 

I'll contact you guys on discord to continue investigating.

Ulyouth, 2023-08-03:
Ok, I finally got the HDMI adapter to do the loop test. I'm not sure how this is done but I suppose I just plug both ends of the cable to the device and it should show a standard image or video at the KVM page. If thats right, it still doesn't work and I keep getting the usual "No signal" message. Here is the log:

```
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Starting streamer ...
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=1069: /usr/bin/ustreamer --device=/dev/kvmd-video --persistent --dv-timings --format=uyvy --encoder=omx --workers=3 --quality=80 --desired-fps=40 --drop-same-frames=30 --last-as-blank=0 --unix=/run/kvmd/ustreamer.sock --unix-rm --unix-mode=0660 --exit-on-parent-death --process-name-prefix=kvmd/streamer --notify-parent --no-log-colors --sink=kvmd::ustreamer::jpeg --sink-mode=0660 --h264-sink=kvmd::ustreamer::h264 --h264-sink-mode=0660 --h264-bitrate=5000 --h264-gop=30
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.137      main] -- Starting PiKVM uStreamer 5.41 ...
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.137      main] -- Using internal blank placeholder
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.137      main] -- Using JPEG-sink: kvmd::ustreamer::jpeg
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.137      main] -- Using H264-sink: kvmd::ustreamer::h264
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.138      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.138    stream] -- Using V4L2 device: /dev/kvmd-video
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.138    stream] -- Using desired FPS: 40
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.138      http] -- Starting HTTP eventloop ...
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.138    stream] -- H264: Initializing encoder ...
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.145    stream] -- H264: Configuring encoder: DMA=0 ...
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => ================================================================================
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [887.168    stream] -- Can't access device: No such file or directory
[2023-08-03 21:53:14 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [887.168    stream] -- Waiting for the device access ...
```

mdevaev, 2023-08-04:
Please show the output of:
```
dmesg | grep tc3
pacman -Q | grep kvmd
```

Ulyouth, 2023-08-04:
> Please show the output of:
> 
> ```
> dmesg | grep tc3
> pacman -Q | grep kvmd
> ```

dmesg | grep tc3
```
[   10.727977] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
[   10.741497] tc358743 10-000f: not a TC358743 on address 0x1e
```

pacman -Q | grep kvmd
```
kvmd 3.234-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.234-1
kvmd-webterm 0.43-1
```

arch1mede, 2023-08-05:
Hi, you made mention of a hat, did you put this together yourself? with the error you are showing, I would check the CSI cable, its the small white FFC, reset or replace with the extra that was included.

Ulyouth, 2023-08-08:
> Hi, you made mention of a hat, did you put this together yourself? with the error you are showing, I would check the CSI cable, its the small white FFC, reset or replace with the extra that was included.

That worked! The problem was indeed the CSI cable. I replaced it with the extra one and video capture started working again. Thanks a lot for the support!

Aside from that, any possible reasons why would the CSI cable suddenly stop working?

mdevaev, 2023-08-08:
This may be due to oxidation of the contacts, if the room is sometimes humid. I think if you clean them on the old cable, everything will work too. Or was it poorly inserted initially. Or if you often disassembled the device, the cable could wear out. Maybe you just had bad luck, it was just low-quality. In any case, remember an important saying: if it works, don't touch it :D I think everything will be fine now.

closed 2023-08-08 by mdevaev