pikvm/pikvm #1062: RFE for docs, v4, DIP switches: please explain when safe to toggle and what ABP is

pcfe, 2023-07-16:
### Description

At https://docs.pikvm.org/v4/#dip-switches I would like to also find the following questions answered;

- can the DIP switches be adjusted while the PiKVM is powered up? I guess the answer is _no_ but it would be nice to be sure.
- What exactly does _advanced backpower protection_ do? Is it perrchance the hardware fix for earlier versions that is discussed in  #382 ?

### Solution I'd like

Nothing fancy needed, just a word or three on when the switches are safe to adjust and a link to a page giving more details on advanced backpower protection

mdevaev, 2023-07-16:
Hi! Added some clarifications. Thank you!

closed 2023-07-16 by mdevaev

pcfe, 2023-07-16:
Thank you, 9398a5b6cbf2caacedf09c76c13f2c182c046871 is exactly what I was looking for. :-)