pikvm/pikvm #1064: ATX Control is not working if made using relay

justicecurcian, 2023-07-19:
**Describe the bug**
I am using v2 made with RPI 4. I DIYed an ATX Control using cheap relay that is being contolled by 3.3V ( https://aliexpress.ru/item/1005001994042589.html?spm=a2g2w.orderdetail.0.0.14cc4aa6jq5hm2&sku_id=12000018351924068 ), but it's not working. I manually triggered relay using 3.3V from RPI and it closed, but when I click power/reset in the web control panel the relay won't close. I am using correct pins (23 for power and 27 for reset), I tried creating my own button using gpio plugin, but had no luck setting it up correctly. I am using only two relays for buttons and nothing for power light, so this might be an issue, but I found nothing about mandatory connection of power led in the docs. 

**To Reproduce**
Click power/reset in ATX menu

**Expected behavior**
Relay closed and PC power/reset button "clicked"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows / Mac OS
 - Browser: Chrome
 - Version: No idea how to find it

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.173
  - uStreamer version: 5.31-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-07-20:
What do you mean, did you perform the direct connection of relay coil to gpio? Without any transistors?

justicecurcian, 2023-07-20:
>What do you mean
If something particular from my message is hard to understand, please quote it and I will try to rephrase it.

>Did you perform the direct connection of relay coil to gpio? Without any transistors?
Yes, direct connection without any transistors, since RPI 4 gpio is 3.3V it should be able to control relay directly. 3.3V power pin from RPI in fact can close relay easily. If I connect LED to pins 23 and GROUND it won't highlight too, so I think it's either a software problem or I am wiring it wrong. When I click on "Click Power" in ATX control is it supposed to emit 3.3V or ground pin 23?

mdevaev, 2023-07-20:
> Yes, direct connection without any transistors, since RPI 4 gpio is 3.3V it should be able to control relay directly.

Don't ever do that! You can burn your Raspberry. The coil consumes a lot of power for operation, and when de-energized generates a reverse polarity current. I recommend you to buy a relay module that has a separate power input and a control wire. But it is better if you follow our scheme.

justicecurcian, 2023-07-20:
>You can burn your Raspberry.

Oh, that's not good. I will use a separate module later, thank you.  Maybe it's good it didn't work, since I could kill my kvm.
But the problem is still presented, LED is not highlighting, 23 pin is not getting power. I didn't follow docs scheme because listed parts are unavailable here.

justicecurcian, 2023-07-20:
On second thought, maybe relay did kill the pin, so this is why led is not highlighing...

mdevaev, 2023-07-20:
Yes, maybe that's the reason.

mdevaev, 2023-07-20:
You can use other pins, but before that, find a good relay module.

justicecurcian, 2023-07-20:
How do I change pins? I thought they are hardcoded.

justicecurcian, 2023-07-20:
I flashed raspbian and pulled up pin 23 using raspi-gpio and it flashed LED, I've got back to pikvm and ATX control works now (without relay, dummy led). Really weird, sorry for useless issue and big thanks for telling me about relays, you saved my PI!

closed 2023-07-20 by justicecurcian

mdevaev, 2023-07-21:
No problem :)