pikvm/pikvm #1069: Provide option to disable authentication on Prometheus endpoint

onedr0p, 2023-07-22:
**Is your feature request related to a problem? Please describe.**

I would like to configure prometheus exporter without a password since my pikvm is internal only.

**Describe the solution you'd like**

Provide option to disable authentication on Prometheus endpoint

**Describe alternatives you've considered**

I am working around this now by doing the following, however I think this needs to be done on every update

```sh
rw
nano /usr/lib/python3.11/site-packages/kvmd/apps/kvmd/api/export.py
# @exposed_http("GET", "/export/prometheus/metrics", False)
ro
systemctl restart kvmd.service
```

**Additional context**
Add any other context or screenshots about the feature request here.

onedr0p, 2023-08-07:
Maybe it would be a good idea to do a network acl instead of no auth?

```yaml
prometheus:
  basic_auth: false
  allowed_networks:
    - 192.168.0.0/24
```

mdevaev, 2023-08-17:
Sup. I made two things in 3.246:
* Option to disable auth:
    ```yaml
    kvmd:
        prometheus:
            auth:
                enabled: false
    ```
* Five-second caching of export results, so it will not be possible to provoke DoS.

Is it enough?

onedr0p, 2023-08-17:
That's perfect, thanks!

mdevaev, 2023-08-17:
:ok_hand:

closed 2023-08-17 by mdevaev