pikvm/pikvm #1070: Two pikvms for two displays for same pc - Can't use both in relative mouse mode

plia7, 2023-07-22:
**Describe the bug**
I have two pikvms which give me two displays, moving the mouse from first screen to second screen only works if I enter first pikvm webpage in a relative mouse mode set - then the mouse seamlessly moves to the second screen - relative mouse mode movement is quite slow compared to absolute mouse mode - How can I configure it so both displays are in relative mouse mode and can work with each other?

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to pikvm 1 web page - the absolute mouse mode defaults
2. Go to pikvm 2 web page - the absolute mouse mode defaults
3. Try to move the mouse pikvm 1 web page far enough so it could reach the edge of the screen.

**Actual behavior**
Mouse cursor remains at the edge of the first screen and there is no indication on the second screen that the mouse cursor is there.

**Expected behavior**
Mouse cursor ends up showing up on the second screen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 11
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
 PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.236-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.236-1
kvmd-webterm 0.43-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  ustreamer 5.41-1
  - Linux kernel: `uname -a`
 Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

plia7, 2023-07-27:
@mdevaev Just for reference, you mentioned that "this is a complex issue, but I plan to solve it in this year".

Some research considerations: 
"have some special feature flag or setting/mode, that links two pikvms together, so they talk to each other, i.e.: pikvm1 to pikvm2: user brought mouse to left edge of screen 1. pikvm2 to pikvm1: ok let me take the mouse from here putting it in right edge of screen 2. " - to which you replied "yes, maybe".

https://www.hardwarecentric.com/how-to-use-2-monitors-2-keyboards-2-mice-1-computer-at-the-same-time/

https://www.youtube.com/watch?v=BzX4u6nPWTY

https://mcmw.abilitynet.org.uk/how-to-use-your-keyboard-to-control-the-mouse-pointer-in-windows-11

Thank you.

plia7, 2023-10-01:
@mdevaev It's already October, just checking to see it's still in the planning, as you said you plan to solve it in this year.
Thank you.

mdevaev, 2023-10-03:
It's not ready yet.

plia7, 2023-10-04:
> It's not ready yet.

Thank you, this could be an amazing feature and stimulate other potential customers to buy additional pikvms for added productivity (a second monitor).

mdevaev, 2023-10-26:
USB devices are not tied to the screen and have a single input field. In any case, I'm closing this since we already have [another](https://github.com/pikvm/pikvm/issues/1071) issue about it, and within the described, everything works as it should.

plia7, 2023-11-30:
> Edited: wrong comment

Hello @mdevaev, is this feature still feasible by EOY (any progress)? Would be nice if Santa includes this feature :)
Thank you.

mdevaev, 2023-12-01:
Yes, it is. I remember the task and do it whenever possible.

Vladimir2989, 2024-02-01:
Hi, any news on this ? its been a while.
I'm still good for 100$ bounty I've offered 💰

plia7, 2024-03-11:
@mdevaev As a short term solution (until we figure out the change for the long term solution), maybe we can add this as an experimental option (i.e. Absolute Mode X Axis Offset) in the pikvm ui (0 by default), where if it's set to positive/negative offset integer number (I believe both positive and negative are needed as the extended display can be to the left of the primary screen or to the right of the primary screen which will affect the the X Axis value), it will offset the x axis absolute mouse coordinates it sends to the host, so people who use the second pikvm for an extended monitor will be able to use the absolute mouse on the extended monitor? This is under the assumption the Y coordinates the second pikvm extended desktop sent to the host is correct (which it does seem to be the case when currently testing it). Just an idea...

So if let's say the absolute mouse mode currently assumes only 1 primary display for the host with hd resolution, absolute mouse sends all coordinates:
From (0,0) to (1920,1080)
This is true for both pikvm 1 and pikvm 2

Now let's say 
If the extended display from the second pikvm to the same host is from the left and the coordinates it has are:
(-1920,0) to (0,0)

When you hover with your mouse in absolute mode to the top left corner of the second pikvm, it sends the coordinate:
(0,0) which causes the mouse to appear on the first pikvm screen top left corner.
If you set the new offset field to -1920, it will send this instead:
(-1920,0) which will make the mouse to appear on the top left corner of the second pikvm extended screen.

So the suggested change is:

Send absolute mouse coordinates
(X + %Absolute Mode X Axis Offset%), Y

For those who don't use the offset, it will just always send (X+0, Y) (or you could just not add it if it's 0)
For those who do use it, like in the example above, it will send that offset configured, so: (X + (-1920)), Y

Does that make sense?

Thank you.

mdevaev, 2024-03-31:
@Vladimir2989 I don't have the opportunity to do this yet :(

lodibrahim, 2024-11-15:
I am facing same issue, it will be cool to use two monitors

mannyking0711, 2025-02-21:
Waiting for #1017 to be resolved .

tenroseUK, 2025-02-25:
This is the dream feature for me. I've got two PiKVMs set up in the same config as the OP, except my experience goes like this.

PiKVM 1 is connected to HDMI and USB, in browser window 1
PiKVM 2 is connected to HDMI and USB, in browser window 2

With both set to Absolute Mouse Mode:
Moving on browser 1 performs as expected
Moving on browser 2 moves the cursor on both browser 1 and browser 2

With both set to Relative Mouse Mode:
Able to control both browsers from either one while control is captured

As the OP said, Relative Mouse Mode is much slower than Absolute. Would be nice to see this feature prioritised if possible?

plia7, 2025-04-07:
Hello @mdevaev, just following up on this, it's been almost 2 years now.
When are you estimating for this feature to be complete? Will it also work if one of the screens have a different orientation (i.e. left screen is set to 90 degrees)?

Thank you.

gnampf1, 2026-05-01:
It goes much further. On Windows a mouse with absolut coordinates is always completely mapped to the main display. No matter which coordinates you give, they are on the main, you can't reach the other screens, as long as you use the builtin driver of MS: So you would need a special driver first to even have a chance to reach the other displays.
You can get around this by not using an absolute mouse, but a pencil or touchpad. But: this behaves different than a mouse does. It doesn't know left/right button, but just one. It reacts different on dragging, scrolling, etc. So it's not what you would expect if you use a mouse. Gets more confusing if you try to send buttons and scrollwheel via a second mouse-item to the machine. Then it depends on the application how it reacts, some react as expected, others ignore scrolling, and on yet others the scrolling happens on a completely differnt application on the other screen. On Thunderbird for example you can't use the scroll wheel but have to drag the text to scroll, like with a touchscreen.

So I guess first thing would be finding a suitable mouse driver for Windows. Writing an own one wouldn't be easy, as it needs to be signed to be trusted.