pikvm/pikvm #1071: Two pikvms for two displays for same pc - VNC to second pikvm ignores mouse/keyboard input

plia7, 2023-07-22:
**Describe the bug**
I have two pikvms which gives me two displays. If I VNC to second pikvm it ignores mouse/keyboard input.

**Prerequisites:** Connect pikvm 1 to first hdmi port in pc 1 and usb port. Connect pikvm 2 to second hdmi port in pc 1.
**Note:** I didn't connect pikvm 2 to a second usb in pc 1, but if you do connect it, it doesn't seem to fix the issue.

**To Reproduce**
Steps to reproduce the behavior, like:
1. VNC to pikvm 2.

**Actual behavior**
You can see the display but mouse and keyboard input is being ignored.

**Expected behavior**
Mouse and keyboard input work just like they do when you vnc to pikvm 1.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 11
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)
Windows TigerVNC Viewer v1.13.1
or
iOS Jump Desktop v9.7.8

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
 PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.236-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.236-1
kvmd-webterm 0.43-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  ustreamer 5.41-1
  - Linux kernel: `uname -a`
 Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

plia7, 2023-07-27:
This is related to https://github.com/pikvm/pikvm/issues/1070

mdevaev, 2023-10-26:
USB devices are not tied to the screen and have a single input field. In any case, I'm closing this since we already have https://github.com/pikvm/pikvm/issues/1070 issue about it, and within the described, everything works as it should.

closed 2023-10-26 by mdevaev