pikvm/pikvm #1072: No signal with pikvm v3

Bluebugs, 2023-07-23:
**Describe the bug**
I am getting no signal on a PiKVM v3. `systemctl status kvmd` report an error:
```
pikvm kvmd[581]: kvmd.apps.kvmd.streamer INFO --- => -- ERROR [112.514 stream] -- Can't access device: No such file or directory
```
Additionally the kernel report the following error:
```
# dmesg|grep fail
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    9.927636] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
[   10.085006] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
```

I have done a full system update to make sure this wasn't a software issue of some sort, but the behavior is the same before and after the update.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Browser Firefox
 - Version 115.0.2

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - PiKVM platform v3
  - Video capture type HDMI HAT
  - KVMD version: 3.236-1
  - uStreamer version: 5.41-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

Not to sure what to look at now. Thanks for any suggestion or idea.

arch1mede, 2023-07-23:
Hi, I would ask if you could join us on [Discord](https://discord.gg/bpmXfz5) to troubleshoot faster but if you cannot, is this a v3 that you have assembled?
```
[    9.927636] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
```
This line here tells me that there is possibily a FCC cable problem. I would reseat or replace this with the replacement FCC.

Bluebugs, 2023-07-23:
I did reseat the cable on both end and the problem went away. I have not looked at the code, but maybe instead of just displaying `no signal`, something saying `no signal, FCC cable probably disconnected` would have been useful in this situation. Thanks.

closed 2023-07-23 by Bluebugs

mdevaev, 2023-07-23:
I plan to do this, but now there are many other tasks.

evanthomas, 2024-06-18:
Perhaps add it to the troubleshooting FAQ. Might have saved me a couple of hours.

mdevaev, 2024-06-18:
@arch1mede could you add this to the faq?