pikvm/pikvm #1075: Mobile/Desktop web version issues from tablet/touch input capable device

plia7, 2023-07-28:
**Describe the bug**
When opening pikvm from chrome browser in Samsung Galaxy s6 lite
There are some issues in the mobile and desktop web versions.

**To Reproduce**
Steps to reproduce the behavior, like:
In mobile web version:
1. Can't go full screen - buttons for maximize window, minimize window or go full screen are missing.
In mobile/desktop web versions: 
2. Can't use touch gestures without using mouse ui buttons in mobile version which could be a problem if issue 1 is resolved (go full screen), because it seems like even in the desktop version, going full screen is ignoring any touch input as well or when moving mouse cursor in desktop version, seems to be offsetting the host mouse cursor away 15 pixels away from where the touch input movement is being made in the desktop version.
3. Scrolling doesn't seem to work using two fingers to scroll (same gesture is working for me right now when doing this gesture in jump desktop).

**Expected behavior**
Being able to go full screen in both mobile/desktop versions. Being able to use touch input movement to accurately make the host mouse cursor to follow (touchpad input style). Being able to do touch input single tap for left click or double tap for a right click (and/or other nice touch gestures if possible, see jump desktop/bvnc for examples). Being able to scroll up or down with two fingers (similar to how it's working for me right now when doing this gesture in jump desktop).

**Screenshots**
If applicable, add screenshots to help explain your problem.

Desktop (please complete the following information):

OS: [e.g. iOS]
Android
Browser [e.g. chrome, safari]
Google Chrome app
Version [e.g. 22]
VNC client (if used)
Tablet:
Samsung Galaxy s6 lite

PiKVM info:
Raspberry Pi board version [e.g. RPi 4]
PiKVM v3 Pre-Assembled
PiKVM platform [e.g. v2-hdmi]
Video capture type [e.g. CSI bridge]
KVMD version: pacman -Q | grep kvmd
kvmd 3.236-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.236-1
kvmd-webterm 0.43-1
uStreamer version: pacman -Q | grep ustreamer
ustreamer 5.41-1
Linux kernel: uname -a
Linux pikvm 5.15.68-3-rpi-ARCH https://github.com/pikvm/pikvm/issues/1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Using mouse ui buttons makes it a clumsy user experience, especially when the host cursor is not accurately following touch movements or the inability to go full screen and/or to use these mouse ui buttons.

plia7, 2023-07-28:
@mdevaev Just for reference, you mentioned on 07/20/2023 that "I don't have time to deal with mobile devices right now... I plan to deal with everything in the fall".

Thank you.

plia7, 2023-10-01:
@mdevaev It's already October, just checking to see it's still in the planning, as you said you plan to solve it in this fall.
Thank you.

mdevaev, 2023-10-02:
We're working :)

DigitalYemeni, 2023-12-01:
Hi,
Any update or ETA for this feature?

Thank you so much!

mdevaev, 2023-12-01:
No ETA, sorry

plia7, 2024-03-21:
> We're working :)

Hello,

Is this the right ticket to track the this work? I just don't see that you assigned it to yourself.

Thank you.

zowle, 2025-02-12:
Hi, is there an update? Would be nice to be able to go Fullscreen/hide UI on an Android browser.

mdevaev, 2025-05-11:
In the next release two fingers scroll will be available with improved mouse scroll algorithm.

mdevaev, 2025-11-26:
In the next release I'll give you floating box with mouse buttons, fast keyboard access and also full-screen mode with toolbar. I even works on the mobile interface. At the moment, I don't see a better solution to prevent user touch misinterpretation.

By source request:

- `full-screen support` - done, full screen and full tab for mobile UI. Navbar panel is also accessible, now you don't need to exit from full* mode to get it.
- `two-finger scrolling` - already done.
- ~~`tap-to-click gestures`~~ - will not be implemented, because in the case of an absolute mouse, it is impossible to distinguish a click from cursor positioning. As an alternative, I made a floating panel with mouse buttons. Maybe in the future I'll make something like a floating pointer, like on Jump VNC.
- ~~`precise touch-to-cursor mapping`~~ - My experience using the new UI on a iPad shows that this is not necessary because of the previous point: all touches are treated as positioning.
- ~~`pinch-to-zoom`~~ - This is necessary for phones, but our interface is primarily aimed at tablets, where it does not seem to be required due to the fact that the screen is large enough.

[image]