pikvm/pikvm #1076: Windows 11 Display orientation Portrait is not working

plia7, 2023-07-28:
**Describe the bug**
Windows 11 Display orientation Portrait is not working. It sometimes useful to rotate a monitor that is capable of portrait orientation to see more screen real estate for many lines of text.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Rotate a monitor that is capable of portrait orientation (it will automatically change to portrait orientation in Window). Note: you can choose for the display resolution 1080x1920.
2. Login into your pikvm wb ui and go into full screen.
3. Go into Windows display settings in the host pc.
4. For the pikvm display, change the display orientation from landscape to portrait. Note: you can choose for the display resolution 1080x1920.

**Expected behavior**
Displays fills in full in portrait mode. Mouse movement should be able to navigate through the entire portrait display.

**Actual behavior**
Displays rotates 90 degrees and not filling the entire black space. The mouse movement also doesn't seem to be right.

**Screenshots**
Initial state in landscape:

[image]

After changing to portrait:

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 114.0.5735.199 (Official Build) (64-bit)
or
Microsoft Edge Version 115.0.1901.183 (Official build) (64-bit)
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
 PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.236-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.238-1
kvmd-webterm 0.43-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  ustreamer 5.41-1
  - Linux kernel: `uname -a`
 Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
It sometimes useful to rotate a monitor that is capable of portrait orientation to see more screen real estate for many lines of text.

mdevaev, 2023-08-06:
Do you REALLY need this? Orientation support is not part of HDMI, it is purely a soft feature and you need to flip the monitor physically for this.

plia7, 2023-08-07:
> Do you REALLY need this? Orientation support is not part of HDMI, it is purely a soft feature and you need to flip the monitor physically for this.

Hi @mdevaev, yes please this is super useful. I have $300 lg view 16+ gram portable monitor and there is great benefit with portrait orientation when it comes to screen real estate (such as academic article reading, seeing more programming lines, etc). I compared it with some other machine streamer software based on h264 and it works great if you flip the monitor and/or it has a right mouse click option on the stream which will flip it and fill in the entire screen.

Thank you.

mdevaev, 2023-08-10:
It didn't work out quickly, there were some problems with video scaling. A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry

BhupiMAD, 2023-08-20:
Any fix for this? i tried your CSS rotate solution, but mouse coordinates does not rotate with screen,

plia7, 2023-08-20:
> Any fix for this? i tried your CSS rotate solution, but mouse coordinates does not rotate with screen,

I don't think @mdevaev solved it yet, it's still work in progress, I think he's still busy with other priorities.

plia7, 2023-10-01:
> It didn't work out quickly, there were some problems with video scaling. A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry

Hi @mdevaev, just checking to see if there are any updates on this? Your last action was on Aug 10.
Thank you.

CC: @BhupiMAD

BhupiMAD, 2023-10-01:
> > It didn't work out quickly, there were some problems with video scaling. A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry
> 
> Hi @mdevaev, just checking to see if there are any updates on this? Your last action was on Aug 10. Thank you.
> 
> CC: @BhupiMAD

This is what @mdevaev replied
You can invert the mouse range. Run kvmd -m and find options tree with mouse_x_range and mouse_y_range.

plia7, 2023-10-02:
> This is what @mdevaev replied
>
> You can invert the mouse range. Run kvmd -m and find options tree with mouse_x_range and mouse_y_range.

So you got it to work? It's not just the mouse movement it's also the screen that's not correctly oriented

BhupiMAD, 2023-10-02:
For me screen orientation was working, the problem was with the mouse not inverting as per screen. I haven’t tried the mouse solution yet.

plia7, 2023-10-02:
> For me screen orientation was working, the problem was with the mouse not inverting as per screen. I haven’t tried the mouse solution yet. 

Not sure what are you saying but there is an issue: Display rotates 90 degrees and not filling the entire black space. The mouse movement also doesn't seem to be right.
So I guess we'll have to wait for it to be resolved, it looks like the dev started to work on it but hasn't finished yet.

mdevaev, 2023-10-02:
No progress here, sorry.

spcharc, 2023-10-02:
I like this issue. Some webpages and e-books / pdfs are more suitable on a vertical display than horizontal.

> A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry

This may fix the web version. Many people use VNC which may also need some other adjustments

plia7, 2023-10-02:
> I like this issue. Some webpages and e-books / pdfs are more suitable on a vertical display than horizontal.
> 
> 
> 
> > A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry
> 
> 
> 
> This may fix the web version. Many people use VNC which may need some other adjustments

Thank you. I think the web version is more functional and common and where you could place the button to activate this mode. But I'm all for adding the support to VNC as well, should probably go in a separate ticket with a link to this ticket.

mdevaev, 2023-10-03:
I can't add it to VNC because there is now the protocol doesn't have any rotation feature and uStreamer can't perform it at all.

spcharc, 2023-10-04:
> I can't add it to VNC because there is now the protocol doesn't have any rotation feature and uStreamer can't perform it at all.

sad

I checked tigervnc client and it cannot rotate images. Guess there's no way to add this to vnc without some major software changes.

plia7, 2023-11-30:
> I can't add it to VNC because there is now the protocol doesn't have any rotation feature and uStreamer can't perform it at all.

Hello, what about adding it to PiKVM UI, using css transform and swap x/y/width/height in getGeometry - Any progress on that or did you hit a plateau?
Thank you.

plia7, 2024-03-11:
> It didn't work out quickly, there were some problems with video scaling. A note for myself not to forget: use css transform and swap x/y/width/height in getGeometry

@mdevaev To use CSS transform and swap the x, y, width, and height properties in getBoundingClientRect() or any similar method, you'll need to follow these steps:

1. Apply CSS Transform: Apply the necessary CSS transform to the element whose geometry you want to manipulate. This can be done using the transform property in CSS.

2. Get Transformed Geometry: After applying the transform, you need to get the transformed geometry of the element. You can do this by using the getBoundingClientRect() method in JavaScript.

3. Swap Properties: Once you have the transformed geometry, swap the x and y coordinates, as well as the width and height properties, to get the desired result.

Here's a basic example:

HTML:

`<div id="element"></div>`

CSS:

```
#element {
  width: 100px;
  height: 100px;
  background-color: red;
  transform: rotate(45deg);
}
```

JavaScript:

```
// Get the element
const element = document.getElementById('element');

// Get the transformed geometry
const rect = element.getBoundingClientRect();

// Swap x and y coordinates
const swappedX = rect.y;
const swappedY = rect.x;

// Swap width and height properties
const swappedWidth = rect.height;
const swappedHeight = rect.width;

// Output the swapped geometry
console.log("Swapped X:", swappedX);
console.log("Swapped Y:", swappedY);
console.log("Swapped Width:", swappedWidth);
console.log("Swapped Height:", swappedHeight);
```

In this example, we apply a rotate(45deg) transform to the element. Then, we use getBoundingClientRect() to get the transformed geometry. Finally, we swap the x and y coordinates as well as the width and height properties to get the geometry as if it were not transformed.

Hopefully this helps.

Thanks.

plia7, 2024-03-15:
> I don't need GPT's help, I've already tried a similar approach and the problem is scaling, as I wrote.

You mean aspect ratio?
To ensure that the video scales properly within the portrait dimensions while maintaining its aspect ratio, you can set the `object-fit` property for the video element. Here's how you can modify the CSS:

```css
#video {
  position: absolute;
  width: 100%; /* Ensure the video fills the container horizontally */
  height: 100%; /* Ensure the video fills the container vertically */
  object-fit: cover; /* Maintain aspect ratio and cover the container */
}
```

By setting `object-fit: cover;`, the video will be scaled to maintain its aspect ratio while covering the entire area of the container. This ensures that the video fills the portrait dimensions completely without distorting or cropping the video content.

With this adjustment, the video will scale properly within the portrait dimensions while maintaining its aspect ratio.

Transform technique:

To transform a video stream from landscape (1920x1080) to portrait (1080x1920) while ensuring that the video scales properly, you can use CSS transforms along with appropriate scaling techniques. Here's a basic approach using CSS:

HTML:
```html
<div id="videoContainer">
  <video id="video" src="your_video_source.mp4" autoplay loop></video>
</div>
```

CSS:
```css
#videoContainer {
  width: 1080px; /* portrait width */
  height: 1920px; /* portrait height */
  overflow: hidden;
  position: relative;
  transform: rotate(90deg) translateY(-100%);
  transform-origin: top left;
}

#video {
  position: absolute;
  width: auto;
  height: 100%;
}
```

In this example:

1. The `#videoContainer` element is styled to have the dimensions of a portrait orientation (1080x1920). 
2. The `overflow: hidden;` property ensures that any overflow content (which may occur due to the rotation) is hidden.
3. We apply a `rotate(90deg)` transform to the video container to rotate the video by 90 degrees clockwise, effectively changing it from landscape to portrait. 
4. The `translateY(-100%)` transform is used to reposition the rotated video to the top-left corner of the container.
5. `transform-origin: top left;` ensures that the rotation and translation are applied from the top-left corner of the container.
6. The `#video` element inside the container is positioned absolutely to fill the container's height and maintain proper aspect ratio.

With this setup, the video will be transformed from landscape to portrait and will scale appropriately to fit within the portrait dimensions while maintaining its aspect ratio. Adjust the dimensions in CSS according to your specific requirements and make sure to replace `"your_video_source.mp4"` with the actual video source.

Hopefully this helps.
If not, could you please show an example/screenshot when you say the video doesn't scale? I'll see if I can help but I want to ensure I understand where the difficulty is.

Thank you.

mdevaev, 2024-03-15:
If you want to figure out the problem on your own, then I'm not stopping you. But please don't ask ChatGPT questions and then ask me to check it all out. ChatGPT does not know the context and does not see the code, and offers only common advice. This will not help solve the problem.

[image]

RSATom, 2024-03-15:
Just an idea, maybe it can be done different way? - If I understand correctly RTP video frames can contain information about video orientation and WebRTC client (browser in our case) can take it into account. So maybe there is a way to set that video orientation info on Janus side?
I can look at it, but unfortunately have  some other tasks with higher priority right now, so it will be  some time later...

mdevaev, 2024-03-15:
@RSATom I like the idea about RTP. I also have a higher priority issue right now, so let me know when you have time, maybe we can do something together.

plia7, 2024-03-15:
> Just an idea, maybe it can be done different way? - If I understand correctly RTP video frames can contain information about video orientation and WebRTC client (browser in our case) can take it into account. So maybe there is a way to set that video orientation info on Janus side?

ChatGPT agrees with you too 🤣:

Yes, you're absolutely correct! If the video frames contain orientation information, the WebRTC client (browser) can utilize this information to properly render the video without the need for manual transformations on the client side.

In the context of Janus, you can indeed set video orientation information on the server side. Janus itself doesn't directly manipulate video frames at this level, but you can use external tools or libraries to handle video processing and orientation metadata.

Here's a potential approach:

1. **Video Processing**: Use a video processing library or tool to analyze and modify the video frames to include orientation metadata. You can extract metadata from the original video, modify it to reflect the desired orientation (e.g., 90-degree rotation for portrait mode), and then repackage the video frames with the updated metadata.

2. **Janus Integration**: Once the video frames contain orientation metadata, Janus can stream them as usual. Janus doesn't need to be aware of the specific orientation metadata; it simply forwards the video frames to the WebRTC clients.

3. **WebRTC Client**: The WebRTC client (browser) receives the video stream with orientation metadata. Modern browsers, such as Chrome and Firefox, have built-in support for processing this metadata and rendering the video correctly according to the specified orientation.

By setting the orientation metadata at the server side (before streaming through Janus), you can ensure that the video is rendered properly in portrait mode without requiring additional transformations or adjustments on the client side. This approach can lead to a more seamless and efficient viewing experience.

Thanks.

mdevaev, 2024-03-15:
Can you stop posting ChatGPT here? I have a strong desire to close this issue, because since you are so good at communicating with ChatGPT, then let it write the code for you. It is unlikely that you need our help in this case.

plia7, 2024-03-15:
> Can you stop posting ChatGPT here? I have a strong desire to close this issue, because since you are so good at communicating with ChatGPT, then let it write the code for you. It is unlikely that you need our help in this case.

Sorry, I'm done with it.
Just one thing, for this new web ui portrait flip switch, is it possible to expose it to the api also so I can add a shortcut trigger when my screen flips to portrait mode, it automatically sends a pikvm api call to flip the orientation? (say mode parameter value 0 is landscape and 1 is portrait). I won't bother anymore :)

Thank you.

plia7, 2024-03-15:
> We can't detect screen orientation automatically.

Sorry, I meant to say my own connecting pc will have a program that can detect the screen orientation change and will send to the pikvm api accordingly.

Thanks.

mdevaev, 2024-03-17:
It's done, but since this is a client-side feature, we have no API on the server side to change orientation.

[image]

closed 2024-03-17 by mdevaev

plia7, 2024-03-17:
> It's done, but since this is a client-side feature, we have no API on the server side to change orientation.
> 
Thank you. Given that it's done as a client-side feature, is it possible to handle it in window orientationchange so when you detect a change there it you pass it (you can get the info from screen.orientation.type) to the streamer?
I tried to register it in stream.js as:

window.addEventListener("orientationchange", function() {
			var orientationType = screen.orientation.type;
			// Code to handle orientation change goes here
			if (__streamer.getMode() === "janus") { // Right now it's working only for H.264
				tools.storage.setInt("stream.orient", 90);
				if (__streamer.getOrientation() != 90) {
					__resetStream();
				}
			}
}

But it's not being fired. Do you know why?
I do see that you register this event already inside wm.js and it it being fired - But I'm not sure how to access the streamer object from the windows manager class? Maybe it can be consolidated somehow?

Last but not least, can I also access/include/add keypad object in the event so I can pass a keyboard shortcut combo from this event so I can notify the host about the change too?

Thanks again.

mdevaev, 2024-03-17:
I don't know why this might not work. And for my taste, this is not so necessary because it is always easy to press the button to explicitly specify the orientation. Especially considering that mobile devices support is limited.

plia7, 2024-03-17:
> I don't know why this might not work. And for my taste, this is not so necessary because it is always easy to press the button to explicitly specify the orientation. Especially considering that mobile devices support is limited.

I'll appreciate if you could try it yourself and see if it fires for you or not (maybe I'm missing something). And this is more for desktop based monitors, so I don't mind if you want to put a condition on it such as if (!isMobile), etc...

Thanks.

plia7, 2024-03-18:
> I don't know why this might not work. And for my taste, this is not so necessary because it is always easy to press the button to explicitly specify the orientation. Especially considering that mobile devices support is limited.

Ok, I was able to get it fired. orientationchange is deprecated and what you need to use instead is:

		  screen.orientation.addEventListener("change", (event) => {
			const type = event.target.type;
			const angle = event.target.angle;
			console.log(`ScreenOrientation change: ${type}, ${angle} degrees.`);*/
		  });

So do you think we could add that? I don't mind making the change locally for myself, but I think it's a good change for everybody, I'll let you decide :) 

Thanks.

plia7, 2024-04-07:
> I really don't think anyone else would need it. The application of this is too specific.
Yes, I had to wrap around my new code with if (!tools.browser.is_mobile) { ... } else { your original code }
otherwise it would break my iOS safari webkit pikvm page. It doesn't seem to know about screen.orientation there, so it still needs window orientationchange, but I don't need it for mobile right now, so it's fine.

Thanks again for the feature, it works great and really improves productivity.