pikvm/pikvm #1078: Map VNC Ports to KVM Hub Devices?

seanwalter, 2023-07-31:
**Is your feature request related to a problem? Please describe.**
PiKVM over VNC is great for connecting to a single device, but when using a KVM hub, it seems the only way to change devices is to make the change through the PiKVM Web UI and then reconnect with a VNC client. Having a way to trigger PiKVM to change hub devices over VNC would be extremely helpful.

**Describe the solution you'd like**
A way I have seen similar things done in other contexts is for the VNC server to adapt based on the port the client accesses. For the feature I am requesting, accessing port 5901 could trigger PiKVM to set the KVM Hub to Input0 before communicating back to the client, 5902 could trigger Input1, and so on, with port 5900 (default) triggering no change to the KVM Hub.

**Describe alternatives you've considered**
I currently use the PiKVM Web UI to change my KVM Hub source, which works fine, but often I find it easier to work directly in a VNC client to access my systems, rather than through the PiKVM UI.

**Additional context**
Hopefully the above is fully explanatory. Maybe the following adds marginal clarity...what I would like is to be able to set up my VNC client to access 4 devices connected to my PiKVM through a KVM hub as follows, with my PiKVM available on the network at PiKVM.local:

- Pi-KVM.local:5900 - access whatever device is currently "active" on the KVM hub
- Pi-KVM.local:5901 - PiKVM switch the KVM hub to the device at Input0, then respond to client
- Pi-KVM.local:5902 - PiKVM switch the KVM hub to the device at Input1, then respond to client
- [same idea for ports 5903 and 5904] 

If something like this is already possible and I have somehow missed it in my review of the documentation, apologies. I have read a lot and searched a lot, and have not found anything like this.

Thank you!

mdevaev, 2023-08-03:
Hello! This is the interesting idea, but there is no way to implement it right now. I'll leave that for the relatively near future.

mdevaev, 2025-05-07:
The future has arrived, but in a different way. In the new release, LeftAlt,LeftAlt,1-8 hotkeys will be available for switching ports of the [PiKVM Switch](https://docs.pikvm.org/switch/). This should cover the needs.

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev