pikvm/pikvm #1080: PiKVM virtual keyboard doesn't allow repositioning via touch, requires mouse (or possibly stylus) input to get 4 way arrow to drag via title bar

spoelstraethan, 2023-08-04:
**Describe the bug**
PiKVM virtual keyboard doesn't allow repositioning via touch, requires mouse (or possibly stylus) input to get 4 way arrow to drag via title bar.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Open PiKVM's KVM interface and view the connected device
2. Click on the top right menu and "Show keyboard"
3. If the remote device isn't currently displaying anything the keyboard shows up in the middle of the screen, if it is showing something it still shows up near the middle
4. Attempt to drag the keyboard by touching the title bar area of the floating "window" inside the PiKVM interface
5. Fail to drag with a finger and resort to an attached mouse or trackpad IF one is available.

**Expected behavior**
Allow repositioning the virtual keyboard/terminal/HDMI view with a finger and not just the mouse.

**Desktop (please complete the following information):**
 - OS: ChromeOS r115
 - Browser:  Chrome
 - Version: 115

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: PiKVM 4 Plus
  - Video capture type: CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
    ```
    [kvmd-webterm@pikvm ~]$ pacman -Q | grep kvmd
    kvmd 3.238-1
    kvmd-fan 0.26-1
    kvmd-oled 0.24-1
    kvmd-platform-v4plus-hdmi-rpi4 3.238-1
    kvmd-webterm 0.43-1
    ```
  - uStreamer version: `pacman -Q | grep ustreamer`
    ```
    [kvmd-webterm@pikvm ~]$ pacman -Q | grep ustreamer
    ustreamer 5.41-1
    ```
  - Linux kernel: `uname -a` `Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux`

arch1mede, 2023-08-06:
While this is a known issue right now, there are plans on improving the mobile experience this fall, loose ETA.

mdevaev, 2023-08-09:
This is not related with generic mobile ui, but I don't know why it's not working. `ontouch*` events is not firing on the desktop chrome.

spoelstraethan, 2023-08-09:
I can try enabling the input logging flag on ChromeOS and see if it shows
the browser interaction portions on the PiKVM page. If there is a verbose
mode for the web server I can enable that as well.

I can also test in Chrome on a Windows touchscreen device to see if Google
is messing with things on ChromeOS causing the bad behavior.

mdevaev, 2023-08-09:
I found a device with touch, but I just don't know how to fix it yet. If I come up with something, I'll fix it.

spoelstraethan, 2023-08-24:
I think I figured out at least a hint of what is happening, when I touch
the screen I'm seeing everything in the PiKVM interface jump a couple
pixels, so that's telling me it is treating it like a hold and drag for the
top level of the interface and not passing the events into the internal
modals.

mdevaev, 2025-05-10:
I found a way to fix it. God only knows why it works. It will be available in the next release.

closed 2025-05-10 by mdevaev