pikvm/pikvm #1081: H.264/ WebRTC not working behind Apache2 Reverse Proxy

jbyerline, 2023-08-04:
**Describe the bug**
H.264/ WebRTC not working behind Apache2 Reverse Proxy

The MJPEG and all other functionality works except for WebRTC

**Proxy Config**
```
<VirtualHost *:80>
  ServerName kvm.byerline.me
  Redirect permanent / https://kvm.byerline.me/
</VirtualHost>

<VirtualHost *:443>
    ServerName kvm.byerline.me

    SSLEngine on
    SSLCertificateFile /etc/apache2/ssl/ca.crt
    SSLCertificateKeyFile /etc/apache2/ssl/ca.key
    SSLCertificateChainFile /etc/apache2/ssl/ca.ca-bundle
    
    SSLProxyEngine On
    SSLProxyVerify none
    SSLProxyCheckPeerCN off
    SSLProxyCheckPeerName off
    SSLProxyCheckPeerExpire off

    RewriteEngine on
    RewriteCond ${HTTP:UPGRADE} ^WebSocket$ [NC]
    RewriteCond ${HTTP:CONNECTION} ^Upgrade$ [NC]
    RewriteRule .* wss:/10.0.0.173%{REQUEST_URI} [P,L]

    ProxyPass /extras/webterm/ttyd/ws wss://10.0.0.173/extras/webterm/ttyd/ws
    ProxyPass /extras/webterm/ttyd/ https://10.0.0.173/extras/webterm/ttyd/
    ProxyPassReverse /extras/webterm/ttyd/ https://10.0.0.173/extras/webterm/ttyd/

    ProxyPass / https://10.0.0.173/
    ProxyPassReverse / https://10.0.0.173/

    <Location /api/ws>
            Order allow,deny
            Allow from all
            ProxyPass wss://10.0.0.173/api/ws
            ProxyPassReverse wss://10.0.0.173/api/ws
    </Location>

    ProxyRequests off
</VirtualHost>
```

**Expected behavior**
Expect WebRTC to work over proxy

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser - Chrome

**PiKVM info:**
  - PiKVM platform v4 plus

**Additional context**
Followed guide from here: https://znil.net/index.php/PiKVM_hinter_Apache_Reverse_Proxy

mdevaev, 2023-08-04:
Hello. You should also pass `/janus/ws` as websocket.

jbyerline, 2023-08-04:
Thank you, that fixed it. I had to add:

```
<Location /janus/ws>
            Order allow,deny
            Allow from all
            ProxyPass wss://10.0.0.173/janus/ws
            ProxyPassReverse wss://10.0.0.173/janus/ws
</Location>
```

Now WebRTC works

closed 2023-08-04 by jbyerline