pikvm/pikvm #1083: TOTP no longer working

semool, 2023-08-11:
**Describe the bug**
TOTP Login is no longer working.
Then i disabeled with "kvmd-totp del" and login works.
Reinit with "kvmd-totp init" and it shows the secret an the QR Code but i cannot login with the generated Codes (testet with Bitwarden and Keepass)

**Log**
`[2023-08-11 11:23:34 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:23:34 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:23:40 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:23:40 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:23:48 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:23:48 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:24:00 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:24:00 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:24:08 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:24:08 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:24:12 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP
[2023-08-11 11:24:12 kvmd.service] --- aiohttp.access                    INFO --- [- / 192.168.1.34] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.1.20:4433/login/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36'
[2023-08-11 11:24:20 kvmd.service] --- kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' by TOTP`

**Desktop (please complete the following information):**
Tested with Vivaldi, Edge and Firefox.

**PiKVM info:**
  - Raspberry Pi 4 - 4GB
  - kvmd 3.244-1
  - kvmd-fan 0.26-1
  - kvmd-oled 0.24-1
  - kvmd-platform-v2-hdmi-rpi4 3.244-1
  - kvmd-webterm 0.43-1
  - ustreamer 5.41-1
  - Video capture CSI bridge
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-08-11:
Hello. Check the `timedatectl` output. It is important that the time on your KVM is synchronized via NTP.

semool, 2023-08-11:
It is:
timedatectl
               Local time: Fr 2023-08-11 19:45:32 CEST
           Universal time: Fr 2023-08-11 17:45:32 UTC
                 RTC time: n/a
                Time zone: Europe/Berlin (CEST, +0200)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

mdevaev, 2023-08-11:
I've just checked, everything works. Try setting it up again and make sure that the time on the phone and kvm are the same.

semool, 2023-08-12:
This morning i setup it again. Its working like before. With the same OTP Apps as before. Im a little confused. Must be some Time Sync Problem in the OTP Apps on my Smartphone yesterday. I thought it was yesterday's update, Sorry.

semool, 2023-08-12:
Not a Bug! Seems time Sync Problem on my Side.

closed 2023-08-12 by semool

mdevaev, 2023-08-12:
No problem.