pikvm/pikvm #1084: Intermittent black screen in WebRTC when connected through an external network after some time

plia7, 2023-08-11:
**Describe the bug**
Intermittent black screen in WebRTC when connected through an external network after some time.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect to pikvm from an external network using tailscale installed on the external network connecting machine.
2. Stay connected and use it for few hours.

**Expected behavior**
Being able to stay connected and use it for few hours without any interruptions assuming stable and fast internet connection on both ends.

**Actual behavior**
Intermittent black screen in WebRTC starts appearing after some time (could be after minutes or hours) where it works fine until that point, then prior to the black screen it might appear like the mouse movement is slower to react (although no observed internet quality reduction on both ends). Then when the black screen appears for few seconds or minutes, it shows like an orange sd memory card icon (all other icons don't have a color), and in the window caption it shows 0 fps:

[image]

These are my runtime settings:

[image]

When opening the log, it shows:

"23-08-09 21:39:00 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [497280146060660] Waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
[2023-08-09 21:39:01 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [497280146060660] Failed to add some remote candidates (added 0, expected 1)"

As well more detailed log for when the issue happens:

[[2023-08-11 001109 kvmd.service] --.txt](https://github.com/pikvm/pikvm/files/12321042/2023-08-11.001109.kvmd.service.--.txt)

After sometime it seem to restore the screen, but the mouse movement and general feel is that the connection is not as stable, and it keeps going back to black screen again after few seconds so it's not very useable at this point.

Note: If I switch to mjpeg, it seem to show the screen then, but doesn't seem like the mouse/input is very reactive in that mode. Trying to click stop stream and start stream while in webrtc doesn't seem to help.

**Possible workaround although not always consistent**
It seems like if I open an incognito browser or the second pikvm web page for the same host pc it might work better for some time and the black screen issue will be gone, but after some time, it starts creeping in into the incognito or second pikvm session again so I have to keep altering between pikvm 1, incognito and pikvm 2 to "escape" the issue but this is not an ideal scenario and I rather not have any interruptions to begin with, just like when I was using it in my local network without this issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 114.0.5735.199 (Official Build) (64-bit)
or
Microsoft Edge Version 115.0.1901.183 (Official build) (64-bit)
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
 PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.236-1
kvmd-fan 0.26-1
kvmd-oled 0.24-1
kvmd-platform-v3-hdmi-rpi4 3.238-1
kvmd-webterm 0.43-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  ustreamer 5.41-1
  - Linux kernel: `uname -a`
 Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
This is the same system setup I used to have in my local network with the connecting pc and host pc - and I never encountered this issue when everything was in the local network. I only started to encounter this issue when the connecting pc is on an external network (I haven't changed any settings from that "local network working setup").

mdevaev, 2023-08-11:
Please update to the latest version and try to reproduce it again.

plia7, 2023-08-12:
> Please update to the latest version and try to reproduce it again.

I updated to kvmd-platform-v3-hdmi-rpi4 3.244-1.
I'll keep you posted if I still encounter this issue with the latest version.

plia7, 2023-08-14:
> Please update to the latest version and try to reproduce it again.

@mdevaev The black screen blinking continues in the latest version. I wouldn't say it's as rare as you would expect, but happens more frequently like every x amount of minutes, although it doesn't seem to last long this time around, it's still a frustrating experience overall.

Here are the logs:

[logs 8.14.2023.txt](https://github.com/pikvm/pikvm/files/12332449/logs.8.14.2023.txt)

It happened like 2-3 times in the last 45 minutes session when I was connected. Although it doesn't seem to persist as long like before.

The logs mention things like (which I also mentioned above originally):

[2023-08-14 10:31:59 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_cb_nice_recv:3078] [1829142975794202]     SRTCP unprotect error: srtp_err_status_replay_fail (len=140-->140)

And:

[2023-08-14 10:07:17 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [7136366473408474] Failed to add some remote candidates (added 0, expected 1)

Internet is stable and fast on both connecting pc (using wifi) and host pc (host pc is using wired ethernet and as well as the pikvm):

Speed test in connecting pc:

[image]

Speed test in host pc:

[image]

Thank you.

plia7, 2023-08-14:
> Please update to the latest version and try to reproduce it again.

@mdevaev Just had some strong encounter with the issue, it went to became slow again in conjunction with the black screen: 

[image]

And the black screen seem to persist longer this time around which makes it quite unusable and unreliable when it happens and keeps happening during this session unless I go incognito or start another pikvm session, here are the logs:

[logs 8.14.2023 take 2.txt](https://github.com/pikvm/pikvm/files/12338593/logs.8.14.2023.take.2.txt)

Internet is still stable and fast on both connecting pc (using wifi) and host pc (host pc is using wired ethernet and as well as the pikvm):

Speed test in connecting pc:

[image]

Speed test in host pc:

[image]

Thank you.

mdevaev, 2023-08-28:
Note: https://github.com/meetecho/janus-gateway/blob/0.x/html/janus.js#L1949

plia7, 2023-08-28:
@mdevaev As discussed, here are the triangles you requested:

[image]

plia7, 2023-10-01:
> I need to add myself to being affected by this but in my case I'm in a completely internal network. I require webrtc/h264 because I need audio - I get audio notifications when there is a notification of new emails/messages coming through from coworkers so I know I need to open the tab to answer. This cannot be done on mjpeg. mjpeg works perfectly but I require the audio which only works on h264 mode. However, I keep getting these black screens and the latency is horrendous, I move the mouse and it does not immediately respond. On mjpeg there is a slight delay for cursor movement but it is reliable and usable. On h264 unusable. Please prioritize this, because not being able to see the screen of the remote device breaks the main use case of PiKVM. mjpeg is not an option due to the lack of audio.

Weird, I never experienced it in an internal network, are you using wired ethernet cable or wifi? The problem is better for me when I'm using it from a different location, so it might depend on location, distance, ISP, etc.

plia7, 2023-10-01:
> Just wanna say changing the polling rate of the mouse to 10ms instead of the default 100ms greatly improved the pointed accuracy and speed. Consider changing the default to 10ms... and I tried to use webrtc in Firefox and have not had any black screens so far so it seems like you might need to investigate how it's working in Chrome, which is the most popular. For now I am fine using Firefox to get what I need to do my job but I am risking getting told off at work and losing my job because Chrome is the only allowed browser.

Mine is already at 10ms. How using the browser is affecting the host pc?

Majestic7979, 2023-10-02:
> Mine is already at 10ms. How using the browser is affecting the host pc?

I don't know why I was having h264 dropouts in Chrome but not in Firefox. I also enabled the USB ethernet with ncm driver in the overrides.yaml and I'm serving wired ethernet via USB to the remote device through the PiKVM and this eliminated my issues. So I think it was a connectivity problem. Probably packet loss which Firefox is somehow handling better than Chrome.

plia7, 2023-10-02:
> I don't know why I was having h264 dropouts in Chrome but not in Firefox. I also enabled the USB ethernet with ncm driver in the overrides.yaml and I'm serving wired ethernet via USB to the remote [...]

The problem is you can't do keyboard shortcuts in firefox when in full screen unlike chrome.

plia7, 2023-10-19:
Hi @mdevaev So I actually started to experience the issue again in other location, no internet speed loss in both connecting or host pc... So is it Tailscale again? and what's interesting that in Firefox there are less black screen moments than there are in google chrome... Is it something that can be improved so there is less black screen in chrome like in firefox for when this issue happens? And again, only from an external network and only through h264 this issue happens - I wonder is there is a way to clean the tailscale pipe so the issue stops... As it could work fine for weeks, months, and in one time, somehow have this massive black screen issue? Besides tailscale are there any other supported vpn alternatives?

Thanks.

mdevaev, 2023-10-25:
I've reported about js error: https://github.com/meetecho/janus-gateway/issues/3283

plia7, 2023-10-26:
> I've reported about js error: [meetecho/janus-gateway#3283](https://github.com/meetecho/janus-gateway/issues/3283)

Thank you, here is the main "trademark" of the issue, when it blinks it shows this prompt:

[image]

"H.264 - 1920x1080 / Invalid PeerConnection / 0 fps dynamic"

As you can see the sd memory card icon is still showing in green color (not yellow)

mdevaev, 2023-12-14:
@plia7 @Majestic7979 I've released small update for this, just a workaround. I don't expect this to significantly improve the situation, but at least it should provide reconnection in case of a bug with unmute. My cursory research shows that this is some kind of bug in Chrome. Since the Janus upstream doesn't know what's going on, I'll try to fix it myself, somehow.

Please update PiKVM OS.

Majestic7979, 2023-12-14:
Thanks @mdevaev you're a really nice guy and a responsive developer. I installed. I'll test and report back.

plia7, 2023-12-14:
> @plia7 @Majestic7979 I've released small update for this, just a workaround. I don't expect this to significantly improve the situation, but at least it should provide reconnection in case of a bug [...]

Thanks, I'll update at my earliest convenience (although I'm not in the condition to test it at the moment).

Majestic7979, 2023-12-15:
@mdevaev whatever you did, has cured the issue for me. Thanks again for your hard work on PiKVM. I wanna know if there is any way to improve the latency on h264? When I move the mouse cursor it takes about a second for it to catch up with the grey dot. Is this just how it's always going to be, or are there any optimizations I can do to make the cursor arrow follow the grey dot more responsively? I am also a relatively fast typist and when I am editing configuration files it also takes about a second for a character to appear after I've typed it. Sometimes the Shift key gets stuck and whatever I type is capitalized, and if I hit the dot key . I get a > instead... Shift randomly stays activated (verified through virtual keyboard) when I hold Shift down to capitalize an acronym for example. And sometimes I'll get repeated characters, for example instead of lemon I will get leeeeeeemon.

mdevaev, 2023-12-15:
The latency when working through a VPN is an unavoidable evil. You can try reducing the mouse polling rate in the system menu and make h264 gop 0. The latency depends on the network conditions, at the moment everything that I could optimize for my part has already been done. I have it at 200ms between continents.

plia7, 2023-12-16:
> @plia7 @Majestic7979 I've released small update for this, just a workaround. I don't expect this to significantly improve the situation, but at least it should provide reconnection in case of a bug [...]

I updated to kvmd-platform-v3-hdmi-rpi4 3.288-1, will try to test when I get a chance and report back. Thanks.

Majestic7979, 2023-12-18:
@mdevaev unfortunately the problem has returned.

mdevaev, 2023-12-18:
Well, that only fixed part of it. I will investigate further.

tgrrr, 2023-12-21:
Hi,

I have the identical problem. It improved slightly after the update. However, still persists.

> Here is the main "trademark" of the issue, when it blinks it shows this prompt:
> [image]
> "H.264 - 1920x1080 / Invalid PeerConnection / 0 fps dynamic"

Same issue:

> Note: It seems like once a session gets a black screen, it seem to be "infected" with it at that point, it never recovers from it and keeps going back to it so it will keep altering to black screen or slowness even if the screen shows up again and might work for some time.
> As you can see, there is no substantial slowdown in the quality or speed of the internet (speedtest screenshots).

Desktop (please complete the following information):

    OS: [e.g. iOS]
    MacOS 14.1.2
    Browser [e.g. chrome, safari]
    Google Chrome Version 120.0.6099.109 (Official Build) (arm64)
    
    or
    FireFox 120.0.1 (64-bit)
    
    VNC client (if used)
    N/A

Before Update:

PiKVM info:

    Raspberry Pi board version [e.g. RPi 4]
    PiKVM v3 Hat
    PiKVM platform [e.g. v2-hdmi]
    Video capture type [e.g. CSI bridge]
    KVMD version: pacman -Q | grep kvmd
    kvmd 3.287-1
    kvmd-fan 0.26-1
    kvmd-oled 0.26-1
    kvmd-platform-v3-hdmi-rpi4 3.287-1
    kvmd-webterm 0.47-1

    uStreamer version: pacman -Q | grep ustreamer
    ustreamer 5.46-1
    Linux kernel: uname -a
    Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux

After update:

PiKVM info:

    Raspberry Pi board version [e.g. RPi 4]
    PiKVM v3 Hat
    PiKVM platform [e.g. v2-hdmi]
    Video capture type [e.g. CSI bridge]
    KVMD version: pacman -Q | grep kvmd
    kvmd 3.288-1
    kvmd-fan 0.26-1
    kvmd-oled 0.26-1
    kvmd-platform-v3-hdmi-rpi4 3.288-1
    kvmd-webterm 0.47-1
    uStreamer version: pacman -Q | grep ustreamer
    ustreamer 5.46-1
    Linux kernel: uname -a
    Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux    

    Log:

    [2023-12-18 04:48:26 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => == ustreamer/main      -- Creating session 0xb0adc758 ...
[2023-12-18 04:48:26 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [252830446426366] Creating ICE agent (ICE Full mode, controlling)
[2023-12-18 04:48:27 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [252830446426366] Waiting for candidates-done callback... (slow gathering, are you using STUN or TURN for Janus too, instead of just for users? Consider enabling full-trickle instead)
[2023-12-18 04:48:28 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [252830446426366] Failed to add some remote candidates (added 0, expected 1)
[2023-12-18 04:48:29 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [252830446426366] The DTLS handshake has been completed
[2023-12-18 04:48:29 kvmd-janus.service] --- kvmd.apps.janus.runner            INFO --- => [WARN] [252830446426366] Failed to add some remote candidates (added 0, expected 1)
[2023-12-18 04:48:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62753.927    stream] -- H264: Requested keyframe by a sink client
[2023-12-18 04:48:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62754.019    stream] -- H264: Requested keyframe by a sink client
[2023-12-18 04:48:29 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62754.628    stream] -- H264: Requested keyframe by a sink client
[2023-12-18 04:48:30 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62754.730    stream] -- H264: Requested keyframe by a sink client
[2023-12-18 04:48:30 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62755.297    stream] -- H264: Requested keyframe by a sink client
[2023-12-18 04:48:30 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [62755.391    stream] -- H264: Requested keyframe by a sink client

PiKVM settings 
[image]

shamsevan, 2024-01-23:
Hi , 
I am experiencing same issue frequently in chrome ! 
while trying to Firefox , the issue seems bit less but it is supper slow there. 
The main frustating part for firefox is , it doesn't accept the keyboard shortcuts , which is making it pretty difficult to run and execute tasks. please give me a solution

blackandcold, 2024-01-24:
Have the same problem - strange it happens in Vivaldi (Chrome) and NOT in Firefox

mdevaev, 2024-01-24:
Is anyone who has this problem playing often and constantly ready to provide me with a TeamViewer session to their computer so that I can look at it?

plia7, 2024-01-31:
> Is anyone who has this problem playing constantly ready to provide me with a TeamViewer session to their computer so that I can look at it?

Hi @mdevaev, 
@shamsevan says the issue happens 10 times in an hour and actually after every few minutes, he's on discord and says he could let you remote to his machine to investigate (he uses the latest os version with the partial fix).

Thank you.

Majestic7979, 2024-02-08:
@mdevaev I found an interesting thing that might help you. I'm using the latest build of PiKVM as of writing this (ran your curl command) so you know which version. Basically, until there is some audio playback the screen will keep cutting out. But the moment there's some audio playback, it stops and stays stable. I can reproduce this every single time. How I do is I logon to Windows while it's in JPEG mode, then I switch to WebRTC and it starts cutting out. I open the Windows volume slider (which makes a sound when you move and release it) and as soon as I hear the Windows make a sound, I get a stable connection. So it seems like until it's silent the screen cuts out!! I use WebRTC for the audio capability over HDMI too, which JPEG mode does not support. Please note that when I swap to WebRTC I have to open the PiKVM menu and move the volume to 100%. I am talking about the Windows volume slider just to make some sound. As long as the remote machine sends a sound the issue will stop. So that narrows it down further for you to investigate.

mdevaev, 2024-02-08:
Do I understand correctly that even if the sound is turned on in PiKVM, this is not enough, and you need the host to start playing the sound itself?

Majestic7979, 2024-02-08:
> Do I understand correctly that even if the sound is turned on in PiKVM, this is not enough, and you need the host to start playing the sound itself?

That's what I was thinking until today but for some reason it was glitching again. I waited a few days just to make sure before making that comment, but today I had issues again. It does seem related to the audio. I'll be monitoring some more to see if it was just "luck" or if there really is a link to audio causing issues.

RSATom, 2024-02-19:
@mdevaev Can you please clarify, setup you are using to access streams from Janus server doesn't use TURN server at all, right?

mdevaev, 2024-02-19:
@RSATom yes, only STUN.

RSATom, 2024-02-20:
Ok, did you see something like "ICE state changed to failed" in JS console right before issue?

mdevaev, 2024-02-20:
@RSATom this is a first option. The second is many "remote track muted", see above.

RSATom, 2024-02-20:
I think there are multiple issues here. So it's required to split them and only after try to fix.

So, first of all it's required to understand how to simplify reproducing this issue. And I think it should be simplest part - if there is firewall present - one can just disable UDP packets passing through firewall and wait ICE disconnected (~1 second), ICE failed (~2 seconds).

Then browser client will try to restore connection. And there are 2 ways to restore WebRTC connection: full reconnect and ICE renegotiation. Since you are using custom plugin I can't say if it supports the second one (please clarify).

Other thing it's required to take into account - full reconnect with Janus can be tricky, especially if you don't detach from Janus plugin on Peer Connection drop.

Maybe all above already known to you, please tell me. I just need to understand what exactly you already tried.

mdevaev, 2024-02-20:
> Since you are using custom plugin I can't say if it supports the second one (please clarify).

No, it's not supported: https://github.com/pikvm/ustreamer/blob/master/janus/src/plugin.c

Although I wrote this, I have a fairly superficial idea of the internal structure of Janus. If there are any problems, I completely terminate the connection with Janus and re-establish it:

https://github.com/pikvm/kvmd/blob/master/web/share/js/kvm/stream_janus.js

Perhaps this is really due to packet loss, but users say that this problem either does not reproduce in Firefox or occurs much less frequently than in Chrome.

If it's convenient for you, let's get in touch via PM at discord: https://discord.gg/bpmXfz5 (mdevaev there)

RandomJerk, 2024-02-23:
I can confirm that Firefox works better and does not see black screen interruptions.  The issue happens on Chrome and Chromium based browsers

plia7, 2024-02-28:
> I can confirm that Firefox works better and does not see black screen interruptions.  The issue happens on Chrome and Chromium based browsers

Yes, maybe Firefox works a little better, but it doesn't support keyboard shortcuts , which is making it pretty difficult to run and execute tasks so it's not really a suitable workaround.

RSATom, 2024-02-28:
did anybody look into `chrome://webrtc-internals`?

plia7, 2024-02-28:
> did anybody look into `chrome://webrtc-internals`?

I believe @mdevaev looked in mine few months ago when I had the issue. Or possibly he looked again recently in @shamsevan machine.

RSATom, 2024-02-28:
I just got an Idea it can be broken by some reason h264 video stream (i.e. ICE connection is fine)... And working Firefox maybe confirms it... Firefox and Chrome have different codecs implementation inside.

RSATom, 2024-02-29:
@plia7 everything is possible. But to say more it's required to know if there is `failed` state happens for ICE before black screen. I'm still waiting my HDMI-to-CSI device to start debugging issue myself. I hope it will arrive on this weekend...

RSATom, 2024-02-29:
Firefox is slowly dying, so yes, it's not a solution in any case...

Majestic7979, 2024-02-29:
Based on comment above I turned off H/w video decoding in chrome flags, and it makes no difference. I also noticed that in Firefox my mouse movements are more fluid, more responsive. The delay appears to be much less, so if I move the mouse from one corner of the screen to the opposite corner, I can see my mouse moving faster in Firefox, in Chrome there is a terrible delay. So Chrome is unusable in its own way, but it's the browser I need to use sometimes due to customer restrictions in their IT.

plia7, 2024-02-29:
> Based on comment above I turned off H/w video decoding in chrome flags, and it makes no difference. I also noticed that in Firefox my mouse movements are more fluid, more responsive. The delay [...]

Did you also try to toggle off "use graphics acceleration when available" setting?

I wonder why the mouse movement and usability appears to be slower in firefox, I guess that's a separate issue.

Thanks.

plia7, 2024-02-29:
> @plia7 everything is possible. But to say more it's required to know if there is `failed` state happens for ICE before black screen appear. I'm still waiting my HDMI-to-CSI device to start debugging issue myself. I hope it will arrive on this weekend...

Looking at the logs on this ticket, I don't see ICE failed state, but it's maybe happening on the first time which causes it to go into this black screen loop (and logs typically gathered after that), so yes, maybe it's good to have full log for it, to make sure 100% if it happens or not so then a proper ticket can be submitted on the chromium bug report website (if it actually turns out not to be pikvm's fault in this case).

Thank you.

RSATom, 2024-03-04:
Finally I've got my HDMI-2-CSI device, did some debugging, and found with current implementation `failed` ICE state never happens. I think the reason is related to handling `disconnected` ICE state by browser and `janus.js`:
1. When peer connection is established and running, UDP packets can stop coming by some reason, and after some amount of missed packets, ICE State switches to [disconnected](https://developer.mozilla.org/en-US/docs/Web/API/RTCIceTransport/state#disconnected) state.
_Important note about `disconnected` ICE state: the name of this state can be confusing but it's not final state. If packets will start flow again - state will switch to `connected` state again. And only if there will be no packets too long amount time state will be switched to `failed` state. And it's final state._
2. Near the same time WebRTC library sends `mute` event to track. And `janus.js` handles it by scheduling [track remove](https://github.com/meetecho/janus-gateway/blob/63ee713102f453447e3c912a7bb45791d4d198c5/html/janus.js#L1926-L1940) from **internal Janus.js structures** (but it doesn't mean track disappears from RTCPeerConnection, it's only `janus.js` internal logic)
3. If there is no `unmute`, after timeout `janus.js` removes track and [sends](https://github.com/meetecho/janus-gateway/blob/63ee713102f453447e3c912a7bb45791d4d198c5/html/janus.js#L1933) `remotestream` event. And this is the point where `stream_janus.js` gets confusing [onremotestream](https://github.com/pikvm/kvmd/blob/4690e330881fcd2ccb30a2ce154be0fdbf1990db/web/share/js/kvm/stream_janus.js#L222) without video tracks. I.e. it's intended behavior, and there is no reason to disconnect from janus at this point. The right time to start reconnect logic is `failed` ICE state (or even better `failed` state of [RTCPeerConnection.connectionstate](https://developer.mozilla.org/en-US/docs/Web/API/RTCPeerConnection/connectionState#failed)

Unfortunately all above has nothing to do with "black screen" issue. When I enabled UDP packets flow on my test environment - reconnect to Janus happened successfully and I was able to see incoming video...

plia7, 2024-03-04:
> Finally I've got my HDMI-2-CSI device, did some debugging, and found with current implementation `failed` ICE state never happens. I think the reason is related to handling `disconnected` ICE state by browser and `janus.js`:
> 
> 1. When peer connection is established and running, UDP packets can stop coming by some reason, and after some amount of missed packets, ICE State switches to [disconnected](https://developer.mozilla.org/en-US/docs/Web/API/RTCIceTransport/state#disconnected) state.
>    _Important note about `disconnected` ICE state: the name of this state can be confusing but it's not final state. If packets will start flow again - state will switch to `connected` state again. And only if there will be no packets for too long amount time state will be switched to `failed` state. And it's final state._
> 2. Near the same time WebRTC library sends `mute` event to track. And `janus.js` handles it by scheduling [track remove](https://github.com/meetecho/janus-gateway/blob/63ee713102f453447e3c912a7bb45791d4d198c5/html/janus.js#L1926-L1940) from **internal Janus.js structures** (but it doesn't mean track disappears from RTCPeerConnection, it's only `janus.js` internal logic)
> 3. If there is no `unmute`, after timeout `janus.js` removes track and [sends](https://github.com/meetecho/janus-gateway/blob/63ee713102f453447e3c912a7bb45791d4d198c5/html/janus.js#L1933) `remotestream` event. And this is the point where `stream_janus.js` gets confusing [onremotestream](https://github.com/pikvm/kvmd/blob/4690e330881fcd2ccb30a2ce154be0fdbf1990db/web/share/js/kvm/stream_janus.js#L222) without video tracks. I.e. it's intended behavior, and there is no reason to disconnect from janus at this point. The right time to start reconnect logic is `failed` ICE state (or even better `failed` state of [RTCPeerConnection.connectionstate](https://developer.mozilla.org/en-US/docs/Web/API/RTCPeerConnection/connectionState#failed) (but obviously it can keep black screen for longer period of time)
> 
> Unfortunately all above has nothing to do with "black screen" issue. When I enabled UDP packets flow on my test environment - reconnect to Janus happened successfully and I was able to see incoming video...

Ok so what this all mean? Do you need the conditions to reproduce the issue to analyze it? Is it possible for you to connect with one of the pikvm users who reported the issue/have the issue and can reproduce it such as through a TeamViewer session to try to analyze the issue on their end (which is what @mdevaev tried to do but couldn't determine the root cause)? Or what is it that you need that will help you to get to the root cause of the issue? Do we need to send HDMI-2-CSI device to one of the users who have the issue? 

Keep in mind it's a very specific set of conditions to reproduce the issue that includes being on an external network and connecting to tailscale , so I'm not sure if you are testing all of this locally with a physical pikvm without actually connecting to a remote pikvm through tailscale.
So either you need to connect to a remote pikvm (on an external network) through tailscale and see if you could get the intermittent black screen at some point and then start analyzing it with the HDMI-2-CSI device or maybe some of the other members in this post have more suggestions or could help with providing you with the conditions needed to reproduce the issue.

Thanks.

RSATom, 2024-03-04:
@plia7 I just shared my findings, nothing more at this moment. HDMI-2-CSI - is device all pikvm users already have actually. It's where you put you HDMI cable :)
Also, I've found another one possible issue and working with @mdevaev on it.

plia7, 2024-03-04:
@mdevaev reached out to me and asked if someone currently using the external network with tailscale and is able to reproduce the issue - @RandomJerk @blackandcold @tgrrr or anyone else who read this - Are you available to assist through a TeamViewer session so the issue can be inspected/further analyzed by @mdevaev and @RSATom? 

Thank you.

RSATom, 2024-03-04:
it would be great if someone could provide me remote access to pikvm with reproducible issue. I don't need real server, I just need video stream. It can be even some tv stick attached to pikvm/

plia7, 2024-03-04:
> **Additional context**
> This is the same system setup I used to have in my local network with the connecting pc and host pc - and I never encountered this issue when everything was in the loca

I found 10 people who have the issue, we created a group in Discord. Please come and join us.

Majestic7979, 2024-03-04:
I feel I need to just remind all reading at this stage that I do not use tailscale. I am accessing the PiKVM from within the internal network... so I'm on 192.168.1.20 and the PiKVM is 192.168.1.21, for example. I get the issue without any VPN or tunneling.

Majestic7979, 2024-03-04:
> it would be great if someone could provide me remote access to pikvm with reproducible issue. I don't need real server, I just need some video stream. It can be even some tv stick attached to pikvm...

Are you able to send me your email address? I will get in touch.

RSATom, 2024-03-05:
@Majestic7979 rsatom_at_gmail_dot_com

RSATom, 2024-03-05:
> I feel I need to just remind all reading at this stage that I do not use tailscale. I am accessing the PiKVM from within the internal network... so I'm on 192.168.1.20 and the PiKVM is 192.168.1.21, for example. I get the issue without any VPN or tunneling.

Are 192.168.1.20 and 192.168.1.21 connected to the same Ethernet switch?

RSATom, 2024-03-05:
> Thanks, you can remove your email now, I've noted it. I'll be writing soon.

:+1: but there is no need to do it. It published on my github profile anyway :smiley:

Majestic7979, 2024-03-05:
> > Thanks, you can remove your email now, I've noted it. I'll be writing soon.
> 
> 👍 but there is no need to do it. It shown on my github profile anyway 😃
> 
> > I'll be bringing the PiKVM to my house, where I have it connected to my UniFi network. It will be connected to a UniFi Flex Mini, that will connect to a USW-24-PoE, and that in turn connected to a UniFi Dream Machine. But it's all the same network within the router configuration.
> 
> and there is no any WiFi in between, right?

I have an Access Point of course, only for the wireless devices. Not for the PiKVM and not for the PC I'll be using because they can be wired and that's how I prefer it.

Majestic7979, 2024-03-05:
I emailed you now, @RSATom

Majestic7979, 2024-03-06:
As a means of updating everyone I'm writing here. @RSATom logged in to my PC via zoom yesterday and showed me how to capture the data he needed to analyze the issue. He now has all the necessary logs, so hopefully some sort of fix can come out of it. I want to thank him for helping and being patient.

Majestic7979, 2024-03-06:
> @Majestic7979 unfortunately I don't have good news for you. According to logs your browser just didn't get UDP packets periodically. So it's network (more likely) or pikvm side (less likely) issue. Did you use VPN yesterday when did issue reproducing?

@RSATom No VPN. Oh :( But it's weird that it doesn't happen on Firefox, only on Chrome... so while it may be the case that network can play a role in this issue, the way it's handled is different. Isn't there a way to code that janus server to stop cutting out? What if it just waits for more UDP packets? I don't know how it works so of course it's just me speculating...

[70 more replies]

closed 2024-03-16 by mdevaev