pikvm/pikvm #1086: First steps password change steps incomplete

agowa, 2023-08-12:
The currently documented steps for changing the passwords within the first steps documentation are incomplete. They only mention root (ssh) and admin (web-if). But currently, it doesn't include /etc/kvmd/vncpasswd (vnc) and /etc/kvmd/ipmipasswd (ipmi).

/etc/kvmd/vncpasswd
```
# This file describes the credentials for VNCAuth. The left part before arrow is a passphrase
# for VNCAuth. The right part is username and password with which the user can access to KVMD API.
# The arrow is used as a separator and shows the relationship of user registrations on the system.
#
# Never use the same passwords for VNC and IPMI users. This default configuration is shown here
# for example only.
#
# If this file does not contain any entries, VNCAuth will be disabled and you will only be able
# to login in using your KVMD username and password using VeNCrypt methods.

# pa$$phr@se -> admin:password
admin -> admin:admin
```

/etc/kvmd/ipmipasswd
```
# This file describes the credentials for IPMI users. The first pair separated by colon
# is the login and password with which the user can access to IPMI. The second pair
# is the name and password with which the user can access to KVMD API. The arrow is used
# as a separator and shows the direction of user registration in the system.
#
# WARNING! IPMI protocol is completely unsafe by design. In short, the authentication
# process for IPMI 2.0 mandates that the server send a salted SHA1 or MD5 hash of the
# requested user's password to the client, prior to the client authenticating. Never use
# the same passwords for KVMD and IPMI users. This default configuration is shown here
# for example only.
#
# And even better not to use IPMI. Instead, you can directly use KVMD API via curl.

admin:admin -> admin:admin
```

mdevaev, 2023-08-12:
VNC and IPMI are disabled by default (and vncauth disabled too even if vnc enabled). Changing their passwords is necessary only when enabled, and is described on the relevant pages

agowa, 2023-08-12:
it still should be part of the "change password" box, as users may use it when they want to change the passwords later on. A short mention and link to these pages would be enough though.

closed 2023-08-14 by mdevaev