pikvm/pikvm #1088: Mouse moves erratically on the bottom of the screen

teckau, 2023-08-13:
PiKVM V3 (Geekworm A3 Kit), Firefox on Windows to use the PiKVM, 4x 8k HDMI Switch w/ hotkeys (seems to work nicely, picked from the list of working HDMI KVM switches). The system that I want to control is an Ubuntu 22.04.

All seems good but the Gnome/Linux/Ubuntu mouse cursor move on the very bottom of the screen not following the movements on the X axis, too. It kind of jumps back and forth, clicking does go through.

The very same Windows (host) is also connected to that HDMI switch (extended desktop, position to the right). Switching to the inpout of that windows box brings up a second Desktop of the Windows systeme within the firefox on desktop one. Mouse moving to the right enters the second screen from the left as expected and all is working. When switching to the Linux input of the HDMI switch I get the correct video, I can click and use the keyboard normally. But the mouse does misbehave.

I have tried rebooting all components, but no cigar. I have configured Linux to just use one work space, no cigar too.
I also have disabled the second desktop of the windows machine (Win+P "just use the primary screen") w/o any different result.

By now I just have these 2 machines connected. Maybe I can make a 3rd connection tomorrow...

Any known problems I am not aware of? Any information needed that I should provide?
Thanks a lot in advance for any help!!!

mdevaev, 2023-08-14:
Hello. I'm sorry, but the hardware you bought is not PiKVM, it is a Chinese copy. We don't provide technical support for this. You should contact the manufacturer, or return it and buy [the original PiKVM](https://pikvm.org/buy), then we will be at your service.

teckau, 2023-08-14:
Hi,

 

I was totally unaware of this. I am sorry that I bought a copy, not because I am unhappy with that, but because I was not giving credits (money) to you. I assume you live on the hardware sales and fund the (free) software with this? In that case – as I am super happy with the PiKVM software – I would like to contribute. Is there any „standard procedure“ to send money out oft hin air to you guys?

Again: I am very happy with the PiKVM software, great work! I also solved my problem when I found out that relative mouse pointer data does work (though it seems that this usually is only needed for UEFI BIOSes, but both of my Ubuntu boxes needed that relative setting).

 

If you let me know how I could send you some money I would like to do so instandly. Keep that great work going on!

Again: Deeply sorry for my mistake, I  just stumbled upon the Chinese thing, bought and used it w/o doing much research. I never ment to bother you w/o paying for it in a commerical setup (that I was unaware of).

 

Best from Germany

Thorsten

mdevaev, 2023-08-14:
Thank you for your understanding and your appreciation. You are absolutely right, the software and support is funded from sales of the hardware, and this is our fulltime job. You can also support us via paypal: https://paypal.me/pikvm

Two questions so I can help you:
* Usually KVM switches have several USB ports. One can be labeled as Keyboard/Mouse, the other is Common. Which port did you insert the PiKVM into? Please take a photo of this.
* Can you record a full-screen video with your desktop and browser, where you will slowly move the cursor so that I can understand the motion distortion?

mdevaev, 2023-08-18:
Sup?

teckau, 2023-08-18:
Hi! 

 

Sup? I do not understand… Sorry for not having answered sooner, but I have missed what kind of information might help here?!

I also have adjusted the mouse to use absolute coordinates which is slow but working. I would highly appreciate help if possible to get behind this but how? What should I gather and upload?

 

But no worries, if you don’t have any idea I am still super happy with this fantastic solution!

 

Best regards

Thorsten

mdevaev, 2023-08-18:
I wrote a little above my questions. This is important for diagnosis. Here they are (copied):
> * Usually KVM switches have several USB ports. One can be labeled as Keyboard/Mouse, the other is Common. Which port did you insert the PiKVM into? Please take a photo of this.
> * Can you record a [...]

teckau, 2023-08-19:
Hello and thanks a lot in advance!

 

I took a picture of the port. There are 2 ports, one is dedicated to mouse and one is dedicated to keyboard (given the symbols). I firstly connected to the other port which did allow me to use keyboard and mouse like now, the notable difference was that the hotkey switching did not work. Once I plugged into the port labels with the keyboard icon I could use Ctrl-Ctrl-# to switch between the ports.

 

The video is 70MB. Too much for an e-mail. I will try to find out if I can upload it on GitHub. If there is anything you can tell me on how to provide that video, please let me know (in case there is no uploaded video because that would indicate me being unable to help myself J ).

Thanks a lot again!!!!

 

Best

Thorsten

teckau, 2023-08-19:
Hi again,

 

it seems that the file hast o be <10MB. My mp4 is not compressable anymore and because of my setup I think that it’s already hard to see the mouse cursor/the problem. So I decided to open a port and share from my computer. Please let me know once you were able to watch it so I can close that private web server.

 

http://trifolium.mooo.com:8000

 

Thanks!

mdevaev, 2023-08-19:
Thank you. I've downloaded video and removed the link from your post. I'll try to figure out what's wrong, I may need additional diagnostics.

mdevaev, 2023-08-20:
Okay, could you try incognito mode in the Firefox without any extensions?

Also please try Chrome or Chromium. I wonder if this will change anything.

teckau, 2023-08-20:
Hi Maxim,

 

thanks a lot for your effort. Unfortunately trying this incognito or with Chrome did not change anything.

 

Cheers

Thorsten

mdevaev, 2023-08-22:
Okay, do you have the opportunity to join the discord chat so that we can try faster diagnostics?

teckau, 2023-08-22:
Hi Maxim, I can try. I have so far no idea what to do to join a chat on Discord. Any hint? Couldn't find any reference to "chat" or "discord" on GIT in the menus, maybe I am missing something?

mdevaev, 2023-08-22:
Just follow the link, this is our community chat.
I'm going to offline now, but you can send me personal message there (mdevaev user). We'll be able to contact when I'm online. https://discord.gg/bpmXfz5

closed 2023-10-07 by mdevaev