pikvm/pikvm #1090: Request for addition of Win+L (Hotkey Lock) to Shortcuts (or easy ability to customize)

mjpcomp, 2023-08-16:
**Is your feature request related to a problem? Please describe.**
There is an existing feature request to have the Win+L (or equivalent for other OS to initiate a LOCK upon close of the terminal window), but, in the mean time, sending Win+L is a convoluted process, and having it added to the Shortcuts bar (or similarly, to the Linux Magic section of shortcuts, have a Windows-key shortcut section where we can have Win+[various letters] to make things easier to do)

**Describe the solution you'd like**
Since it appears the customization to the Web interface doesn't stay with updates, please add at the least a Win+L shortcut, but, ideally, a master list similar to the Alt+SysRq section where various commonly used hotkey/shortcuts based on the Windows Key are available to use.

**Describe alternatives you've considered**
Manually editing the web/kvm/index.html and web/kvm/navbar-shortcuts.pug files to add the keys, however, with multiple PiKVM units deployed, maintaining custom updates is quite the exercise

**Additional context**
Shortcut page in question:
[image]

visit1985, 2023-08-16:
It would be awesome to have a config file where we can add custom shortcuts.

I'm controlling a Windows Client from macOS and would like to map the Pt/Sq button into the Shortcuts, so I don't have to bother about the on-screen keyboard during screenshots.

mdevaev, 2023-08-18:
Menu customization is a big feature that should be done but with low priority.
* Added Win+L for you @mjpcomp
* And PrintScreen for you @visit1985 
Have fun :)

closed 2023-08-18 by mdevaev

visit1985, 2023-08-18:
Thanks @mdevaev, you are awesome! 👍

mjpcomp, 2023-08-18:
> Menu customization is a big feature that should be done but with low priority.
> 
> * Added Win+L for you @mjpcomp
> * And PrintScreen for you @visit1985
> 
> Have fun :)

Thanks, so much! That is perfect - my 3x KS v4Plus thank you 👍 Have so much interest in them after a few people wondered what they were - damn amazing product..

If the customizable shortcut is a low priority, I wonder if doing it as a custom driver would possibly work? Or is the formatting of that current shortcut box only going to be possible through the hardcoded UI files?

mdevaev, 2023-08-18:
I am gradually moving towards providing simple editing through the interface

Josho-SAI, 2024-08-26:
Hi mdevaev, I'm using a PiKVM on a [TESmart KVM Switcher](https://www.amazon.com.au/TESmart-Inputs-Control-Computers-Servers/dp/B07K1X1BPZ). This KVM switcher doesn't have a working mouse however is connected to multiple headless linux servers so no mouse required. Machine BIOS settings are also easily navigated with keyboard only... No mouse required!

The keyboard shortcut to toggle thru the HDMI inputs is "ScrLk + ScrLk + <keys 1 to 8>". The "ScrLk + ScrLk" can be changed to "(left) Ctrl + (left) Ctrl" by entering "ScrLk + ScrLk + F1", and toggled back using "Ctrl + Ctrl + F1"

Can I please request "ScrLk + ScrLk" and "(left) Ctrl + (left) Ctrl"? If this is too much of a PITA, I'm very keen to see if / when there's a option to customise Shortcuts. TIA!