pikvm/pikvm #1093: Help Janus H.264 Indirect Black Screen

TelDragon, 2023-08-22:
Help Janus H.264 Indirect Black Screen...

Is it because my network is unstable??

```
Aug 22 23:08:59 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_traffic_handle:5077] [8269221333422530] ... only sent -1 bytes? (was 1216)
Aug 22 23:08:59 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_traffic_handle:5077] [8269221333422530] ... only sent -1 bytes? (was 1211)
Aug 22 23:09:00 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_stats_handle:4692] [8269221333422530] Got 1226 SRTP/SRTCP errors in the last few seconds (last error: srtp_err_status_replay_old)
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [WSS-0xb2500ae0] Destroying WebSocket client
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => Destroying session 1371484477383498; 0xb2500768
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => Detaching handle from ustreamer; 0xb3c06338 0xb2500f80 0xb3c06338 (nil)
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.124: Agent 0xb1107b78, freeing conncheck_list of stream 0xb2505c28
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [8269221333422530] WebRTC resources freed; 0xb3c06338 0xb2500768
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Removing session 0xb2500f80 ...
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.125: Free socket sources for component 0xb25060f0.
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.125: Detaching source 0xb25092c8 (socket 0xb1103e58, FD 20) from context 0xb3c06410
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.126: Detaching source 0xb250afe8 (socket 0xb1103e10, FD 19) from context 0xb3c06410
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.126: Detaching source 0xb2509dc8 (socket 0xb1103dc8, FD 18) from context 0xb3c06410
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.127: Destroyed NiceComponent (5 created, 5 destroyed)
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:02.128: Destroyed NiceStream (5 created, 5 destroyed)
Aug 22 23:09:02 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink closed
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => Creating new session: 686792985926076; 0xb25008b8
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => Creating new handle in session 686792985926076: 5993682632653368; 0xb25008b8 0xb3c06708
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Creating session 0xb2511b08 ...
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [5993682632653368] Creating ICE agent (ICE Full mode, controlling)
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink opened; reading frames ...
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.640: Agent 0xb1107c48 : Property set, role is already "controlling".
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.640: Created NiceStream (6 created, 5 destroyed)
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Created NiceComponent (6 created, 5 destroyed)
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Agent 0xb1107c48 : allocating stream id 1 (0xb2505c28)
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Agent 0xb1107c48 : In ICE-FULL mode, starting candidate gathering.
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Agent 0xb1107c48 : libnice compiled without UPnP support
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Agent 0xb1107c48: Trying to create host candidate on port 37604
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Agent 0xb1107c48: Could not set IPV6 socket ToS: Protocol not available
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Component 0xb25060f0: Attach source (stream 1).
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.641: Attaching source 0xb250d320 (socket 0xb1103b70, FD 19) to context 0xb2511488
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Agent 0xb1107c48 : Adding new srv-rflx candidate discovery 0xb2512568
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Agent 0xb1107c48: Trying to create host candidate on port 39251
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Component 0xb25060f0: Attach source (stream 1).
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Attaching source 0xb2511998 (socket 0xb1103b28, FD 20) to context 0xb2511488
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Agent 0xb1107c48: Trying to create host candidate on port 26858
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Component 0xb25060f0: Attach source (stream 1).
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.642: Attaching source 0xb25092c8 (socket 0xb1103ae0, FD 21) to context 0xb2511488
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Agent 0xb1107c48 : discovery - scheduling cand type 1 addr 172.217.213.127.
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Agent 0xb1107c48 : stream 1 component 1 STATE-CHANGE disconnected -> gathering.
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Detach source 0xb25092c8, socket 0xb1103ae0.
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Detaching source 0xb25092c8 (socket 0xb1103ae0, FD 21) from context 0xb2511488
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Detach source 0xb2511998, socket 0xb1103b28.
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Detaching source 0xb2511998 (socket 0xb1103b28, FD 20) from context 0xb2511488
Aug 22 23:09:03 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => (process:21836): libnice-DEBUG: 23:09:03.643: Detach source 0xb250d320, socket 0xb1103b70.
[... cut ...]
Aug 22 23:09:08 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_traffic_handle:5077] [5993682632653368] ... only sent -1 bytes? (was 1216)
Aug 22 23:09:08 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_traffic_handle:5077] [5993682632653368] ... only sent -1 bytes? (was 1216)
Aug 22 23:09:08 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] [ice.c:janus_ice_outgoing_traffic_handle:5077] [5993682632653368] ... only sent -1 bytes? (was 1216)
Aug 22 23:09:08 kvmd kvmd-janus[6442]: kvmd.apps.janus.runner            INFO --- => [ERR] 
```

TelDragon, 2023-08-23:
And this kind of connection is completely disconnected. But the display resolution and rate traffic are 0

```
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.168: Detach source 0xaf10e498, socket 0xaf104860.
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.168: Detaching source 0xaf10e498 (socket 0xaf104860, FD 18) from context 0xaf116480
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Detach source 0xaf12ab50, socket 0xaf104818.
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Detaching source 0xaf12ab50 (socket 0xaf104818, FD 17) from context 0xaf116480
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Reattach source (nil).
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Detaching source (nil) (socket 0xaf104860, FD 18) from context (nil)
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Attaching source 0xaf12b150 (socket 0xaf104860, FD 18) to context 0xb3c06308
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Reattach source (nil).
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Detaching source (nil) (socket 0xaf104818, FD 17) from context (nil)
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.169: Attaching source 0xaf116368 (socket 0xaf104818, FD 17) to context 0xb3c06308
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.296: Agent 0xaf1082a8: inbound STUN packet for 1/1 (stream/component) from [172.217.211.127]:19302 (32 octets) :
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-stun-DEBUG: 09:26:14.297: STUN demux error: no cookie!
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-stun-DEBUG: 09:26:14.297: STUN unknown: 0 mandatory attribute(s)!
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-stun-DEBUG: 09:26:14.297: Received 32-bytes STUN message
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-stun-DEBUG: 09:26:14.297:  No XOR-MAPPED-ADDRESS: 1
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-stun-DEBUG: 09:26:14.297:  Mapped address found!
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.297: Agent 0xaf1082a8 : stun_bind_process/disc for 0xaf12b3f0 res 0.
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.297: agent_recv_message_unlocked: Valid STUN packet received.
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.312: Agent 0xaf1082a8 : Candidate gathering FINISHED, stopping discovery timer.
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.312: Agent 0xaf1082a8: gathered UDP local candidate : [10.9.174.16]:29973 for s1/c1. U/P '(null)'/'(null)'
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.312: Agent 0xaf1082a8: gathered UDP local candidate : [fe80::5453:8c5f:da16:1f7a]:24090 for s1/c1. U/P '(null)'/'(null)'
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.312: Agent 0xaf1082a8: gathered UDP local candidate : [112.224.149.157]:61046 for s1/c1. U/P '(null)'/'(null)'
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.516: Agent 0xaf1082a8: set_remote_candidates 1 1
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.516: Agent 0xaf1082a8 : Adding UDP remote candidate with addr [10.164.164.111]:47716 for s1/c1. U/P '(null)'/'(null)' prio: 2113937151
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.517: Agent 0xaf1082a8 : creating a new pair
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.517: Agent 0xaf1082a8 : pair 0xb1b06be8 state FROZEN (priv_add_new_check_pair)
Aug 23 09:26:14 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:14.517: Agent 0xaf1082a8 : new pair 0xb1b06be8 : [10.9.174.16]:29973 --> [10.164.164.111]:47716
[... cut ...]
Aug 23 09:26:38 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:38.488: Destroyed NiceComponent (4 created, 4 destroyed)
Aug 23 09:26:38 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => (process:4340): libnice-DEBUG: 09:26:38.488: Destroyed NiceStream (4 created, 4 destroyed)
Aug 23 09:26:38 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- Memsink closed
Aug 23 09:26:39 kvmd kvmd-janus[4335]: kvmd.apps.janus.runner            INFO --- => == ustreamer -- No active watchers, memsink disconnected

```

TelDragon, 2023-08-23:
Okay, through investigation, it should be due to the issue with libnice. Determine the version as v0.1.18 recommended

So, for Janus' P2P udp ports, many network operators basically block udp based peer-to-peer services. What are the better methods in this situation? For example, using TCP for P2P tunneling??

mdevaev, 2023-08-23:
I recommend libnice 0.1.21.

> What are the better methods in this situation? For example, using TCP for P2P tunneling??

You need to make TURN server in the local network for this and use janus static configuration.

TelDragon, 2023-08-23:
> I recommend libnice 0.1.21.
> 
> > What are the better methods in this situation? For example, using TCP for P2P tunneling??
> 
> You need to make TURN server in the local network for this and use janus static configuration.

Great and respected author, thank you for your reply.

I have tried using a newer release version of 0.1.21 and still encountered the same error.

Can it be that there is a Janus version of Libnice that is relative to it? ≥? The version and ustreamer are relatively matched. Because I am currently hesitant to try upgrading to a higher level, my kernel version is less than 5.15.32.

Additionally, I have deployed a TURN server on my server. The default is still a UDP based P2P tunnel. Here, it is emphasized that the type of P2P tunnel is UDP. Instead of the port number that connects to the TURN service. Do you have any knowledge about switching to TCP P2P? Because I cannot make any changes directly. Uncertain how janus static matches and changes relative to TCP when switched to, and uncertain about libjanus_ There is no need for linkage changes to related configurations such as ustreamer.

mdevaev, 2023-08-23:
> I have tried using a newer release version of 0.1.21 and still encountered the same error.

I'm following the simple rule: latest libnice, latest janus.

> Do you have any knowledge about switching to TCP P2P?

No, but I suggest to check janus documentation.

TelDragon, 2023-08-24:
> > I have tried using a newer release version of 0.1.21 and still encountered the same error.
> 
> I'm following the simple rule: latest libnice, latest janus.
> 
> > Do you have any knowledge about switching to TCP P2P?
> 
> No, but I suggest to check janus documentation.

Okay, thank you very much,

But Janus seems to have two release lines
[https://github.com/meetecho/janus-gateway/tags](https://github.com/meetecho/janus-gateway/tags)
1.2.0 0.14.0

Which one should I use?

mdevaev, 2023-08-24:
0.14

closed 2023-08-27 by mdevaev

jangerer-github, 2023-11-30:
Is there anywhere I can get a step-by-step process to solve for this. (e.g. installing a local TURN server) I cannot find any help on the pikvm.org site and this is the closest to an answer I have found as it describes what I am seeing in the logs exactly.

I just received my PiKVM mini V4 and I am have had this issue continually.

Any assistance would be appreciated.