pikvm/pikvm #1095: Download floppy drive icon missing/broken on macOS

reedy, 2023-08-24:
**Describe the bug**
Use of apparently not widely supported `🖪`/`U+1F5AA`/`BLACK HARD SHELL FLOPPY DISK Emoji`

**To Reproduce**
Steps to reproduce the behavior, like:
1. Click on Drive
2. Look at the `Download image` button

**Expected behavior**
Floppy Disk Emoji to actually appear

**Screenshots**
Chrome:
[image]
FF:
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.5
 - Browser Chrome/FF
 - Version 116.0.5845.96/116.0.3

reedy, 2023-08-24:
Same on iOS.

https://emojipedia.org/black-hard-shell-floppy-disk#technical apple maybe hasn't implemented/drawn it?

https://emojipedia.org/floppy-disk#technical might be a replacement...

reedy, 2023-08-24:
[image]

mdevaev, 2023-08-26:
I've replaced it to "arrow down"

closed 2023-08-26 by mdevaev