pikvm/pikvm #1097: Clipboard pastes immediately when focusing VNCViewer

roberestarkk, 2023-08-29:
**Describe the bug**
If the clipboard on the client system has updated since the VNCViewer was active, when focusing VNCViewer it will immediately send the contents of the client's clipboard as keypresses.

**To Reproduce**
1. Setup PiKVM with VNC Server running
2. Connect to VNC Server via viewer (I have tried with TigerVNC and TurboVNC) with 'Clipboard sharing' options enabled
3. Position your cursor somewhere text can be pasted (eg: Notepad) in the VNCViewer
4. Copy something from the client machine
5. Refocus the VNCViewer
6. Observe the immediate pasting of whatever was copied, unsolicited by user input

**Expected behavior**
Nothing should happen unless/until I trigger a paste while the VNCViewer is focused, at which point it should paste the copied text.

**~~Screenshots~~Videos**
[Simple Example (unmute for Commentary)](https://imgur.com/diRVqu7)
[Longer Example (unmute for Commentary](https://imgur.com/FAyBVec)

**Client Desktop (please complete the following information):**
 - OS: Win10
 - Browser N/A
 - Version N/A
 - VNC client: TigerVNCViewer
 - VNC client version: 1.13.1

**KVM'd Desktop (extra info just in case it's relevant):**
 - OS: Win11
 - Browser N/A
 - Version N/A

**PiKVM info:**
  - Raspberry Pi board version: Rpi3
  - PiKVM platform: Not sure, I don't see anything like that mentioned anywhere
  - Video capture type: HDMI?
  - KVMD version: `3.238-1`
  - uStreamer version: `5.41-1`
  - Linux kernel: `Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux`

**Additional context**
When I tested with TurboVNC Viewer to try to isolate where the fault is, it didn't auto-paste on re-focus, but instead provided me with a pop-up Clipboard Manager built into the viewer. It did keep the VNCViewer Clipboard in sync with my client's clipboard, but it would immediately start typing the keys if I hit the 'Send Paste' via the manager, via the F8 menu for TurboVNC. So I assume PiKVM VNCServer is still effectively auto-pasting as soon as it detects clipboard input from the VNCClient.

I first raised this first as an issue on [TigerVNC's GitHub](https://github.com/TigerVNC/tigervnc/issues/1661), as it seemed like perhaps the Viewer would need to handle the clipboard like TurboVNC does, assuming PiKVM's VNCServer might not have a way to detect a clipboard update through the viewer was different to actual user input. However they are indicating that VNCViewers should be able to sync the clipboard with the VNCServer, rather than immediately sending the paste.

I had a look through the issues here, and in the documentation, to see if there were any mention of 'paste' or 'clipboard' which looked related to my issue, but didn't find anything.

I also did mention clipboard sync from the server to the client didn't work either in the TigerVNC issue, but I've since found the 'OCR' part of PiKVM that might probably possibly address that part, and that direction is not really a big deal to me anyway.

mdevaev, 2023-08-31:
Hello. I'll test it and find out what's wrong with this.

mdevaev, 2023-10-10:
I got my hands on it. Alas, there is no technical way to fix this, this is a limitation of the fact that we don't have access to the clipboard.

roberestarkk, 2023-10-10:
That's a shame, but understandable. 
Is there at least a config option or something to prevent it from sending clipboard updates on the Pi as keypresses to the locally connected device?

I'm thinking I can finagle an alternate way to sync the clipboard between the Pi and the 'local' machine attached to it over LAN. 
Then when I VNC into the Pi and VNC syncs the clipboard between Remote and Pi as normal, it won't be an issue, and the alternate sync between Pi and local will take care of the remaining hop.

This kind of setup might even enable it to sync in reverse as well, which would be nice, depending on how the VNC Server is set I assume.

mdevaev, 2023-10-10:
I can make option to disable clipboard completely. Will it fit?

roberestarkk, 2023-10-10:
From the word 'entirely' I assume not.

As long as a VNC Viewer connecting to the Pi can still sync the clipboard of the VNCViewer's machine with the clipboard of the Pi, and the Pi's clipboard updating doesn't trigger the Pi to auto-type the clipboard into the hard-wired connected machine, that will be great. 
If not, that won't really help unfortunately.

mdevaev, 2023-10-10:
I'm confused. I can make it so that the clipboard change event will be ignored. This means that you will not be able to insert text via VNC. Any other third-party tools will work.

roberestarkk, 2023-10-10:
My understanding of how everything ties together is as follows:
[image]

What I'd like to have happen instead, is this:
[image]

The key difference being in the third step of stage 2.
Everything prior to that would still work and update things as expected, PiKVM just wouldn't try to communicate the clipboard changes across USB to the connected machine via emulated keypresses anymore.

mdevaev, 2023-10-10:
Then this is what I meant, disabling the clipboard on PiKVM server.

roberestarkk, 2023-10-10:
Which of the following options would best reflect what impact that would have on the current process?

A:
[image]

Or B:
[image]

At the moment it sounds like option B, which is basically how it already works if I disable the sync in the VNCViewer options. 
So option B would not be that helpful in PiKVM as it's already a VNCViewer setting, while Option A would be very helpful as only really PiKVM can control whether the clipboard auto-typing happens or not.

mdevaev, 2023-10-10:
It will be like A.

roberestarkk, 2023-10-10:
Ah wonderful, that sounds great!

That should let me sync "Remote" clipboard with "Pi" clipboard via VNC, and then sync "Pi" clipboard with "Local" clipboard via something else like Uniclip, which would mean full end-to-end bi-directional clipboard sync 😃

mdevaev, 2023-10-10:
There is no clipboard inside the Pi, at all. The server receives the clipboard change event and simply discards it.

roberestarkk, 2023-10-10:
It's built off Linux though, so it surely must have a system clipboard built in?

mdevaev, 2023-10-10:
There is no system clipboard in Linux. This is part of X11 or wayland, which is not in PiKVM. If you need to save the clipboard content somewhere, then a temporary file on /tmp in PiKVM.

roberestarkk, 2023-10-10:
Oh right, fair enough, I just figured something like that'd be a prerequisite for the VNCServer.

So if the clipboard entry has nowhere to go on the PiKVM other than the VNCServer's internal implementation of a clipboard, and that is disabled, then effectively what'll happen is closer to Option B right? 
The Pi won't do anything with clipboard updates that get pushed to it by the VNCViewer?

mdevaev, 2023-10-10:
Yes.

roberestarkk, 2023-10-10:
Rightio, well we can already do that with the toggle on the VNCViewer, so no need for that in PiKVM I don't think, all good!
Thanks anyway :)

mdevaev, 2023-10-10:
Okay. Then I'm closing this.

qwertier, 2024-07-13:
Alright, from what I've read here, it looks like we cannot use fully-featured VNC to access remote hosts? Clipboard capabilities is a part of VNC protocol, and I'm frustrated that it's not available for the remote system control (((( Of course, we can't talk about proper remote control without clipboard feature!

@mdevaev is there any plans to implement this feature in the future? From what @roberestarkk described, it seems the code is already in there, if auto-paste takes place, when VNC server tries to sync clipboard buffers. We just need to store the clipboard contents somewhere and paste them as key types on receiving the according event from VNC client. That's how it should be, I think. Again, it's obvious from the current ticket's description, that most of the code is present. X11 buffer may not be too helpful, as we still need to pass the received data from PiKVM to the host! It would be helpful if VNC server was on the receiving host, which is not our case!

@roberestarkk I found a workaround of how to paste local machine's clipboard contents, when the remote system doesn't provide clipboard capabilities. On Linux, you can write the script like the following:
`xclip -selection c -o | xargs -0 -I '{}' xdotool type "{}"`
... and then assign a shortcut on it like Ctrl+Alt+V.
The script above will get the local machine's clipboard contents as a text and type it by generating key press sequence. Of course, this is a workaround and it suffers from different issues, like sporadically "eating" some letters at the beginning and so forth. But it's better, than nothing...

roberestarkk, 2024-07-13:
Thanks, but I am using Windows as both my Local and Remote machine, only the PiKVM itself is Linux (I assume with a reference to xdotool that your script needs to be run in the context of an x server, and thus on the local and not the PiKVM itself?).

My eventual solution was to disable clipboard sharing entirely in the VNC Client, and create a KeePass entry to perform an auto-type of my clipboard at the Local machine level.
Which is effectively the much more irritatingly roundabout Windows-y version of your suggestion anyway lol.

qwertier, 2024-07-14:
@roberestarkk  Yes, the provided script is for Linux/Unix and must be run on Linux local machine (the one, from which you are connecting to PiKVM). It was just an idea of how to avoid clipboard limitations:
disable clipboard management in VNC and write the script, that will translate clipboard contents of the local machine into the series of key types.

qwertier, 2024-07-14:
> @mdevaev is there any plans to implement this feature in the future? From what @roberestarkk described, it seems the code is already in there, if auto-paste takes place, when VNC server tries to [...]

@mdevaev This is more specific "to-do" plan of how to get clipboard feature in PiKVM, when using VNC connection...

From the description of the current ticket, it's evident, that we already have: 
- Subscription on VNC client event to sync clipboards of VNC client on local machine and PiKVM
- Translation of received clipboard contents to the series of key presses

What we need to do to get clipboard functionality for VNC connection, that allows to copy data from the local to the remote machine:

1. On clipboard contents sync stage, do not trigger translation clipboard => key presses, but just store these contents into a temporary file, like /timp/clipboard.
2. Create a subscription on "Ctrl-V" event from the local machine. When received, just check for the tmp file presense, and if it exists, get the file contents and execute already existing translation from text to the series of key types.
3. Remove the tmp file to avoid potential issues with event duplications.

I think, it's not that hard to implement all that. In any case, this feature just needs reasonable code changes and extensions, not writing everything from scratch!

tachang, 2025-05-08:
> In the next release, the hotkey `LeftAlt,LeftAlt,P` will be used for pasting so the described problem will gone.

That's awesome to hear. That means pressing each key in succession right? Not holding them down.

mdevaev, 2025-05-08:
Yes

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev