pikvm/pikvm #1098: HP Elite Mini 800 G9 - not entering startup menu with PiKVM

bogd, 2023-08-30:
**Describe the bug**
The unit does not enter the startup menu when connected to PiKVM. If I connect a "normal" monitor via HDMI, everything works fine.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Power on the unit
2. Press Esc to enter startup menu
3. The unit hangs at "Entering startup menu..."

If I repeat the steps using an external monitor (instead of PiKVM), the unit enters startup menu without any issues.

**Expected behavior**
The startup menu should be displayed, allowing for firmware updates, boot menu, etc.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 Windows 10
 - Browser [e.g. chrome, safari]
 Chrome
 - Version [e.g. 22]
115.0.5790.171
 - VNC client (if used)
N/A

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
 RPi 4
  - PiKVM platform [e.g. v2-hdmi]
 v3
  - Video capture type [e.g. CSI bridge]
 CSI
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.252-1
kvmd-fan 0.26-1
kvmd-oled 0.25-1
kvmd-platform-v3-hdmi-rpi4 3.252-1
kvmd-webterm 0.43-1

  - uStreamer version: `pacman -Q | grep ustreamer`
 ustreamer 5.42-1

  - Linux kernel: `uname -a`
Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**

Initially, I had problems using the PiKVM virtual keyboard before boot (even the Esc keypress was not recognized). I fixed it by disabling mass storage device in PiKVM (which kinda makes it impossible to use PiKVM to install the OS on the unit? But that's another problem, for another issue). Now I can press Esc on PiKVM, I see the "Entering startup menu..." message, but the HP unit hangs there.

The same issue persists even when I connect an external keyboard directly to the HP unit's USB port, and trigger the startup menu by pressing Esc on the physical keyboard - unit still hangs.

If I connect the physical keyboard and external monitor directly to the HP unit, the unit enters boot menu without any issues.

mdevaev, 2023-08-31:
Hello. Do you mean that you don't see the BIOS or there is a hang? What happens if you connect a physical monitor during a potential hangup?

bogd, 2023-09-01:
Unclear which one it is. As far as I can tell, the HP unit hangs - however, it makes all the "normal" noises, so I am not really sure if it actually hangs or it simply is unable to display the screen.

If I switch to a physical monitor during the hangup, all I get is a black screen.

The strangest thing is that no other connection (other than HDMI) is moved - I have USB connected to both PiKVM and an external keyboard, and those stay in place. I just switch the HDMI output - connect it to a "normal" monitor, everything is fine; connect it to PiKVM, the screen hangs.

I have also reproduced the problem on an HP Elite 600 G9 mini (not really surprising, since the hardware is all but identical).

mdevaev, 2023-09-01:
Try some alternative edids for display emulation: https://docs.pikvm.org/edid/

mdevaev, 2023-09-02:
> Before I close this, there is still the problem of "keyboard is ignored before boot, unless I disable mass storage device". Which means that I cannot really use PiKVM to mount an OS installation ISO and install the OS remotely. To avoid opening another issue for this - is there a workaround I could use in this case? A way to get the keyboard working, AND MSD support? Can you point me in the right direction?

I believe that the reason for the problem is that the BIOS on HP wants something strange that is missing in the Linux kernel modules on PiKVM. I could fix this, since I have a USB protocol analyzer, but there is not a single sample of devices with a similar problem.

As a temporary solution, you can use a [dynamic USB configuration](https://docs.pikvm.org/usb_dynamic). This is a bit unsafe for KVM (described in detail in the documentation), but it works. Even if a kernel panic occurs in V3, the device will simply reboot.

bogd, 2023-09-07:
Closing as "should have read the documentation, but in my defense I had no idea that EDID even existed, let alone that it could be changed... :) " . Thank you once again for all your help!!

closed 2023-09-07 by bogd

mdevaev, 2023-09-07:
Ur welcome :)