pikvm/pikvm #1100: PiKVMv4 Plus: I2C device not found on address: 0x3C

donkerm, 2023-09-01:
Received a new PiKVMv4 Plus and oled doesn't work.
When starting the servers I receive the following errors:

```
Sep 01 14:50:55 pikvm systemd[1]: Starting PiKVM - A small OLED daemon...
Sep 01 14:50:55 pikvm kvmd-oled[2175]: Traceback (most recent call last):
Sep 01 14:50:55 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/luma/core/interface/serial.py", line 96, in command
Sep 01 14:50:55 pikvm kvmd-oled[2175]:     self._bus.write_i2c_block_data(self._addr, self._cmd_mode,
Sep 01 14:50:55 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/smbus2/smbus2.py", line 643, in write_i2c_block_data
Sep 01 14:50:55 pikvm kvmd-oled[2175]:     ioctl(self.fd, I2C_SMBUS, msg)
Sep 01 14:50:55 pikvm kvmd-oled[2175]: OSError: [Errno 121] Remote I/O error
Sep 01 14:50:55 pikvm kvmd-oled[2175]: During handling of the above exception, another exception occurred:
Sep 01 14:50:55 pikvm kvmd-oled[2175]: Traceback (most recent call last):
Sep 01 14:50:55 pikvm kvmd-oled[2175]:   File "/usr/bin/kvmd-oled", line 232, in <module>
Sep 01 14:50:55 pikvm kvmd-oled[2175]:     main()
Sep 01 14:50:55 pikvm kvmd-oled[2175]:   File "/usr/bin/kvmd-oled", line 170, in main
Sep 01 14:50:55 pikvm kvmd-oled[2175]:     device = luma_cmdline.create_device(options)
Sep 01 14:50:55 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/luma/core/cmdline.py", line 246, in create_device
Sep 01 14:50:56 pikvm kvmd-oled[2175]:     device = Device(serial_interface=interface(), **params)
Sep 01 14:50:56 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/luma/oled/device/__init__.py", line 176, in __init__
Sep 01 14:50:56 pikvm kvmd-oled[2175]:     self.command(
Sep 01 14:50:56 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 48, in command
Sep 01 14:50:56 pikvm kvmd-oled[2175]:     self._serial_interface.command(*cmd)
Sep 01 14:50:56 pikvm kvmd-oled[2175]:   File "/usr/lib/python3.10/site-packages/luma/core/interface/serial.py", line 101, in command
Sep 01 14:50:56 pikvm kvmd-oled[2175]:     raise luma.core.error.DeviceNotFoundError(
Sep 01 14:50:56 pikvm kvmd-oled[2175]: luma.core.error.DeviceNotFoundError: I2C device not found on address: 0x3C
Sep 01 14:50:56 pikvm systemd[1]: kvmd-oled.service: Control process exited, code=exited, status=1/FAILURE
Sep 01 14:50:56 pikvm systemd[1]: kvmd-oled.service: Failed with result 'exit-code'.
Sep 01 14:50:56 pikvm systemd[1]: Failed to start PiKVM - A small OLED daemon.
```
```
[root@pikvm etc]#  pacman -Q | grep kvmd
kvmd 3.199-1
kvmd-fan 0.25-1
kvmd-oled 0.19-1
kvmd-platform-v4plus-hdmi-rpi4 3.199-1
kvmd-webterm 0.42-1
```
```
[root@pikvm etc]# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --         
```
Any idea what the problem can be?

mdevaev, 2023-09-01:
Hello. It looks like your device has a faulty screen. Is this a kickstarter device?

donkerm, 2023-09-02:
This is a unit I bought from https://thepihut.com/products/pikvm-v4-plus
How can you tell it's a faulty screen? just wondering and thank you for the support.

mdevaev, 2023-09-02:
If it is not detected by I2C, then it is broken :) Because we have other working components on I2C.

As far as I understand, you have already sent an exchange request, right?

donkerm, 2023-09-04:
I'm mailing with thepihut to see if this is possible.. Could you tell me how a correct output of I2C looks like?
The UU made me believe it detected something.
Thank you for the feedback and help. If I know how a correct output looks like I will close this issue.
Great support!

donkerm, 2023-09-04:
I contacted the vendor and we are going to swap.
Thank you for the support!

closed 2023-09-04 by donkerm

mdevaev, 2023-09-04:
Thank you for buying or product!