pikvm/pikvm #1102: Distorted KVM text display

BackSeat, 2023-09-03:
PiKVM + ezCoo connecting to three servers, two of them Linux servers without X installed (so just a text mode console). Both Linux servers running Debian 12 (Bookworm). One of those servers displays fine; the other has a a very distorted display (see attached). That server is an HP ProLiant MicroServer Gen 10.

Any suggestions for getting this text mode display working correctly?
[image]

mdevaev, 2023-09-07:
Hello. Probably the server does not respect EDID. Try to change it: https://docs.pikvm.org/edid

BackSeat, 2023-09-08:
Thanks for the suggestion. That wasn't the fix, but [this was](https://unix.stackexchange.com/questions/583585/setting-the-most-basic-text-console-video-mode-on-linux).

Appreciate your help.

closed 2023-09-08 by BackSeat

mdevaev, 2023-09-08:
:ok_hand: