pikvm/pikvm #1104: Add "Changes you made will not be saved" to Term and Text states when "Ask page close confirmation" is off.

adamoutler, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I try to close PiKVM Window and a "Changes you made will not be saved" popup occurs. It causes all chrome windows to lock out until Confirmed/Cancelled.  Disabling this option removes it's usefulness for the text menu.

*To reproduce*
Scenario 1.
1. open KVMD with "Ask page close confirmation" turned on
2. Open terminal window
3. press reload
4. Observe popup - Expected behavior

*Scenario 2*
1. open KVMD with "Ask page close confirmation" turned on
2. Type text into Text Menu and do not paste
3. press reload
4. Observe popup - Expected behavior

Scenario 3. 
1. open KVMD with "Ask page close confirmation" turned off
2. Open terminal window
3. press reload
4. Observe popup - Expected behavior because data loss is possible

*Scenario 4*
1. open KVMD with "Ask page close confirmation" turned on
2. Type text into Text Menu and do not paste
3. press reload
4. Observe no popup - Unexpected behavior as data loss is possible

Scenario 4 demonstrates the problem. The only true form data on the page is in the Text menu and the Term.   All other settings persist through multiple instances due to the use of immediate Local Storage within the browser.  Since all other settings will persist,  the Text and Term are the only items which should always trigger popup.  

Turning off the "Ask page close confirmation" still allows page close confirmation from Term, but it does not from Text Menu.  
[image]

The "Changes you made will not be saved" should be unified with the on/off states of "Ask page close confirmation" for both Term and Text menu. Both of these items should act the same.  Attempting to close the page with any non-persistent item present should lead to a popup.
[image]

**Describe the solution you'd like**

When "Ask page close confirmation" is off, if text was entered into Text menu, but not pasted then a popup should be displayed

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
- [ ] Turn off "Ask page close confirmation". [image]  This does not account for the Text Menu data. This is technically the only form on the page and should be covered via confirmation.
- [ ] Limit the "Ask page close confirmation" to only the Term and Text menu states.  Since this is a default behavior, it may be a part of some people's workflows and would require review before modification.  Due to this being a default setting, I feel this change is more risky.
- [x] Add Text menu confirmation when "Ask page close confirmation" is turned off. This seems to make the most sense because the Term and the Text menu are non-persistent data and the only things which could be lost if the page were closed/reloaded. This would make the behavior of opening the term or typing in the text menu without pasting act the same way.

**Additional context**
This popup locks out all other chrome windows until it is dismissed, therefore it is important to have proper and expected behavior. 
[image]

mdevaev, 2023-12-03:
It seems to me that the close confirmation should act on the entire page, regardless of the terminal and the input text. If the user disables this setting, he clearly does not want any confirmations. After I disabled confirmation for the terminal, we only have a simple consistent behavior left: either popup is always shown, or never at all if it is turned off. I wouldn't add any implicit logic here.

mdevaev, 2023-12-03:
In fact, first of all, this is protection against accidentally pressing Ctrl+W, because we cannot intercept this combination in the browser. I don't think there is any point in protecting the text input field, because it is unlikely to contain data that you are afraid of losing, because you copied it from somewhere here.

mdevaev, 2024-04-07:
I returned a warning about closing the terminal when it is running in full-screen mode (the button from the menu after login). To close the terminal in KVM mode, just use the "X" button. The terminal will also be closed along with the tab after one warning. As for the insert warning, I still adhere to the position described earlier. Right now it's the best I can offer.

closed 2024-04-07 by mdevaev