pikvm/pikvm #1106: No UEFI/boot output on the 10th gen Intel NUC with PiKVM v3 DIY

GentleHoneyLover, 2023-09-07:
**Describe the bug**
I just assembled and setup my PiKVM v3 DIY kit and hooked it up to my Intel NUC ([NUC10i5FNKN](https://www.intel.com/content/www/us/en/products/sku/214595/intel-nuc-10-performance-kit-nuc10i5fnkn/specifications.html)) which I use as a headless Linux server. Problem is I'm getting no video on PiKVM before the OS is booted (no UEFI, no booting process). The first screen that appears is this one:

[image]<br><br>

I followed [this guide](https://docs.pikvm.org/edid/) and tested a few EDID options — no result. Do I need some special EDID here? Or do I need to change other settings somewhere? Any help is highly appreciated!<br><br>

**To Reproduce**
Steps to reproduce the behavior: Reboot the Intel NUC with PiKVM connected

**Expected behavior**
When Intel NUC starts you should be able to have video output straightaway, be able to enter and use UEFI and see the consequent boot process (GRUB, kernel, etc)

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.5.1
 - Browser: Safari
 - Version: 16.6

**PiKVM info:**
  - Raspberry Pi board version: Raspberry Pi 4 Model B Rev 1.5
  - PiKVM platform: v3 HAT DIY kit
  - Video capture type: not sure — whatever comes with the kit
  - KVMD version: 
```
kvmd 3.252-1
kvmd-fan 0.26-1
kvmd-oled 0.25-1
kvmd-platform-v3-hdmi-rpi4 3.252-1
kvmd-webterm 0.43-1
```
  - uStreamer version: `ustreamer 5.42-1`
  - Linux kernel: `Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux`

closed 2023-09-07 by GentleHoneyLover

GentleHoneyLover, 2023-09-07:
Sorry — stupid me. Was too quick with logging the issue — it had nothing to do with PiKVM or the NUC :)

Everything works as expected. Thanks for the great project!

mdevaev, 2023-09-07:
:ok_hand:  :)