pikvm/pikvm #1107: Unable to see / access BIOS / UEFI

bigjohns97, 2023-09-16:
After switching to an intel a380 video card I am unable to access the BIOS / UEFI

**To Reproduce**
Reboot the PC connected to the PoKVM and don't see a video signal until the Windows login screen

**Expected behavior**
Expect to see a BIOS boot image with option to enter the BIOS and make changes

**Screenshots**
Not needed IMO

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 116
 - VNC client: not used

**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.5
  - hdmi
  - Intel A380
  - KVMD version: 3.121-1
  - uStreamer version: 5.16-1
  - Linux kernel: Linux pikvm 5.15.34-4-rpi-ARCH #1 SMP Tue Jul 5 12:10:16 MSK 2022 armv7l GNU/Linux

**Additional context**
This was working previously with a nvidia 1660

bigjohns97, 2023-09-16:
Odd, I just ran the update procedure and now the WebUI is telling me that the page requires javascript and I need to disable my ad blocker...

bigjohns97, 2023-09-16:
Added the PiKVM back to the block internet rule on my router and rebooted the device and now I can't access the webUI at all, is internet connectivity required for the latest version?

bigjohns97, 2023-09-16:
Intel A380 does require resizable bar enabled as well as using the DP connection for the first attempt to enable resizable bar.

https://www.reddit.com/r/IntelArc/comments/z2t0rs/a380_giving_black_screen_until_loaded_in_windows/

https://www.reddit.com/r/IntelArc/comments/16j5ea8/no_display_output_without_resizeable_bar_enabled/

Either enable this before installing your card or get a DP to HDMI cable like I had to do.

closed 2023-09-16 by bigjohns97

mdevaev, 2023-10-03:
Good to know, thank you.