pikvm/pikvm #1109: pikvm v4 mini not turning on (possible hw problem)

RootLUG, 2023-09-17:
Hello,
I've bought 2 identical pikvm v4 minis recently and I've left one of them running connected to the laptop hdmi & usb to provide me with a remote access while traveling. Recently I discovered I can no longer connect and when I physically came to the pikvm it was "powered on" in a sense the RED led light is on + the light next to the HDMI port, screen was however off as well as green light next to the 5V power input, the power draw was also 0W.

I tried to reboot the pikvm by reconnecting the power but it now never starts up, the red light led next to 5V input is on, green light next to eth port is on as well and also the green light next to hdmi is on. The "system" green light on the left side of the 5V never turns on, even after trying several reconnects to power.

Since I have 2 kvms I tried to swap the sd cards and the OS on there is OK and boots fine in the other pikvm, and the same behaviour is when I insert a known good SD card from the other kvm it also doesn't boot. From that point of view I think I can rule out it being some sort of SD/OS or configuration problem.

When I compare the startup sequence with a known good pikvm, on power insert the RED led next to 5V input turns on, it will "blink" (briefly turn off) and then it will light also the green led light on the other left side of 5V input. The "blinking" of red led also happens on the malfunctioning pikvm but after it's turned back on the green led won't turn as well which I think indicates the OS has not started to boot.

Is there any way how I can debug further or perform a "hw" reset? I read the new pikvm4 contains a supercapacitor, is it possible  that it keeps interfering or something set on where I need to discharge it? Any other recommendations I can try? The pikvm was powered on for roughly 3 days when I noticed it's no longer responding.

mdevaev, 2023-09-19:
Hello. It looks like you got a defective CM4. This happens occasionally. Where did you buy the device? You can get a replacement.

Amt921, 2023-09-19:
Hello! @RootLUG 

What WiFi settings do you use? Exactly channel number and bandwidth?

RootLUG, 2023-09-20:
@Amt921 The wifi settings are default ones generated by wpa_supplicant (just ssid + psk) + the only additional thing is that I configured 2 networks, "wifi1" with priority 2 and "wifi2" with priority 3. The "wifi2" wasn't used as I configured it only as a backup network I use when traveling.
The "wifi1" network is b/g/n/ax on channel 11 with 40MHz bandwith and -25 dBm power at the place where pikvm was.

The network is also technically available in 5 GHz as well (channel 100, 80Mhz) but I don't think cm4 has 5ghz connectivity. Router hardware is Mikrotik hAP ax2.

Amt921, 2023-09-25:
Hello @RootLUG !
Sorry for delay. It seems you got some hardware defect, we will replace it with new.
Please write email to support@pishop.us and mention issue github #1109, support will guide you with replace procedure.
We got some ideas in work, just in case - in further please use external antennas with wifi.

closed 2023-09-27 by mdevaev