pikvm/pikvm #1110: zImage kernel linux-rpi-pikvm  . gzip to lz4

cgm999, 2023-09-18:
This is for package linux-rpi-pikvm . I think if zImage uses lz4 instead of gzip it will speedup booting .

On my pi2zero here are some times for read+decompress:
**
zcat /boot/kernel7.img.gz.9 : real    0m1.557s user    0m1.310s sys     0m0.039s

lz4cat /boot/kernel7.img.lz4.best: real    0m0.775s user    0m0.283s sys     0m0.062s
**

Size will increase a little for /boot/kernel7.img , in my case from 8.7M to 11M

mdevaev, 2023-09-18:
I adhere to a policy of minimal interference with Arch kernel packages. I'm sorry, but one second and a couple of megabytes is not a very significant gain for a device that should rarely reboot.

cgm999, 2023-09-18:
Well, I understand the ideea with minimal change from upstream. However not pikvms are ment to be online all the time :)
Anyway I will close this for now.
Thank you for making this amazing project opensource btw !

closed 2023-09-18 by cgm999

mdevaev, 2023-09-18:
Thank you! :ok_hand:

cgm999, 2023-09-20:
Adding a little more details for whoever will read this ,  GZIP vs LZ4 for kernel zImage only gains 0.5sec on rpi2z.