pikvm/pikvm #1111: No UEFI/boot output on the 7th gen Intel NUC with PiKVM v3

n3gwg, 2023-09-19:
Developers, et alia:

NUC7i7DNHE and PiKVM V3 (fully assembled version)

I am unable to see video when the UEFI/BIOS is control of the system, though can see video succeeding a proper booting of Linux (ProxMox VE Version 8).

In succession to doing much research I realized it is a problem with the EDID in PiKVM. I have tried some several different EDID files with varying results. Some are to high of a resolution to be readable some are perfect in resolution but flicker into oblivion.

I obtained an EDID editor (for my Mac) from the website as follows hereupon:
https://www.analogway.com/americas/products/software-tools/aw-edid-editor/

I downloaded this repo with tons of EDID configurations!
https://github.com/linuxhw/EDID

I have a call into Intel about the issue so I can try to ascertain the totality of graphical modes the Intel NUC model I have (NUC7i7DNHE) will expel via its HDMI port. I am also going to try to obtain an EDID emulator.

If someone can explain what are the parameters I can tune and tweak in an EDID file that matter for solving this issue I would gladly try to create an EDID file that would work for myself.

I also had some chats with several people on the Discord channel as well.

Thank you in advance for your time and consideration regarding the instant matter before us.

Stuart

mdevaev, 2023-10-03:
Hello. Sorry for the late responce. If it is not solved - did you try our special edids from that page? https://docs.pikvm.org/edid

n3gwg, 2023-10-04:
Maxim,

No I have not yet solved it and yes I tried some of the EDIDs in that link. People keep saying buy the V4, but I already have this and I have zero appetite for re-buying the device when I can't even get reasonable help. The people in the support channel are nice but they are all guessing at solutions as far as I can tell and do not really understand the EDIDs. They basically just hand out links and cannot troubleshoot much.

I purchased a Gefen Detective Plus HDMI box, but that has not helped.
I also opened a ticket with Intel, but the people in support have gotten much worse lately and only write down questions and ask someone else and call or email back, so there is no way to have a real technical discussion with an engineer.

I have several NUCs (4 of them) and an EVCOO switch (which I am happy to subtract out of the equation for now until we get this figured out).

As far as I understand it, I need to ask Intel what resolution and frequency the NUC uses prior to starting the BIOS? That is hard, these people at Intel are outside the US and just write stuff down and call someone else. I am suffering and about to throw this thing at the wall.

Thank you so much!
Stuart

mdevaev, 2023-10-05:
Personally I hate NUCs because of all these BIOS issues because it absolutely ignores the HDMI specifications. It ignores the EDID requirements and sends whatever signal it wants.

We did V3 when we didn't know how deep the rabbit hole was yet, and tried to fix it in V4. The difference between them is that V4 supports the 60Hz mode natively, while V3 can't due to Raspberry hardware limitations.

Unfortunately, I'm not much better than other tech support in the sense that this is really a hardware problem, and no matter how much I want to, I can't solve it with some software patches, sorry :(

I asked @arch1mede  to find a downscaler for you that might work.

arch1mede, 2023-10-05:
Hmm I think there needs to be some update to this git issue, the cx came onto discord, realized their device wasn't programmed [correctly](https://discord.com/channels/580094191938437144/896216359963332628/1158844746685489214)

This is a known issue with these NUC's, they enforce a certain resolution + hz and ignore anything else. I really hope that Intel will release a BIOS that can relax this enforcements but I would not be surprised if they did not answer this request.

Here are some options:
1x v2 with a usb dongle
This [downscaler](https://www.amazon.com/Splitter-Multi-Resolution-J-Tech-Digital-downscale/dp/B0839PRQ2V) will work. Also as an alternative, if you purchased the v3 from pishop within the warranty period, and you are willing to accept the restocking fee, you can get this exchanged for a v4 which will for sure work.

mdevaev, 2023-10-10:
Probably I can close it now. Let me know if it's not solved.

closed 2023-10-10 by mdevaev

n3gwg, 2023-10-11:
Maxim,

With the Jewish holidays this weekend and then the war in Israel this issue got away from my attention. It is not yet resolved, I will try to contact PiShop tomorrow to see if my device in still in warranty, likely not. I did not really realize this was a warrantee issue or I'd have spoken up much sooner. I just presumed I was configuring it incorrectly.

A downscaler was recommended (I will order it), but I am wondering what specifications specifically should I look for in a downscaler? There are other downscalers too and I wanted to know what specs to look for if I wanted a different model?

Thank you so very much!

Stuart

mdevaev, 2023-10-12:
It's a DIY device with USB dongle capture.

bamaas, 2024-01-04:
Any updates on this?

n3gwg, 2024-07-07:
Arch1mede, Maxim, et alia:

It has been some time since I have thought about the PiKVM. Recently, I updated the software and am immersing myself back into trying to use it again.

[...clipped...]

> 
> This is a known issue with these NUC's, they enforce a certain resolution + hz and ignore anything else. I really hope that Intel will release a BIOS that can relax this enforcements but I would not be surprised if they did not answer this request.

I am curious, exactly what is the "certain resolution + Hz" that you refer to? Do we not know what it is? If we do, knowing that might be helpful to trying to resolve this issue.

ASUS has purchased the NUC line from Intel, so maybe they will be more forthcoming with technical information or be updating the BIOS in new ways that Intel had not.

> 
> Here are some options: 1x v2 with a usb dongle This [downscaler](https://www.amazon.com/Splitter-Multi-Resolution-J-Tech-Digital-downscale/dp/B0839PRQ2V) will work. As an alternative, if you purchased the v3 from pishop within the warranty period, and you are willing to accept the restocking fee, you can get this exchanged for a v4 which will for sure work.

Clicking the above entitled link now stands demonstrative of several different models. I presume it is the 4K 1x2 4:4:4 (with a "downscale on/off" slide switch) that you are implying could solve the issue for me?

Do any of the 4-port switches also provide any such downscaling capabilities?

I had spoken to someone at PiShop.US (Steve?) back then (when it was within the warranty period) about doing that but then had gotten busy traveling for work and did not get to it in time.

> This of course assumes the purchase of official pikvm hardware, if this is related to "other" hardware aka clones, then you should seek tech support from them.

Yes, it was an official PiKVM purchased from PiShop.US

Thanks again to everyone for their assistance and for creating such a great project.

PS - Any chance a version of PiKVM will come out based on Debian at any point?

arch1mede, 2024-07-07:
> I am curious, exactly what is the "certain resolution + Hz" that you refer to? Do we not know what it is? If we do, knowing that might be helpful to trying to resolve this issue.

1080P@60hz

> Clicking the above entitled link now stands demonstrative of several different models. I presume it is the 4K 1x2 4:4:4 (with a "downscale on/off" slide switch) that you are implying could solve the issue for me?

Its possible, a v4 would be better though.

> Do any of the 4-port switches also provide any such downscaling capabilities?

No

> PS - Any chance a version of PiKVM will come out based on Debian at any point?

Yes but no ETA, why do you need a Debian based PiKVM?

n3gwg, 2024-07-07:
Some of my comments herein below were subject to a discord chat we had, but I feel it is worthy of notation for future readers to post some of them ideas herein below.

> > Clicking the above entitled link now stands demonstrative of several different models. I presume it is the 4K 1x2 4:4:4 (with a "downscale on/off" slide switch) that you are implying could solve the issue for me?
> 
> Its possible, a v4 would be better though.

I realize that you cannot speak for the manufacturer of any downscaler product nor do you follow any manufacturers products regularly to know of changes in specifications. The foregoing notwithstanding, I now understand that I need a downscaler that will go from 1080P@60Hz to 1080P@50Hz.

I am going to contact the manufacturer of the product/model you originally linked to and find out if it might be vaiable.

> > PS - Any chance a version of PiKVM will come out based on Debian at any point?
> 
> Yes but no ETA, why do you need a Debian based PiKVM?

The reason that people do not want Arch is because it is a niche Linux that most folks using Linux are unfamiliar with and it surely offers no advantage over any of the more popular and stable Linux distributions and their subordinate distributions stemming from Red Hat, Debian, or SuSE. Thus, it seems that the real question you are asking is why use a less known distribution that offers zero advantages over the popular and highly stable choices? I do not know why anyone would select Arch Linux.

mdevaev, 2024-07-17:
Debian stability is a myth. It is just as prone to problems that may not be fixed there for years, because everyone is used to their presence. Debian also provides packages with its own weird patches, which cause the software to behave differently from what the developer intended. Arch provides packages as is, with minimal patching. Just as an example, when I needed to use Debian, I wanted to use rng-tools. And none of the two packages provided in Debian were the original rng-tools. None of them are compatible with upstream version. One of them is ancient and is not being updated. The second one is some kind of strange offshoot from the maintainer of the package, and also does not look like anything.

In 2024, Debian is still unable to completely get rid of init scripts, despite the fact that systemd was introduced about 10 years ago. Instead of the generally accepted and standard solutions from systemd, Debian uses its own reinvented wheels, which do not work if you need something unusual, such as read-only rootfs.

My story about the disadvantages of Debian can be very long.