pikvm/pikvm #1114: 4k 120Hz monitor alongside the PiKVM

aamaa5, 2023-09-23:
I'm in a situation where I need to connect to my home PC from elsewhere. The PC is originally connected to a 4k 120 Hz monitor. I would like to use the PiKVM to connect to the PC remotely from a laptop and access the BIOS, and I don't mind having a lower resolution.  However, when I'm home and next to the PC I want to use my 4k monitor. Is there a way to connect to my PC using the PiKVM whilst maintaining my 4k 120 Hz setup for when I'm home and use the PC normally? So, 4k 120Hz when using normally, and 1080p when connecting remotely. Two situations for the same PC. Is it possible?

arch1mede, 2023-09-24:
Hi, you would need some sort of HDMI splitter like what j-tech digital offers however, no splitter on the market uses 120hz, only 60Hz. I have one of these and it works fine, the only thing you'll need to do is 2 things, 1) use a v4 PiKVM and 2) once in the os, change the scaling to something higher as the 1:1 scaling is unreadable. If you wish to keep using 120hz, you'll likely need to disconnect from the splitter when you want to use that monitor.

aamaa5, 2023-09-24:
Thank you for the reply! What if I connect the PiKVM to one port on the GPU and the monitor to another port and mirror the outputs, Would this be a better solution?

mdevaev, 2023-09-27:
Yes, it's really better solution.

closed 2023-10-07 by mdevaev