pikvm/pikvm #1118: upload iso error

LordHumphrey, 2023-09-25:
**Describe the bug**
```log
Can't upload image to the Mass Storage Drive:
{ "ok": false, "result": { "error": "MsdIsBusyError", "error_msg": "Performing another MSD operation, please try again later" } }
```

**Screenshots**
[image]

**Additional context**
```log
Sep 25 20:04:33 orangepizero3 kvmd[22291]: kvmd.plugins.msd.otg              INFO --- Writing 'USBOSV3.iso' image (2103969792 bytes) to MSD ...
Sep 25 20:04:45 orangepizero3 kvmd[22291]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [1976.223      http] -- HTTP: DEL client (now=0): [172.18.4.85]:0, id=137135377206c0ff, Resource temporarily unavailable (reading,eof)
Sep 25 20:04:45 orangepizero3 kvmd[22291]: kvmd.plugins.msd.otg              INFO --- Closing image writer ...
Sep 25 20:04:45 orangepizero3 kvmd[22291]: kvmd.plugins.msd.otg             ERROR --- Written 116588544 of 2103969792 bytes to MSD image 'USBOSV3.iso': INCOMPLETE
Sep 25 20:04:45 orangepizero3 kvmd[22291]: kvmd.htserver                    ERROR --- API error: MsdIsBusyError: Performing another MSD operation, please try again later

```

mdevaev, 2023-09-26:
Hello. How can I reproduce it?

LordHumphrey, 2023-09-27:
> Hello. How can I reproduce it?

I don't know, but it was happen.

mdevaev, 2023-09-27:
Every time when you try uploading?

LordHumphrey, 2023-09-28:
> Every time when you try uploading?

Yeah，everytime. I just spilt my SD card and set a ext4 part, then mount it on.

mdevaev, 2023-10-08:
I can't reproduce it. And we don't support orange* and other Raspberry clones. So I don't know what could be the reason, and I have no way to deal with it.

closed 2023-10-08 by mdevaev