pikvm/pikvm #1120: Add second image source

onlineapps-cloud, 2023-09-26:
hi, i used pikvm to connect to my proxmox servers, i have situation when i need to forward pikvm from host to guest OS (to use in pikvm in guest OS).
All working fine, but i cannot see in pikvm guest screen, solution is to add second image source in pikvm.
Very good solution is novnc. Integration guide: https://github.com/novnc/noVNC/blob/master/docs/EMBEDDING.md
It will be great to add switch in header of web page that switch between novnc and standard HDMI source input.
Novnc already have all needed options to connect to proxmox vnc server, also this feature permit to control host and they guest  machines, very useful for IT administrators. Best regards.

onlineapps-cloud, 2023-09-29:
hi, i found old project that have this feature, link: https://github.com/purplepalmdash/pikvmx86
[image]
`Could run both on Physical-Server Or VMs.`

mdevaev, 2023-10-03:
VNC on the page is a solution for a very special case, so it will not be done in this form in upstream. I'm thinking about how to capture multiple sources, but it won't be soon.