pikvm/pikvm #1122: Character Pasting Issue: Incorrect or Missing Characters on Remote Server

jordan30001, 2023-09-28:
**Describe the bug**
When pasting certain characters, they either get pasted incorrectly or do not paste at all on the remote server which occurs across multiple programs, however, for this bug report I will use notepad++

**To Reproduce**
In the "Paste text as keypress sequence", paste the following: 
`` `¬ ! " £ $ % ^ & * ( ) _- =+ [{ ]} ;: '@ #~ ,< .> /? \|`` 
On the remote machine the following gets pasted: 
`` `¬ ! " £ $ % ^ & * ( ) _- =+   ;: '@ #~ , . /? ¦ ``

**Expected behaviour**
The remote machine should have the following sequence of characters pasted: 
`` `¬ ! " £ $ % ^ & * ( ) _- =+ [{ ]} ;: '@ #~ ,< .> /? \|``

**Screenshots**
In the below screenshot, the left text panel in notepad++ is showing the text that I am pasting into the client "Paste text as keypress sequence" textbox.
The right text panel in notepad++ is showing the text that is pasted on the remote server in notepad++
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 (both server and client)
 - Browser Tried Chrome, Opera GX, Edge
 - Version Chrome: 117.0.5938.92, Opera GX: 102.0.4880.64, Edge: 117.0.2045.43

**PiKVM info:**
  - Raspberry Pi board version Pi CM4
  - PiKVM platform PiKVM 4 Plus
  - Video capture type CSI
  - KVMD version: 
    kvmd 3.256-1
    kvmd-fan 0.26-1
    kvmd-oled 0.25-1
    kvmd-platform-v4plus-hdmi-rpi4 3.256-1
    kvmd-webterm 0.43-1
  - uStreamer version: 5.42-1
  - Linux kernel: `Linux [redacted website url] 5.15.92-1-rpi-ARCH #2 SMP Sat Sep 9 00:58:18 MSK 2023 armv7l GNU/Linux`

**Additional context**
`/etc/kvmd/override.yaml` config: 
`
kvmd:
    msd:
        type: disabled
    hid:
        keymap: /usr/share/kvmd/keymaps/en-gb
`

mdevaev, 2023-12-03:
Hello. Did you change the keyboard layout in paste menu?

jordan30001, 2023-12-03:
Yes, i tried en-gb, and just in case i also tried en-us, both of which do not work correctly.

closed 2023-12-04 by mdevaev