pikvm/pikvm #1125: PiKVM V4-Plus not accepting video from EZCOO EZ-SP12H28S

vini-nu, 2023-09-30:
**Describe the bug**
A clear and concise description of what the bug is.

https://www.amazon.de/-/en/dp/B0B56D7NJW EZCOO EZ-SP12H28S

connect scheme: PC -->EZCOO splitter --> LG C2 (it gets 4K 120Hz mode) + PiKVM (pikvm gets downscaled 1080p)

LG C2 works flawlessly. when i connect second HDMI to another monitor EZCOO works too. Problem specific to PiKVM V4-Plus HDMI-IN not accepting 1080p from EZCOO

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: EZCOO EZ-SP12H28S
 - Browser firefox chrome
 - Version [e.g. 22]

Maybe EZCOO 1080p outputs 120Hz. not sure yet.

Will update this with logs and etc as time goes.

vini-nu, 2023-09-30:
60Hz same problem

vini-nu, 2023-09-30:
```bash
[kvmd-webterm@pikvm ~]$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.68-3-rpi-ARCH (alarm@ad2de596cb15) (gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP Mon Oct 31 20:56:54 MSK 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 4 Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 128 MiB at 0x0000000030000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s36300 r8192 d21044 u65536
[    0.000000] pcpu-alloc: s36300 r8192 d21044 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 489792
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:E4:8F:4D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=LABEL=PIROOT ro cma=128M rootwait console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 audit=0
[    0.000000] audit: disabled (until reboot)
[    0.000000] Unknown kernel command line parameters "selinux=0", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x00000000293dc000-0x000000002d3dc000] (64MB)
[    0.000000] Memory: 1715292K/1966080K available (14336K kernel code, 1617K rwdata, 4152K rodata, 2048K init, 629K bss, 119716K reserved, 131072K cma-reserved, 1048576K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000020] Switching to timer-based delay loop, resolution 18ns
[    0.000333] Console: colour dummy device 80x30
[    0.000784] printk: console [tty1] enabled
[    0.000836] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000870] pid_max: default: 32768 minimum: 301
[    0.001090] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001123] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002140] cgroup: Disabling memory control group subsystem
[    0.002305] CPU: Testing write buffer coherency: ok
[    0.002357] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.002607] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003251] Setting up static identity map for 0x200000 - 0x20003c
[    0.003367] rcu: Hierarchical SRCU implementation.
[    0.004369] smp: Bringing up secondary CPUs ...
[    0.005059] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005818] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006529] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006623] smp: Brought up 1 node, 4 CPUs
[    0.006682] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006702] CPU: All CPU(s) started in HYP mode.
[    0.006717] CPU: Virtualization extensions available.
[    0.007321] devtmpfs: initialized
[    0.018123] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.018292] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.018331] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.021325] pinctrl core: initialized pinctrl subsystem
[... 347 lines cut ...]
[    9.852093] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[    9.860665] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[    9.870353] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[... cut ...]
[   21.621954] EXT4-fs (mmcblk0p4): re-mounted. Opts: ,errors=remount-ro. Quota mode: none.
[   21.900184] EXT4-fs (mmcblk0p4): re-mounted. Opts: ,errors=remount-ro. Quota mode: none.
[   22.049415] dwc2 fe980000.usb: new device is full-speed
[   35.804090] systemd-journald[239]: Time jumped backwards, rotating.
[   36.327832] cam-dummy-reg: disabling
[   47.952767] use of bytesused == 0 is deprecated and will be removed in the future,
[   47.960456] use the actual size instead.
[kvmd-webterm@pikvm ~]$ 
```

mdevaev, 2023-10-03:
Hello. Sorry for the late responce. Did you try different edids? https://docs.pikvm.org/edid

mdevaev, 2023-10-08:
Sup?

vini-nu, 2023-10-10:
Yes I tried few different EDID. I think the problem is HDMI-CSI bridge hardware.

mdevaev, 2023-10-10:
We know for sure that the previous versions of Ezcoo hardware work fine. So I'm inclined to blame their switch - apparently they overdid it with EDID override. I can't fix it on my side.

mdevaev, 2023-10-24:
I'm closing it since it has no new info. Please contact us via discord support chat if you need something else.

closed 2023-10-24 by mdevaev

threepack, 2024-01-15:
Not sure if it‘s ok to add here but I have a similar issue with a PiKVM V4 mini and an EZ-SP12HAS Splitter (does that count as older hardware?) (https://www.easycoolav.com/products/4k60-hdmi-splitter-1-in-2-outdolby-vision-optical-audio-breakout); the computer is a Mac M1 mini, the screen is a ~12 year old LG TV

Output to screen is 1080p@60Hz, signal is just duplicated; I‘m currently using the EDID provided for “Splitter stays dark” (DELL D2721H) but I’ve used another one before.

The splitter works with other screens.

I get kind of a flickering screen on the stream, so there’s something - it says 1080p / ~4-6 fps

mdevaev, 2024-01-15:
For support questions please join to discord: https://discord.gg/bpmXfz5