pikvm/pikvm #1129: Resolution dont't scale

Westhoff-MeWe-Consulting, 2023-10-03:
**Describe the bug**
The resolution on my device is changed but the PiKVM doesn't change. 

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'Resolution Settings'
2. Click on 'Change Res to  1920x1080'
3. See error

**Expected behavior**
The windows hast to adapt to the new res.

**Desktop (please complete the following information):**
 - OS: Windows on an VirtualBox
 - Browser Edge 
 - Version 117.0.2045.47 
 - VirtualBox Display Driver WDDM 

**PiKVM info:**
  - RPI 4 4GB
  - PiKVM platform V3
  - Video capture type CSI bridge
  - KVMD version: `kvmd 3.256-1`
  - uStreamer version: `ustreamer 5.42-1`
  - Linux kernel: `5.15.92-1-rpi-ARCH`

Any ideas to get this working?

arch1mede, 2023-10-03:
Hi, which window are you specifically looking for support on?
For the main windows, go into display settings and change to 1920x1080 50hz, there is also an additional setting that might need to be adjusted thats in the advanced display settings, then display adapter properities, then list modes and this might need to be toggled to 192x1080p 50hz. We have sometimes seen this say that its at 1080p, I would change to something else then back to fully sync this setting.

I would also encourage you to jump on [Discord](https://discord.gg/bpmXfz5) to get some live assistance, if you do, make sure and mention this github ticket.

Westhoff-MeWe-Consulting, 2023-10-04:
I think i found the problem. The WWDM driver sends every resolution option with 64Hz. 
I was talking about the window inside of the Edge Tab, after you logged in an clicked on KVM. 
What are the possibilities for my case? Do i have to live with small resolution?

mdevaev, 2023-10-05:
Try to change resolution here: https://www.alphr.com/set-custom-resolution-windows-10/

mdevaev, 2023-10-12:
Sup?