pikvm/pikvm #1130: WebRTC (H.264) does not show video on firefox-esr when installed via the apt package manager on Debian GNU/Linux

Glass2721, 2023-10-06:
**Describe the bug**
When using PiKVM's web interface on firefox-esr (from debian repositories) with the WebRTC/H.264 Video mode selected, it only shows a small black rectangle in the video output window (title of window is: "H.264 – 1280x720 / 0 kbps / 0 fps dynamic").

**To Reproduce**
1. Install/Open Debian Linux (Bug tested on Debian 12 and Debian Testing). Use of a virtual machine may be required.
2. Open a termianl/console and update system by typing: `sudo apt update && sudo apt upgrade`
3. The firefox-esr package should already be installed. Type `sudo apt install firefox-esr` to verify, and install if needed.
4. Open browser called "Firefox ESR" (eg. by searching) and open PiKVM URL.
5. Navigate to the "KVM" page of the PiKVM web interface, and verify that "video mode" is set to WebRTC/H.264
6. You should see that there is no video output (switching to MJPEG/HTTP shows video output normally)

**Expected behavior**
There should be video output shown, instead of a black rectangle.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Debian GNU/Linux
 - Browser: Firefox ESR (Installed via apt using official repositories; irrespective of firefox version)
 - Version: 12 or Testing (Tested on both)

**PiKVM info:**
  - Raspberry Pi board version: RPi 4 (2GB RAM)
  - PiKVM platform: V3 (Third Party Kit: However, I am pretty sure that is not relevant to this problem)
  - Video capture type: CSI bridge (Pretty sure)
  - KVMD version: kvmd 3.257-2, kvmd-fan 0.26-1, kvmd-oled 0.25-1, kvmd-platform-v3-hdmi-rpi4 3.257-2, kvmd-webterm 0.43-1
  - uStreamer version: ustreamer 5.43-1
  - Linux kernel: `Linux (hostname).(domain) 5.15.92-1-rpi-ARCH #2 SMP Sat Sep 9 00:58:18 MSK 2023 armv7l GNU/Linux`

**Additional context**
On the same Debian machine: if I install Firefox ESR via Firefox's Website (https://www.mozilla.org/en-US/firefox/all/#product-desktop-esr), it works **normally** and this bug does not appear. Flatpak, snaps and rpms also don't seem to be affected by this issue.

mdevaev, 2023-10-06:
Hi. If I remember correctly, Debian has some problems with having H264. Either they don't support it because of license paranoia, or something else. Check here: https://bugzilla.mozilla.org/show_bug.cgi?id=1708570

In any case, this is a problem on the Debian side, and there's nothing I can do about it except help you with the search :)

Glass2721, 2023-10-07:
Thanks for the information. After further investigation, I've identified the solution: By default on Debian, the "OpenH264" Plugin is **disabled**. Enabling it fixes the issue.

**Recommendation**: Add a further bullet-point to https://docs.pikvm.org/webrtc/#troubleshooting that says the following: 

> If on Firefox, make sure the OpenH264 Plugin both exists and is enabled (known issue on Debian Linux). How? Press Ctrl+Shift+A to open the Add-ons Manager, then press "Plugins". You should see "OpenH264 Video Codec provided by Cisco Systems, Inc.". Make sure it is enabled by pressing the "more options" button (3 horizontal dots), then pressing "Always Activate". 

Feel free to use or edit the recommendation and close this issue if you consider this issue completed.

mdevaev, 2023-10-07:
Added to the doc, thank you.

closed 2023-10-07 by mdevaev