pikvm/pikvm #1133: Unable to figure out where to set - "--drop-same-frames=120"

cooljimy84, 2023-10-11:
**Describe the bug**
I have edited the override with the below but when runing htop i can see the command is still being run with - "--drop-same-frames=30"

**To Reproduce**
Steps to reproduce the behavior, like:
Edit /etc/kvmd/override.yaml and add option to set drop frame to something else, reboot, use htop to see if the command has run with that option set.

**Expected behavior**
the command should be using the set override options

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: PI4 image

```
vnc:
    keymap: /usr/share/kvmd/keymaps/en-gb
    auth:
        vncauth:
            enabled: true

kvmd:
    msd:
        type:  disabled
    ocr:
        langs:
            - eng

otg:
    manufacturer: Logitech
    product: USB Receiver
    serial: 0
    devices:
        serial:
            enabled: false
        drives:
            enabled: false
        ethernet:
            enabled: false

streamer:
    h264_bitrate:
        default: 5000
    cmd:
        - "--drop-same-frames=120"
```

mdevaev, 2023-10-11:
```yaml
kvmd:
    streamer:
        cmd_remove:
            - "--drop-same-frames=30"
        cmd_append:
            - "--drop-same-frames=120"
```

cooljimy84, 2023-10-11:
> ```yaml
> kvmd:
>     streamer:
>         cmd_remove:
>             - "--drop-same-frames=30"
>         cmd_append:
>             - "--drop-same-frames=120"
> ```

That seems to remove the option in the command when it runs, but doesn't add the appended option. strange

cooljimy84, 2023-10-11:
Also just edited all the files under 
'/usr/share/kvmd/configs.default/kvmd/main/*.yaml'
and changed the option to 120 but it still runs the command with 30 (when i remove the cmd_remove option above)

mdevaev, 2023-10-11:
**Never** edit config.default files.

After changing override, run `kvmd -m` and attach dump here.

cooljimy84, 2023-10-12:
Changed the files back (did an update as well)
override.yaml now looks like this

```
vnc:
    keymap: /usr/share/kvmd/keymaps/en-gb
    auth:
        vncauth:
            enabled: true

kvmd:
    msd:
        type:  disabled
    ocr:
        langs:
            - eng
    streamer:
        cmd_remove:
            - "--drop-same-frames=30"
        cmd_append:
            - "--drop-same-frames=120"

otg:
    manufacturer: Logitech
    product: USB Receiver
    serial: 0
    devices:
        serial:
            enabled: false
        drives:
            enabled: false
        ethernet:
            enabled: false
```
output of kvmd -m is below
```
[root@pikvm ~]# kvmd -m
ipmi:
    auth:
        file: /etc/kvmd/ipmipasswd

    kvmd:
        timeout: 5.0
        unix: /run/kvmd/kvmd.sock

    server:
        host: '::'
        port: 623
        timeout: 10.0

    sol:
        device: ''
        proxy_port: 0
        select_timeout: 0.1
        speed: 115200

janus:
    check:
        interval: 10.0
        retries: 5
        retries_delay: 5.0

    cmd:
        - /usr/bin/janus
        - --disable-colors
        - --plugins-folder=/usr/lib/ustreamer/janus
        - --configs-folder=/etc/kvmd/janus
        - --interface={src_ip}
        - '{o_stun_server}'
    cmd_append: []
[... 399 lines cut ...]
        keepalive:
            count: 3
            enabled: true
            idle: 10
            interval: 3

        max_clients: 10
        no_delay: true
        port: 5900
        tls:
            ciphers: ALL:@SECLEVEL=0
            timeout: 30.0
            x509:
                cert: /etc/kvmd/vnc/ssl/server.crt
                key: /etc/kvmd/vnc/ssl/server.key

    streamer:
        timeout: 5.0
        unix: /run/kvmd/ustreamer.sock

watchdog:
    interval: 30
    rtc: 0
    timeout: 300
```

mdevaev, 2023-10-12:
Did you perfrorm reboot after this change?

cooljimy84, 2023-10-15:
I have indeed.

mdevaev, 2023-10-20:
Sorry for the wait. I checked it myself and everything works. Are you sure that the option hasn't changed after cmd_append and cmd_remove? Because judging by the dump, it happened.

cooljimy84, 2023-10-20:
Agreed it says it clearly in the drump, but i can't see it in the command that is run, see htop screenshot.
If i remove the "cmd_remove" and "cmd_append" then i can see in the command "drop-same-frames=30"

[image]

mdevaev, 2023-10-20:
You should check `journalctl -u kvmd` log. It will there, I assure you.

mdevaev, 2023-10-24:
Sup?

mdevaev, 2023-10-26:
Should be solved, I guess.

closed 2023-10-26 by mdevaev

cooljimy84, 2023-11-01:
Yes totally shows in that output at the end, thanks for your help. Was totally my misunderstanding of the command being run.

mdevaev, 2023-11-01:
:ok_hand: