pikvm/pikvm #1135: Build failing with "IsADirectoryError" with V0 for RPI3

sjthespian, 2023-10-17:
**Describe the bug**
When building an image for V0 on the RPI3, the os build succeeds, but the image build is failing with the following error:
```
IsADirectoryError: [Errno 21] Is a directory: './.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs/usr/share/licenses/common/RUBY'
```
This is a true statement, that is a directory containing just the license.txt file, as is everything else in that directory.

The build is being run on Ubuntu 22.04.3.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Clone the pikvm repo
2. CD to the os directory
3. Create a config.mk file (BOARD = rpi3, PLATFORM = v0-hdmi, HOSTNAME = pikvm. LOCALE = en_US, TIMEZONE = America/Los_Angeles)
4. run `make os`
5. run `make image`

**Expected behavior**
The build should create an image in the images/ directory.

**Screenshots**
The tail end of the build showing the failure, stack trace, and error:
```
#
docker save --output=./.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs.tar pi-builder.pikvm-os.v0-hdmi.arch-rpi3-arm-result
docker run --rm --tty \
                --volume=/home/drich/Src/pikvm/os/.pi-builder/v0-hdmi-rpi3/./.cache:/root/./.cache --workdir=/root/./.cache/.. \
        pi-builder.pikvm-os.v0-hdmi.toolbox \
                /tools/docker-extract \
                        --remove-root \
                        --root=./.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs \
                        --set-hostname="pikvm" \
                        --remove-qemu \
                ./.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs.tar
:: Extracting rootfs to './.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs' ...
Traceback (most recent call last):
  File "/tools/docker-extract", line 182, in <module>
    main()
  File "/tools/docker-extract", line 161, in main
    _extract_rootfs(image_tar, _get_image_layers(image_tar), options.root)
  File "/tools/docker-extract", line 98, in _extract_rootfs
    layer_tar.extractall(
  File "/usr/lib/python3.11/tarfile.py", line 2264, in extractall
    self._extract_one(tarinfo, path, set_attrs=not tarinfo.isdir(),
  File "/usr/lib/python3.11/tarfile.py", line 2331, in _extract_one
    self._handle_fatal_error(e)
  File "/usr/lib/python3.11/tarfile.py", line 2327, in _extract_one
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name),
  File "/usr/lib/python3.11/tarfile.py", line 2418, in _extract_member
    self.makelink(tarinfo, targetpath)
  File "/tools/docker-extract", line 120, in makelink
    self.__remove(targetpath)
  File "/tools/docker-extract", line 125, in __remove
    os.unlink(targetpath)
IsADirectoryError: [Errno 21] Is a directory: './.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs/usr/share/licenses/common/RUBY'
make[1]: *** [Makefile:286: extract] Error 1
make[1]: Leaving directory '/home/drich/Src/pikvm/os/.pi-builder/v0-hdmi-rpi3'
make: *** [Makefile:82: install] Error 2
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS, Docker version 24.0.5, build 24.0.5-0ubuntu1~22.04.1
 - Browser N/A
 - Version N/A
 - VNC client N/A

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]: RPI 3
  - PiKVM platform v0-hdmi
  - Video capture type CSI bridge
[... 69 lines cut ...]
:: Extracting rootfs to './.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs' ...
Traceback (most recent call last):
  File "/tools/docker-extract", line 182, in <module>
    main()
  File "/tools/docker-extract", line 161, in main
    _extract_rootfs(image_tar, _get_image_layers(image_tar), options.root)
  File "/tools/docker-extract", line 98, in _extract_rootfs
    layer_tar.extractall(
  File "/usr/lib/python3.11/tarfile.py", line 2264, in extractall
    self._extract_one(tarinfo, path, set_attrs=not tarinfo.isdir(),
  File "/usr/lib/python3.11/tarfile.py", line 2331, in _extract_one
    self._handle_fatal_error(e)
  File "/usr/lib/python3.11/tarfile.py", line 2327, in _extract_one
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name),
  File "/usr/lib/python3.11/tarfile.py", line 2418, in _extract_member
    self.makelink(tarinfo, targetpath)
  File "/tools/docker-extract", line 120, in makelink
    self.__remove(targetpath)
  File "/tools/docker-extract", line 125, in __remove
    os.unlink(targetpath)
IsADirectoryError: [Errno 21] Is a directory: './.cache/pikvm-os.v0-hdmi.arch-rpi3-arm.rootfs/usr/share/licenses/common/RUBY'
make[1]: *** [Makefile:286: extract] Error 1
make[1]: Leaving directory '/home/drich/Src/pikvm/os/.pi-builder/v0-hdmi-rpi3'
make: *** [Makefile:89: image] Error 2
```

mdevaev, 2023-10-17:
Hello. I can't reproduce it :/ Can I debug it on your machine using `tmate`?

anton-tokarev-softeq, 2023-10-17:
+1 the same error

mdevaev, 2023-10-17:
Found a bug, fixing...

mdevaev, 2023-10-17:
Fixed. Please re-clone entire `os` buildenv.

closed 2023-10-17 by mdevaev

sjthespian, 2023-10-18:
Looks good, I just got a successfully image build! Thanks!!!

mdevaev, 2023-10-19:
:ok_hand: