pikvm/pikvm #1136: Differentiate dual stream data

sreedharch, 2023-10-18:
I'm trying for dual streaming using pikvm. I'm able to run the ustreamer utlity as two different processes for capturing two streams parallelly. From front-end I notice `/streamer/stream?key=${__key}` is being used to stream the data. But, I'm unable to fetch the information how this request call is able to fetch the data from back-end ustreamer. I'm looking how to differentiate the data coming from ustreamer be used on front-end side parallel. Any help on this please?

mdevaev, 2023-10-18:
Hello. `/streamer/stream` is just a proxy to `/stream` handle of the ustreamer instance on /run/kvmd/ustreamer.sock. One ustreamer instance can handle only one camera.

sreedharch, 2023-10-18:
If I use something like /run/kvmd/ustreamer2.sock file in ustreamer utility, will it help streaming the second source?

mdevaev, 2023-10-19:
You will need to change nginx config for this (/etc/kvmd/nginx).

sreedharch, 2023-10-19:
Thanks for the detail @mdevaev , it worked!!

mdevaev, 2023-10-19:
:ok_hand: