pikvm/pikvm #1137: PiKVM folder /kvmd: No such file or directory

rafawallaa, 2023-10-18:
Hello all,

Due to mouse problems in NVR Reolink, i am trying to change setting of absolute mouse, as per PiKVM suggestion/guide.
When i try to change the /etc/kvmd/override.yaml file, i receive response that no such file or directory exist, altough is shows on "ls" insice /etc.
Any toughts?

[image]

I am acessing trough pikvm
 terminal, via w11 PC on chrome 118.0.5993.71

I am using:
  - RPI-3
  - Fresh install of PIKVM (4/5 days ago)
  - Also using Ezcoo 4 port hdmi/usb switch

Thanks

arch1mede, 2023-10-19:
Hi, I would encourage you to follow the first steps document [here](https://docs.pikvm.org/first_steps/#configuring-pikvm) and follow those directions exactly as there seems to be some confusion how you are traveling around the OS.

kvmd and /kvmd are not the same. One is within the same dir, the other is from the root dir which does not exist.

As always, you are more than welcome to join [Discord](https://discord.gg/bpmXfz5) and ask your questions there, we have an active community that will help you out.

mdevaev, 2023-10-19:
This is a basix linux/unix question so please join discord.

closed 2023-10-19 by mdevaev

rafawallaa, 2023-10-19:
Thank you, will do!