pikvm/pikvm #1138: pacman cannot get community.files

pompfe, 2023-10-19:
**Describe the bug**
Just tried to update via pacman and was unable to get community.files (others were ok).  Looked at the mirror (de3 in Germany), the file does not exist.  Looked at other mirrors.  This file is missing there, too.  Directory in mirrors is pretty empty...

**To Reproduce**
Just run pacpman to update installation...

**Expected behavior**
No error.

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - Linux nxpikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-10-19:
Which directory is empty? Please attach the log.

pompfe, 2023-10-20:
The directory in the repository does not have a "community.files":
https://de3.mirror.archlinuxarm.org/armv7h/community/
Missing in other mirrors, too.
Maybe the mirrors are currently broken...
But I have no clue about this flavor, I come from debian.

mdevaev, 2023-10-20:
Ah, I understand. Just remove community repo from /etc/pacman.conf.

pompfe, 2023-10-20:
Thanks a lot!

mdevaev, 2023-10-20:
:ok_hand:

closed 2023-10-20 by mdevaev