pikvm/pikvm #1142: Unable to detect Raspberry PI V2 camera

sreedharch, 2023-10-26:
I tried connecting "Raspberry PI 5MP Camera Board Module" V2 camera to pikvm and started running the command:
$ sudo modprobe bcm2835-v4l2
$ libcamerify ./ustreamer --host :: -e m2m-image

I notice that the ustreamer throws following error:
Device fd=8 opened
Can't query device capabilities: Inappropriate ioctl for device
Device fd=8 closed

pikvm is unable to capture the camera streaming. Can you please help on this.

mdevaev, 2023-10-26:
Try option `--device` with various /dev/videoX.

sreedharch, 2023-10-26:
yes tried, but didn't work. same error seen.

mdevaev, 2023-10-26:
Sorry, but I can't help with this. I don't have V2 camera.

sreedharch, 2023-10-26:
i tried running
ustreamer --host :: --buffers=3 -r 640x480 --device=/dev/video0 --debug

Following is the Output:
Starting PiKVM uStreamer 5.44 ...
Using internal blank placeholder
Installing SIGINT handler ...
Installing SIGTERM handler ...
Ignoring SIGPIPE ...
Binding HTTP to [::]:8080 ...
Listening HTTP on [::]:8080
Using V4L2 device: /dev/video0
Using desired FPS: 0
_stream_init_loop: stream->run->stop=0
Starting HTTP eventloop ...
==============================================
Device fd=8 opened
Querying device capabilities ...
Using capture type: single-planar
Using input channel: 0
Using TV standard: DEFAULT
Probing device format=YUYV, stride=1280, resolution=640x480 ...
Using resolution: 640x480
Using format: YUYV
Querying HW FPS ...
Querying HW FPS changing is not supported
Using IO method: MMAP
Requesting 3 device buffers for MMAP ...
Requested 3 device buffers, got 3
Allocating device buffers ...
Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
Mapping device buffer=0 ...
Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
Mapping device buffer=1 ...
Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
Mapping device buffer=2 ...
Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
Device fd=8 initialized
Starting device capturing ...
Can't start capturing: Invalid argument
Releasing device buffers ...
Closing device ...
Device fd=8 closed
Sleeping 1 seconds before new stream init ...
HTTP: Updating exposed frame (online=0) ...
HTTP: Exposed frame: online=0, exp_time=0.000000
==============================================

I see below error:
Can't start capturing: Invalid argument

Don't get which argument it's referring as invalid.

sreedharch, 2023-10-26:
Sorry for the confusion and thanks for your support @mdevaev . Issue resolved and its working.
I didn't enable legacy camera mode in raspi.

mdevaev, 2023-10-26:
How did you do this? I want to add this for the doc.