pikvm/pikvm #1144: Unable to update using pacman -Syu due to conflicting dependencies

Majestic7979, 2023-10-27:
**Describe the bug**
```
[root@pikvm kvmd-webterm]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
warning: python-ajsonrpc: local (1.2.0-4) is newer than extra (1.2.0-3)
:: Replace raspberrypi-bootloader-x-pikvm with pikvm/raspberrypi-bootloader-pikvm? [Y/n] n
resolving dependencies...
looking for conflicting packages...
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: raspberrypi-utils and raspberrypi-firmware-pikvm are in conflict
```

also tried answering yes but same issue:
```
[root@pikvm kvmd-webterm]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
warning: python-ajsonrpc: local (1.2.0-4) is newer than extra (1.2.0-3)
:: Replace raspberrypi-bootloader-x-pikvm with pikvm/raspberrypi-bootloader-pikvm? [Y/n] y
resolving dependencies...
looking for conflicting packages...
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: raspberrypi-utils and raspberrypi-firmware-pikvm are in conflict
```

I don't have access to Discord so I cannot use that to obtain support.

mdevaev, 2023-10-27:
See here: https://www.reddit.com/r/pikvm/comments/17dm3is/the_big_pikvm_os_update/

closed 2023-10-27 by mdevaev

Majestic7979, 2023-10-27:
Thank you very much, it worked. Have a nice day!

mdevaev, 2023-10-27:
:ok_hand:

rotanigroc, 2023-11-14:
Copying the actual resolution here in case the Reddit post is removed in the future for any reason --

```
Posted by [u/Liksys](https://www.reddit.com/user/Liksys/)
The big PiKVM OS update

There have been significant changes in the upstream distribution, so a new OS update requires a few more actions than usual. But don't panic! I wrote a script that will do everything for you.

To install the new update, run under the root user:

curl https://files.pikvm.org/update-os.sh | bash

You can also pre-download the script and view its contents if you don't trust me %)
```

mdevaev, 2024-01-17:
@asad26 You haven't updated the OS for a long time, there have been significant changes in the upstream. Please reflash the memory card: https://docs.pikvm.org/flashing_os/