pikvm/pikvm #1145: Please support audio

yurivict, 2023-10-31:
**Is your feature request related to a problem? Please describe.**

In order to fully control a computer it is necessary to control Video-Keyboard-Mouse-Audio.
Based on the description, audio is missing.
PIKVM should present to a computer a simple USB audio device and to pass audio bi-directionally to the client.

Use model: companies nowadays disallow any type of connections to their laptops except Video-Keyboard-Mouse-Audio. In order to still use such laptops virtually, from any other hosts, and without breaking any rules, control over Video-Keyboard-Mouse-Audio is needed. Missing audio is a significant deficiency for such use model.

**Describe the solution you'd like**
USB audio device needs to be supported.

**Describe alternatives you've considered**
None exists.

mdevaev, 2023-10-31:
Hello. We already have HDMI audio capture. Is that not enough?

arch1mede, 2023-10-31:
Hi, please see https://github.com/pikvm/pikvm/issues/97
Just know that 2 way audio is software based AKA client/server based, right now PiKVM is clientless.

yurivict, 2023-10-31:
I don't have the physical device yet, so I only saw what is in the description.
----------------------------
> Hello. We already have HDMI audio capture. Is that not enough?

What about playback?

> Just know that 2 way audio is software based AKA client/server based, right now PiKVM is clientless.

I was going to propose another feature: support the RDP protocol, so that network RDP clients, like Remmina, would be able to control the computer.

PiKVM could be essentially a converter between HDMI+USB and RDP, so that the user would be able to work on this computer via RDP from any location the same way as he would be able to work when he is physically on that computer.
This requires audio capture and playback.

arch1mede, 2023-10-31:
**_What about playback?_**
Audio playback is supported

**_I was going to propose another feature: support the RDP protocol, so that the network RDP clients, like Remmina, would be able to control the computer._**
You are perhaps mixing a software solution to a hardware, this is a hardware solution - its not going to be possible to make this into a software solution, at least not in the same sense as you are thinking.

**_PiKVM could be essentially a converter between HDMI+USB and RDP, so that the user would be able to work on this computer via RDP from any location the same way as he would be able to work when he is physically on that computer.
This requires audio capture and playback._**
Please see above.

yurivict, 2023-10-31:
> You are perhaps mixing a software solution to a hardware, this is a hardware solution - its not going to be possible to make this into a software solution, at least not in the same sense as you are thinking.

What would be a software solution?

Why is this not possible?

mdevaev, 2023-10-31:
> What about playback?

You mean `client -> pikvm -> remote host` direction?

> I was going to propose another feature: support the RDP protocol, so that the network RDP clients, like Remmina, would be able to control the computer.

You mean RDP instead of the Web UI? If yes, we have VNC as an alternative. I have no desire to implement RDP because it's almost the same. And then, we have already added H.264 support to TigerVNC, and there is no point in doing the same work for RDP.

yurivict, 2023-11-01:
> You mean RDP instead of the Web UI?

Yes.

> If yes, we have VNC as an alternative. I have no desire to implement RDP because it's almost the same.

RDP is different from VNC in these respects:
1) RDP supports audio, and VNC doesn't
2) RDP supports file sharing, and VNC doesn't
3) RDP has a lot better performance, because a lot of performance improvements were implemented in it, compared to almost none in VNC

But RDP doesn't need to be reimplemented. In order to add RDP on my FreeBSD box I just installed the xrdp package and slightly adjusted the configuration to work through VNC. I can provide the details if necessary.

RDP is implemented on top of VNC. The same should work on PiKVM, and you should be able to just add the RDP daemon on Raspberry Pie.

From the users' perspective RDP is a lot better compared to VNC.

NiKiZe, 2023-11-01:
RDP is mainly used for known drawings from the host, such as rectangles, lines and glyphs.
VNC on the other hand is image/screenshot based.

RDP is not really suitable for the kind of transfers that a image grabbing device like KVM is. Sure it can be done, but the benefits that is mentioned for performance really isn't there in this usecase. RDP do have h262 support as well, but not all clients support it, so you need to implement fallbacks for that as well.

You will be adding quite a lot of complexity, for not that much gain. (Sure audio support is nice, but just use the browser implementation instead)

mdevaev, 2023-11-02:
@yurivict 

> RDP supports audio, and VNC doesn't

VNC supports pcm audio, but clients usually don't implement it. There are two reasons why this is irrelevant for us now:
* uStreamer can only transmit audio via webrtc, that is, there is no mechanism to feed the audio stream to another server.
* There is a draft extension for Opus audio in MTS, which is very suitable for us. That is, if I decide to make audio not only on the web (more precisely, I will find time for this), I will prefer to implement it in VNC because we already have experience for this (I will explain below)

> RDP supports file sharing, and VNC doesn't

With all our desire, we will not be able to do anything about it because PiKVM does not have software access to the target host and you will not be able to transfer the file there in any way except using Mass Storage emulation.

> RDP has a lot better performance, because a lot of performance improvements were implemented in it, compared to almost none in VNC

The VNC server in PiKVM was written from scratch with a huge number of optimizations for our specifics. It works faster than the Web UI. In addition, we implemented H.264 in VNC, pushed it into the standard and TigerVNC already supports it natively. Moreover, compression occurs by hardware. According to my calculations, we almost ran into the physical latency limit, except that something can be optimized in TigerVNC.

If you do RDP on top of VNC, I think it will only add latency and will not improve anything, because our VNC server does everything very optimally anyway. But if you want to give it a try, I'd be interested to see the results.

dyehya, 2023-11-17:
I'm out of my depth here and the way audio is referenced I need to separate out into 2 different parts. There is audio out from the machine and audio into the machine from remote system. The PiKVM already does audio out from the machine to WebRTC but not VNC if I am understanding correctly.

The part that I think is being requested or that I would request is pass-through microphone support in WebRTC and VNC. Just like we can emulate mass storage, keyboard, mouse, via USB, a microphone input being added/supported and configurable like VEN/DEV Id overrides. This is a big ask but would allow for remote meetings and other use cases where we need to pass live audio into it.

yurivict, 2023-11-17:
> The PiKVM already does audio out from the machine to WebRTC but not VNC if I am understanding correctly.

My reading is that PiKVM does support VNC audio but the audio can only work on clients that support VNC audio, which is a minority of VNC clients.

mdevaev, 2023-11-17:
@dyehya We are running out of [USB endpoints](https://docs.pikvm.org/_usb_limits/) to do this. I will look for other ways to implement the microphone, but this is discussed in another issue.

@yurivict Audio in PiKVM works only in the browser over WebRTC. Supported TC358743 video capture device or our ready-made device from the store. We don't support audio in VNC at all, because, as you said, very few clients can do it, and also because it will require a significant redesign of the backend on PiKVM side. If there are many people willing, we will modify TigerVNC to support OPUS and make audio in VNC.

As for VNC clients, I just advise using TigerVNC. Everyone else has some problems with the set of features or non-compliance with the standard.

yurivict, 2023-12-04:
Your customer base should be employees of companies that have to work from home on company-issued laptops.

The companies often issue cheap laptops of very poor quality. Some Dell Windows laptops which are very thick, weighing 5+ pounds, and barely able to connect to WiFi, and failing to hibernate when the lid is closed. It is very inconvenient to carry such laptop. In my case it fails to connect to 50% of WiFi networks without an apparent reason.

In case when the person needs to travel or to be at some conference or at a coffee shop - it is very difficult to carry such laptop because it is heavy and often doesn't connect to WiFi.

It is much easier to leave it at home and to connect to it through PiKVM from some good sleek laptop like MacBook Air.

This way it is easy to work while traveling, and easy to integrate the work environment with the external monitor, keyboard and mouse.

At least this is a problem that I face.

People usually either have a regular KVM switch, or an extra monitor, keyboard and mouse. But these solutions don't integrate into the external desktop system, and don't work during travel. The PiKVM allows to integrate work into any existing system, and works great during travel.

In order to work on a system all day access though the browser would hardly work.

VNC should support audio in addition to video for a comfortable full control for extended periods of time.

mdevaev, 2023-12-04:
Okay, so be it.

closed 2023-12-04 by mdevaev

mdevaev, 2025-01-20:
Official [PiKVM V3 and V4 devices](https://pikvm.org/buy) gets two-way audio with microphone.

Now you can use voice applications remotely, or you can just emotionally tell the server what you think about it.

**https://docs.pikvm.org/audio**

*PS: I know that many people wanted this feature. I spent all the New Year holidays on it, so if you found PiKVM useful and finally got what you wanted, you can support my work on PiKVM or just give me a beer with donation to [Patreon](https://www.patreon.com/pikvm) or [Paypal](https://paypal.me/pikvm). Happy PiKVMing!*

izphi78, 2025-02-26:
Hi there! Thank you for this feature !  :) @mdevaev 
Is there a technical reason for not supporting the DIY flavor of PiKVM?

mdevaev, 2025-02-26:
DIY does not support HDMI audio capture. Other options may be unstable and
I didn't test them.

mdevaev, 2025-03-04:
I didn't try.

izphi78, 2025-03-12:
After some research on [this site](https://wiki.blicube.com/blikvm/en/hdmi-csi-i2s/#software-demo), I successfully got it working with the V2.

I have a Geekwork X630 and recently bought a Geekworm X1300-A2 to enable audio via I2S right after discovering that PiKVM now supports two-way audio.

For fellow V2 users, [here](https://wiki.geekworm.com/Enable_pikvm_hdmi_audio) is a guide you can follow to enable two-way audio support.

Once again, a huge thank you to @mdevaev for implementing this fantastic feature!