pikvm/pikvm #1148: Large then one line text recognition in firefox causes error

damige, 2023-11-06:
When in firefox when selecting multiple lines to OCR, the error: 
"Can't copy text to the clipboard: NotAllowedError: Clipboard write was blocked due to lack of user activation."  appears

Steps to reproduce the behavior, are
1. Click Text on menu
2. Select area for eng
3. Select an area that's more then one line (selecting one line works, chromium works regardless of line count)

I would have expected no error, and text is copied
issue was also raised here: https://www.reddit.com/r/pikvm/comments/sxkijf/kvmd_355_copying_text_from_the_pc/

This is on:
 - OS: Linux (Arch)
 - Firefox 119

PiKVM v4 running latest updates + reboot

mdevaev, 2023-11-07:
Note for me: https://web.dev/articles/async-clipboard?hl=en

mdevaev, 2023-12-02:
I made a workaround, but I want to point out that [Firefox lives in its own world](https://bugzilla.mozilla.org/show_bug.cgi?id=1560373).

closed 2023-12-02 by mdevaev

damige, 2023-12-02:
Confirmed fixed, thanks!