pikvm/pikvm #1151: Please consider adding a feature comparison table to the website

yurivict, 2023-11-10:
**Is your feature request related to a problem? Please describe.**
It is quite confusing to understand what's the difference between PiKVMv3, PiKVMv4, PiKVMv4 Mini.

**Describe the solution you'd like**
Please consider adding a product comparison matrix with products along the horizontal axis, features along the vertical axis, and checks where features available for products.

**Describe alternatives you've considered**
The textual description present on the website is not easy to read.

**Additional context**
It isn't clear at all what the differences are. Is PiKVMv4 the most feature full version?

mdevaev, 2023-11-10:
I'm working on it :)

> Is PiKVMv4 Plus the most feature full version?

Yes.

arch1mede, 2023-11-20:
> USB host support (external USB device ports)

KVM Switches for example

> Additional USB storage support (internal USB port)

USB flash drive can be inserted, along with a ssd/nvme 

> RJ-45 console port (this appears to mean the second Ethernet port, but it isn't clear why is it needed)

This is a console port much like what Cisco uses, you can either use this port or the usbc port for console access

mdevaev, 2024-01-27:
It's done.

closed 2024-01-27 by mdevaev

anssih, 2024-05-04:
It is not discoverable from the website (www.pikvm.org), though.

I spent quite a long time going through the various pages there trying to find a feature comparison, before resorting to google and finding this bug which pointed me to README of this repo :)

mdevaev, 2024-05-04:
Our website is a fossil, we will update it soon.

mikulik86, 2024-09-21:
Still nothing on the website. Google search doesn't bring up the readme table.