pikvm/pikvm #1152: Make Image throws rsync error

Black616Angel, 2023-11-11:
**Describe the bug**
While trying to create a Pi3V2-hdmi image, the `make image` operation throws multiple errors while trying to rsync.

**To Reproduce**
Steps to reproduce the behavior, like:
1. follow the steps of https://docs.pikvm.org/building_os/
2. config.mk:

```
# Base board
BOARD = rpi3

# Hardware configuration
PLATFORM = v2-hdmi

# Target hostname
HOSTNAME = pikvm

# ru_RU, etc. UTF-8 only
LOCALE = en_US

# See /usr/share/zoneinfo
TIMEZONE = Europe/Berlin

# For SSH root user
ROOT_PASSWD = root

# Web UI credentials: user=admin, password=adminpass
WEBUI_ADMIN_PASSWD = admin

# IPMI credentials: user=admin, password=adminpass
IPMI_ADMIN_PASSWD = admin
```

3. `make os`
4. `make image`

**Expected behavior**
Usually an image would be created.

**Error returned**
<details>
  <summary>Error</summary>

```
Allocating group tables: done
Writing inode tables: done
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done

CMD [ /tools/disk ] ==> /usr/bin/partprobe /root//dev/loop13
CMD [ /tools/disk ] ==> /usr/bin/partx -vu /root//dev/loop13
partition: none, disk: /root//dev/loop13, lower: 0, upper: 0
/root//dev/loop13: partition table type 'dos' detected
/root//dev/loop13: partition #1 added
/root//dev/loop13: partition #2 added
/root//dev/loop13: partition #3 added
/root//dev/loop13: partition #4 added
CMD [ /tools/install ] ==> /tools/disk mount /root//dev/loop13 __mnt__
CMD [ /tools/disk ] ==> /usr/bin/mkdir -p __mnt__//
CMD [ /tools/disk ] ==> /usr/bin/mount /root//dev/loop13p3 __mnt__//
CMD [ /tools/disk ] ==> /usr/bin/mkdir -p __mnt__//boot
CMD [ /tools/disk ] ==> /usr/bin/mount /root//dev/loop13p1 __mnt__//boot
CMD [ /tools/disk ] ==> /usr/bin/mkdir -p __mnt__//var/lib/kvmd/pst
CMD [ /tools/disk ] ==> /usr/bin/mount /root//dev/loop13p2 __mnt__//var/lib/kvmd/pst
CMD [ /tools/disk ] ==> /usr/bin/mkdir -p __mnt__//var/lib/kvmd/msd
CMD [ /tools/disk ] ==> /usr/bin/mount /root//dev/loop13p4 __mnt__//var/lib/kvmd/msd
CMD [ /tools/install ] ==> rsync -a --quiet ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/bin ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/boot ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/dev ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/etc ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/home ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/lib ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/mnt ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/opt ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/proc ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/root ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/run ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/sbin ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/srv ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/sys ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/tmp ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/usr ./.cache/pikvm-os.v2-hdmi.arch-rpi3-arm.rootfs/var __mnt__
rsync: [generator] chown "/root/__mnt__/boot" failed: Operation not permitted (1)
rsync: [generator] chown "/root/__mnt__/boot/overlays" failed: Operation not permitted (1)
[... 362 lines cut ...]
rsync: [receiver] chown "/root/__mnt__/boot/overlays/.watterott-display.dtbo.IZcNbn" failed: Operation not permitted (1)
rsync: [receiver] chown "/root/__mnt__/boot/overlays/.waveshare-can-fd-hat-mode-a.dtbo.vNeza9" failed: Operation not permitted (1)
rsync: [receiver] chown "/root/__mnt__/boot/overlays/.waveshare-can-fd-hat-mode-b.dtbo.fUICse" failed: Operation not permitted (1)
rsync: [receiver] chown "/root/__mnt__/boot/overlays/.wittypi.dtbo.V3g8TB" failed: Operation not permitted (1)
rsync: [receiver] chown "/root/__mnt__/boot/overlays/.wm8960-soundcard.dtbo.CqT8OJ" failed: Operation not permitted (1)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1336) [sender=3.2.7]
CMD [ /tools/install ] ==> losetup -d /root//dev/loop13
make[1]: *** [Makefile:316: image] Fehler 1
make[1]: Verzeichnis „/mnt/NVMe/Programmierung/Zeug/PiKVM/os/.pi-builder/v2-hdmi-rpi3“ wird verlassen
make: *** [Makefile:91: image] Fehler 2
```
</details>

**Desktop (please complete the following information):**
 - OS: Pop_OS 22.04
 - Browser firefox
 - Version 119

**PiKVM info:**
  - Raspberry Pi board version 3
  - PiKVM platform v2-hdmi
  - Video capture type CSI bridge

**Additional context**
-

mdevaev, 2023-11-11:
Hello. Why you're using V2 with RPi3? It doesn't supported.

mdevaev, 2023-11-17:
No answer, closing.

closed 2023-11-17 by mdevaev