pikvm/pikvm #1158: I'm trying to figure out how to map override vnc keymaps... and I'm not sure if I'm doing this wrong, or this is a bug...

brighton36, 2023-11-15:
I have the following keymap loading, in my /etc/kvmd/override.yaml:
```
  vnc:
      keymap: /usr/share/kvmd/keymaps/en-us-caps2ctrl
```

And, as indicated by the log:
```
  kvmd.keyboard.keysym              INFO --- Reading keyboard layout /usr/share/kvmd/keymaps/en-us-caps2ctrl
```
This configuration directive would appear to be parsed and active. 

However, it doesn't seem as if anything I do this in file, has any effect in vnc. As you can guess by the name, I wanted to switch the caps and control keycodes... But, that doesn't seem to be working. And, even just trying to change the 'a' keypress to send 'b' (and vice versa) doesn't seem to achieve this.  Here's the relevant changes to my keymap for reference:
```
  [root@pikvm ~]# cat /usr/share/kvmd/keymaps/en-us-caps2ctrl | grep -P '0x1e|0x30'
  # evdev 30 (0x1e), QKeyCode "a", number 0x1e
  a 0x30
  A 0x30 shift
  # evdev 48 (0x30), QKeyCode "b", number 0x30
  b 0x1e
  B 0x1e shift
```
That file was duplicated from en-us, which, has these values reversed:
```
  [root@pikvm ~]# cat /usr/share/kvmd/keymaps/en-us | grep -P '0x1e|0x30'
  # evdev 30 (0x1e), QKeyCode "a", number 0x1e
  a 0x1e
  A 0x1e shift
  # evdev 48 (0x30), QKeyCode "b", number 0x30
  b 0x30
  B 0x30 shift
```
I just ran a pacman -Syu before starting on this path, this morning. And, it's uptodate, and restarted thereafter. 

Any ideas what I'm missing?

mdevaev, 2023-11-17:
VNC has two methods of working with the keyboard: using keymap and using direct hardware codes. PiKVM uses the second as a priority, and the first, unless otherwise supported.

All major clients work using the second method, and the first one exists only for compatibility.

Maybe I'll just make an option to swap caps and ctrl for you?

brighton36, 2023-11-17:
Thank-you so much for this reply. And the offer is very gracious. I'm using tigervnc, and interception. So, the key bindings are a little crazy on my end, and, I'm not sure what makes the most sense. 

Do you know if tigervnc supports the keymap mode of operation? I'm looking through the help output, and I don't think that's an option. Perhaps you can tell me which client supports this mode?

It seems to me that the caps swap... is maybe not the most encompassing solution. I like the keymap a lot, and I'd vote to apply this feature to the 'direct hardware code' mode, if that's an option...

mdevaev, 2023-11-17:
> Do you know if tigervnc supports the keymap mode of operation? I'm looking through the help output, and I don't think that's an option. Perhaps you can tell me which client supports this mode?

TigerVNC does not support this and I don't know any other clients that could do this.

> I like the keymap a lot, and I'd vote to apply this feature to the 'direct hardware code' mode, if that's an option...

It's not possible. This will create more problems than it will solve because the keymap conversion is not perfect. Hardware keys are not uniquely determined by them from the logical codes. I generally plan to get rid of them in the future, when all major VNC clients will switch to physical keys.

So, should I make you the option?

brighton36, 2023-11-17:
That is such a generous offer, and I'm going to say 'no', because ... well, that's not actually what I'm looking for. (I've bound a lot of keys on my end, such as tab to alt, shift to del, etc) I think I can make this work through interception a bit differently. But, it would be very useful if you could show me some kind of documentation on how x keyboard events, get sent to vnc, and/or the keyboard of the pikvm machine. (Perhaps even showing me where in the source code that's being done here, would be useful) . If/when I figure out things on my end, perhaps I can share a pr, if I see that I can add any value to the project here...

mdevaev, 2023-11-17:
Here the VNC server: https://github.com/pikvm/kvmd/blob/master/kvmd/apps/vnc/server.py

It simply converts VNC physical keys into KVMD physical key names. If VNC uses X11 codes, then first the conversion takes place using keymap.

brighton36, 2023-11-17:
Thank-you so much for your help, and your dedication to this useful project. Have a great weekend!

closed 2023-11-17 by brighton36

mdevaev, 2023-11-17:
:ok_hand: